nRF24/RF24 #716: printPrettyDetails - Primary Mode

DrClaudio73, 2021-01-07:
**Describe the bug**
A clear and concise description of what the bug is.

I am wondering if in file RF24.cpp the code at line 631 - within implementation of method "void RF24::printPrettyDetails(void)"- shouldn't it be like this:

    printf_P(PSTR("Primary Mode\t\t= %cX\r\n"), config_reg & _BV(PRIM_RX) ? 'R' : 'T');

instead of:

    printf_P(PSTR("Primary Mode\t\t= %cX\r\n"), config_reg & _BV(PWR_UP) ? 'T' : 'R'); 

?

Please include:
1. Code to reproduce

....... init RF .......
  radio.startListening();
  radio.printPrettyDetails();
  radio.stopListening();                                    // First, stop listening so we can talk.
  radio.printPrettyDetails();
.... program continues

2. Expected behaviour

.... 1st call to printPrettyDetails previous output ...

Primary Mode            = RX

.... 2nd call to printPrettyDetails previous output ...

Primary Mode            = TX

3. What device(es) are you using? Please specify make and model.
nRF24L01+

**Additional context**
Thank you for your great work. Please accept my deepest apologies if this is not the proper way to submit this request.

2bndy5, 2021-01-07:
Thank you for bringing this to our attention. You are correct in raising this as a separate issue, but @combs has already made us aware via a comment on another issue. This is issue will be corrected by PR #707.

DrClaudio73, 2021-01-07:
As per the received reply I close this issue. Thank you.

closed 2021-01-07 by DrClaudio73

2bndy5, 2021-01-07:
If that's what you want. I only opened it because the Pull Request I linked to this issue will automatically close it when the Pull Request is merged.

DrClaudio73, 2021-01-07:
Okay thank you so much.

PR #707 (2021-01-08): Fix print pretty
Fixes #706 and closes #716. Both fixes in this PR resolves errors noticed by @combs & @DrClaudio73

1. Use binary representation to show the auto-ack feature when configuration differs per pipe
2. Uses correct mnemonic to output "primary mode"

Tested changes using python wrapper; it works as expected.