nRF24/RF24 #717: Problem to compile RF24Gateway sample

arnold-b, 2021-01-09:
root@Rasp-VPN:~/rf24libs/RF24Gateway/examples# make install all
g++ -std=c++0x -Wall -I../ -lrf24-bcm -lrf24network -lrf24mesh -lrf24gateway RF24GatewayNode.cpp -o RF24GatewayNode
/usr/bin/ld: //usr/local/lib/librf24-bcm.so: undefined reference to symbol 'delay'
/usr/bin/ld: //usr/lib/libwiringPi.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make: *** [Makefile:37: RF24GatewayNode] Fehler 1

Hardware Rasp 3 , 5.4.83-v7+

Thanks for helping

2bndy5, 2021-01-09:
>/usr/bin/ld: //usr/lib/libwiringPi.so: error adding symbols: DSO missing from command line

I'm not sure if the wiringPi driver still works. I raised a similar issue about this  #669 

Can you use SPIDEV for your project?

TMRh20, 2021-01-09:
WiringPi isn't specifically supported with RF24Gateway, so you need to add `-lwiringPi` in the `${PROGRAMS}: ${SOURCES}` section of the makefile for it to compile correctly.

Also not sure if wiringPi driver works/still works especially with the newer RPis.

arnold-b, 2021-01-09:
Thank you. 
I change to SPIDEV 

After that it get compiled

arnold-b, 2021-01-09:
Thanks, 
yes i changed. Now I get it compiled

closed 2021-01-09 by 2bndy5