nRF24/RF24 #719: Display(s) driver conflicts (SPI and I2C) with the RF24 library

donniebergh, 2021-01-10:
I'm have trouble getting any TFT/LCD display to work along side the RF24 library on an Arduino Nano. To date I have tried 3 different displays all using some form of the Adafruit or Arduino drivers.

I have narrowed my problem to the moment  the radio is initialized with  "RF24 radio(10,9); // CE, CSN" the display(s) stop working. I have also tried moving the CE,CSN pins around with no success. The radio continues to work and I can display the data on the Serial monitor - but no long on the display.

I'm using the following hardware:

- Arduino Nano V3
- Several displays - the latest I have tried is a "240x240 IPS ST7789" using SPI. Have also used a 1306 (128x64 pixels) on the I2C bus
- NRF24L01 transceiver module

Can you help me - Is there a fix for this?

Thank you,
-Donnie

2bndy5, 2021-01-10:
Can you provide store links for the displays? "1306" isn't helpful. ST7789 is the driver IC for the display, but it doesn't describe the board the display is on. I'm curious because most of the SPI display boards that I saw (when recently shopping for a TFT display) included an SD card reader. So, that might be detrimental to using the nRF24L01 over the same SPI bus.

@Avamander I did happen to buy a TFT (ST7789 + SD card reader) & an OLED (default I2C xor optional SPI interfaces - no SD card reader) for exploring #714

@donniebergh Maybe when they arrive I will be better suited to help you debug this issue.

donniebergh, 2021-01-10:
Yes see below. All were purchased within the last year.

I’m interested in small displays. I plan to build (well 3D print) a box for the project I’m building. The transmitter portion of the project will be battery powered and portable. None of the displays I’m interested in has an SD reader.

These two use the SPI bus:
https://www.amazon.com/gp/product/B01N1LZQA8/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 <https://www.amazon.com/gp/product/B01N1LZQA8/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1>

https://www.amazon.com/gp/product/B08FB77YY2/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 <https://www.amazon.com/gp/product/B08FB77YY2/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1>

This one used the I2C Bus:
https://www.amazon.com/gp/product/B07F3KY8NF/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 <https://www.amazon.com/gp/product/B07F3KY8NF/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1>

Thank you,
-Donnie

TMRh20, 2021-01-12:
Seems odd you would have issues with I2C, but sometimes there are SPI conflicts which means you need an alternate BUS for either the display or RF24. See RF24_config.h for enabling software SPI and select your chosen pins.

2bndy5, 2021-01-12:
@donniebergh just to be clear, the radio's CSN pin is not the same pin used for the ST7789's CS pin right? And you are only trying to connect 1 display at a time? These questions are so basic I forgot to ask.

donniebergh, 2021-01-13:
Hello-

I used the I2C bus display in the early testing/development. I was having similar communications issues with it plus the display is mono and small. I moved on to a larger color display(s) which use the SPI bus. It is possible that I had issues with the I2C bus test that were my fault - I’ll revisit it in a bit.

Below is the initialization code for the SPI display app. The 1.3 inch display does not seem to have a CS pin just SCL, SDA, RES, DC and BLK pins. I can get text to display - when the radio is started the display goes blank.

I'm just using one display and one radio built into the Arduino Nano RF (NRF24L01) . 

I created a small app to test the issue I have - it is attached below.

#include <SPI.h>
#include <RF24.h>
#include <Adafruit_GFX.h>   // Core graphics library by Adafruit
#include <Arduino_ST7789.h> // Library for ST7789 (with or without CS pin)

#define TFT_RST   8     // ST7789 Reset *1 (8,6)
#define TFT_DC    4     // Data/Command *1 (9,7)
#define TFT_MOSI  11    // SPI data pin
#define TFT_SCLK  13    // SPI sclk pin

Arduino_ST7789  tft = Arduino_ST7789(TFT_DC, TFT_RST, TFT_MOSI, TFT_SCLK); //for display without CS pin
RF24 radio(10,9); // CE, CSN (Arduino Nano RF)

-Donnie

2bndy5, 2021-01-13:
ok, glad I asked. Most SPI devices need to have a CS (Chip Select) or CSN (Chip Select Not -- same behavior, different label) so that the MCU (the nano) can tell which SPI device to pay attention to the communication happening on the SPI bus (MISO, MOSI, & SCK/SCLK pins). Without attaching the ST7789's CS pin to the MCU, the display is likely paying attention to all communication over the SPI bus, even when the data over MOSI is only intended for the nRF24.

According to the code you posted, and the [adafruit library for the ST7789](https://github.com/adafruit/Adafruit-ST7735-Library/blob/55bed633293e1be2510d64bf3c05884e6129273c/Adafruit_ST77xx.cpp#L48), you are trying to use software driven SPI data over the nano's designated pins for hardware driven SPI. I highly recommend you change the `TFT_MOSI` & `TFT_SCLK` pins to something that isn't also attached the nRF24. [Or just use the ST7789's CS pin](https://github.com/adafruit/Adafruit-ST7735-Library/blob/55bed633293e1be2510d64bf3c05884e6129273c/Adafruit_ST77xx.cpp#L63). Notice that the ST7789 is one of those few SPI devices that doesn't use the MISO (because its an output only device).

p.s. SCL & SDA are terms usually reserved for the I2C bus, but they are often used as a reference for the SPI pins SCK & MOSI/MOSI (as evident by the ST7789 datasheet which doesn't explicitly mention using I2C). I've also seen SDO & DO used for MOSI and SDI & DI used for MISO. The SCK/SCLK pin shares the same identical role as the SCL (again just different labels for a different interfacing bus).

donniebergh, 2021-01-15:
> @donniebergh <https://github.com/donniebergh> I can confirm that there must've been something wrong with your wiring/code using the I2C to the SSD1306 driven display. I recently received my displays, and I encountered no problems with the I2C OLED display + SPI nRF24 radio. More to come when I have a try at the SPI TFT display driven by the ST7789...

I too can confirm the I2C / SSD1306 issues was indeed my problem. I rebuild the prototype using the same parts - same problem. When I loaded some of the example display test sketches - the Arduino would begin to fail within a few moments. I swapped the Arduino for another and the issue was resolved. That Arduino Nano is now in the trash can. 

I revisited the SPI display issue using a different Nano (Arduino Nano RF) and have the same issue. The stock example sketches work as designed with no issues.

I look forward to your finding.

Thank you,
-Donnie

2bndy5, 2021-01-16:
@donniebergh The hardware driven SPI pins on the ATMega328(p) (the CPU used on the Uno and the Nano) are D11, D12, D13 (that's the MOSI, MISO, and SCK/SCLK respectively). [image]
According to this schematic from the [PDF provided by emakefun about the RF Nano](https://github.com/emakefun/emakefun-nano-plus/blob/master/RF-Nano/RF-NANO%E4%BD%BF%E7%94%A8%E8%AF%B4%E6%98%8E%E4%B9%A6_V.1.2.pdf), the nRF24L01's SPI bus is using the ATMega328's hardware driven SPI bus. [One of your SPI displays (the bigger one)](https://www.amazon.com/gp/product/B08FB77YY2/) does not provide access to the ST7789's CS pin. This means that you can't connect the display (without the CS pin) to the same SPI pins that the nRF24L01 is using (please re-[read my previous comment for more explanation about the CS pin](https://github.com/nRF24/RF24/issues/719#issuecomment-759154963)).

## Simply said
1. When using the RF Nano, the display without a CS pin has to use software driven SPI via pins that **are not** D9, D10, D11, D12 and D13.
2. If you use a regular Nano (with an external nRF24L01 radio), then the nRF24L01 can be connected using some software driven SPI pins (not the hardware driven SPI pins), and the display without a CS pin can be connected to the hardware driven SPI pins.
3. If using a display with a CS pin, then both the display and the radio can use the same hardware driven SPI pins, **but make sure that the CS pin for the display is not connected to the same pin that the CSN pin for the nRF24L01 is connected to**.

### Moving on
I'm going to try and use the Adafruit library for the ST7789 (& ST7735) for my TFT display because I can use the same Adafruit GFX library to draw the same graphics/text that I did on the I2C OLED display.

donniebergh, 2021-01-16:
Thank You for the explanation of the smaller display and the missing CE pin. I’ve been concerned about it working anyway and clearly it is not going to share on the SPI bus very well.

Also the schematic helped enormously. The schematic I have appears to have issues (see below in yellow). It appears the CE and CSN pins are not connected to the correct pins and are backwards. That being said I moved a couple of wires on the the larger displays project, changed a few pin definitions in the code and it is working!

Again I thank you - all of my issues are resolved or explained.

-Donnie

Pin changes:

#define TFT_RST   5     // ST7789 Reset  (8)
#define TFT_DC    6     // Data/Command  (9)
#define TFT_CS    10    // CS (CE)       (10)
#define TFT_MOSI  11    // SPI data pin
#define TFT_SCLK  13    // SPI sclk pin

Schematic:

closed 2021-01-16 by 2bndy5