nRF24/RF24 #720: ESP32 and RF24 settings not changing[Question]

Mungy123, 2021-01-24:
Hi wonder if somebody could help with this - using the RF24 library with Nano's works great (Top Job)
settings all change fine in the setup and comms is good.

following wiring examples in the blog i have tried using the same code and get some "wild" configuration back from the NRF24
if using "PrintPrettyLittleDetails" can anybody shed any light on this (RF24 Library is uptodate)
[image]
[image]

Regards
Andy

2bndy5, 2021-01-24:
When you doubt what `printPrettyDetails()` is saying, you should try the old `printDetails()`. Seeing that your addresses all return `0xff` (and all other settings show a value that translates to `0xff`),  my guess is that the wiring is somehow wrong. You should also see `radio.begin()` returns `false` if your wiring is bad.

What esp32 board are you using?

Mungy123, 2021-01-24:
2bndy5  - thanks for the tips, tuned out to be a bad module - all is now good
thank you for your support

regards
Andy

closed 2021-01-24 by Mungy123