nRF24/RF24 #722: [Question] How to use ESP32 VSPI

szerwi, 2021-01-25:
Hi,

Is it possible to use ESP32's VSPI instead of HSPI to communicate with nRF24L01? I would like to use pins 12, 13, 14 insead of 19, 23, 18.

TMRh20, 2021-01-26:
It should be  possible but not currently supported. Requires a few modifications.

szerwi, 2021-01-26:
I found the solution. I simply added `SPI.begin(14, 12, 13, 15);` before `radio.begin();`. Unfortunately, I have no idea if `SPI.begin` changed VSPI to HSPI or it just changed VSPI's pins. May it cause any problems or can I use it without any worry?

2bndy5, 2021-01-26:
Remember that the library instantiates the default SPI bus [on this line](https://github.com/nRF24/RF24/blob/0281ea2048f5295d382257b51980f0004a0f4d1f/RF24.cpp#L715). Not sure if that helps, but you could check if the radio is working by seeing what `radio.begin()` returns. Checking the configuration with `radio.printDetails()` is another more finite debugging option.

I've been toying with the idea of making the c'tor take a reference to an SPI object... Haven't reached any conclusion due to all the platforms that this library supports.

mikechy2005, 2021-01-27:
I did small modification that can use HSPI.
Create a SPIClass pointer in sketch(SPIClass* hspi = new SPIClass(HSPI); Then pass it to RF24 constructer(RF24(hspi, cePin, csnPin);)

RF24.h
Constructor
RF24(SPIClass* spi, ...):ptrSPI(spi), ...
Private
Add SPIClass* ptrSPI = NULL;

RF24.cpp
Replace all _SPI. With ptrSPI->

Works. Hope can help you.

szerwi, 2021-01-27:
Is the modification in the library really neccessary? My solution (adding `SPI.begin(14, 12, 13, 15);` before `radio.begin()`) works well too :)

2bndy5, 2021-01-27:
@mikechy2005 is this code on your fork or is it private? Your modifications are exactly what I was thinking. TBH, I think this idea could use a separate issue/discussion.

@szerwi its possible your solution works without modifying the library because of the SPI implementation for the esp32 core (namely treating the `SPi` object as a singleton).

mikechy2005, 2021-01-28:
> Is the modification in the library really neccessary? My solution (adding `SPI.begin(14, 12, 13, 15);` before `radio.begin()`) works well too :)

Obversely, your code also ok.
In my way, you could choose VSPI/HSPI in sketch level,not library level. Depends on your projects.:)

mikechy2005, 2021-01-28:
> @mikechy2005 is this code on your fork or is it private? Your modifications are exactly what I was thinking. TBH, I think this idea could use a separate issue/discussion.
>
> @szerwi its possible your [...]

Just private :)
Because I used SPI pointer in my other library, so modify this too and share to here:)

ex-caliper, 2021-02-10:
@szerwi
from your earlier comment, "I found the solution. I simply added SPI.begin(14, 12, 13, 15);"
I am assuming these numbers are MOSI, MISO, SCK and SS arguments.  I have tried relating this to the documentation and I am unfamiliar with the ESP32, so please could you put them in order. i.e. which one of the four does 14 represent.   

Many thanks..

szerwi, 2021-02-10:
@ex-caliper have a look at SPI.cpp: https://github.com/espressif/arduino-esp32/blob/6b0114366baf986c155e8173ab7c22bc0c5fcedc/libraries/SPI/src/SPI.cpp#L37
The correct order is: SCK, MISO, MOSI, SS

mauricio-linhares, 2021-02-13:
@mikechy2005 
I am working in a project that requires 2 SPI buses like yours. I intend to use the nRF24l01 in the HSPI, but can't manage to make the changes you suggested work. I modified the RF24.h and RF24.cpp files, but it fails to compile. If you could detail it more, it would help a lot.

mikechy2005, 2021-02-13:
> @mikechy2005
> I am working in a project that requires 2 SPI buses like yours. I intend to use the nRF24l01 in the HSPI, but can't manage to make the changes you suggested work. I modified the RF24.h and RF24.cpp files, but it fails to compile. If you could detail it more, it would help a lot.

@mauricio-linhares 
First of all, i did it in OOP way. 

1. I remove Line 113-124 and add new 1 line SPIClass* ptrSPI; in RF24.h
2. Add replace line 180 (constructor) with RF24(SPIClass* spi, uint16_t _cepin, uint16_t _cspin, uint32_t _spispeed = RF24_SPI_SPEED); in RF24.h
3. Add ptrSPI(spi), before ce_pin(_cepin) in line 424 in RF24.cpp
4. Find all "_SPI." and replace with  "ptrSPI->" in RF24.cpp
5. Finally create HSPI pointer (SPIClass* _hspi = SPIClass(HSPI); ) in sketch and init RF24 with (_hspi, cePin.......

I create all object in object pointer since i always wrote my project by OOP. You may follow OOP concept to try my code.
Hope can help you.

mauricio-linhares, 2021-02-13:
@mikechy2005 
It is working perfectly now. Much better to have each SPI device its own dedicated bus. I've been smashing my head for several days trying to make this work, luckily I found this issue open.
You helped a lot! Thank you!!

gfvalvo, 2021-02-20:
> I've been toying with the idea of making the `begin()` take in a reference to an SPI object... Haven't reached any conclusion due to all the platforms that this library supports.

Or a pointer ..... that would be great!!! I was thinking about hacking it in myself.

2bndy5, 2021-03-20:
I'm working on this feature to use a secondary SPI bus by overloading `begin()`... I think its ready for testing (need testers). The [overload-begin branch](https://github.com/nRF24/RF24/tree/overload-begin) has the changes. I've also added [example source code specific for the ESP32](https://github.com/nRF24/RF24/blob/overload-begin/docs/arduino.md#esp32-example).

makerMcl, 2021-03-27:
Just for report/feedback: I tried workaround from @szerwi with SPI.begin() before radio.begin() -> result was kernel panic to illegal write at 0x400d6828: SFE_BMP180::writeBytes(unsigned char*, char) .pio\libdeps\waterSensorCOM\BMP180/SFE_BMP180.cpp:207
This configuration also uses I2C - I did not expect this side effect but there was no other change.
After removing that SPI.begin()-hack my sketch runs without kernel panics as before.
I will now test that solution in the branch before starting my own fork...

makerMcl, 2021-03-27:
I have tested the solution in that branch now, I can confirm it works! 
Good work, I like the OOP approach (it's my world too).
Tested with a ESP-32 DevKit v1, with some interfaces (I2C, Onewire, UART0, AsyncTCP) in parallel, without any problems.
Used the pattern from the example but referenced inside a class instance (instead of static/main loop).
Do you need more docs?

I would really appreciate it if you can merge that branch so I can get back to get the lib via platformio registry...

szerwi, 2021-03-27:
@makerMcl hmm, that's strange. I successfully used nRF24L01 together with SC16IS752 expander over I2C.
Here is part of my setup:
```C++
void setup()
{
  Serial.begin(9600);

  Wire.begin(4, 16);
  i2cuart.beginA(SC16IS752_DEFAULT_SPEED);

  SPI.begin(14, 12, 13, 15); //ESP32 - zmiana VSPI na HSPI
  radio.begin();
  radio.setChannel(100);
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_MAX);
  radio.openReadingPipe(1, masterPipe);
  radio.openWritingPipe(slavePipe);
  transmit = 0;
  radio.startListening();
  radio.printDetails();
}
```
Maybe that's because I use I2C on different pins (4 and 16).

@2bndy5 when do you plan to implement it into stable release?

2bndy5, 2021-03-27:
@szerwi my branch needs testing. It covers more than just esp32. I don't own a teensy board that could test my solution. 

To be clear, my solution is not the same as @szerwi solution. @szerwi your solution breaks features that others may depend on

@makerMcl thank you so much for validation. If you want you can post you code, so others (not just me) can follow your lead. Although, the example snippet I added the the docs is meant to be as simple and straight forward as possible (to avoid confusion).

Also I think pio has a mechanism to use a specific lib folder as a dependency in your project's platformio.ini. we are aware that v1.3.12 is not showing in pio registry

szerwi, 2021-03-27:
@2bndy5 what exactly can my solution break?
Currently I am just testing if all components are working for bigger project.
I guess I will use your solution in the final code :D

2bndy5, 2021-03-27:
@szerwi I'm sorry. I was confusing your solution with @mikechy2005 solution. @mikechy2005 solution breaks features.

@szerwi solution will work _better_ with my branch because my branch actually allows the user to call `SPI class::begin()` separately from `radio.begin()`. Additionally, I think my solution actually supports espressif's ArduinoCore's soft SPI implementation.

closed 2021-04-11 by 2bndy5

makerMcl, 2021-04-11:
@2bndy5 fyi - Regarding the code I used - it is the same as in your example;
```
    void init(const byte pinCe, const byte pinCsn)
    {
        hspi = new SPIClass(HSPI); // by default VSPI is used
        hspi->begin();
        radio = new RF24(pinCe, pinCsn);
        if (radio->begin(hspi))
        {
            // AutoACK is on by default
            radio->setChannel(67);
            radio->setPALevel(RF24_PA_MIN);
            radio->enableDynamicPayloads();
            radio->openReadingPipe(0, receiveAddress);
            radio->openWritingPipe(sendAddress);
            radio->startListening();
            ui.logInfo() << "radio initialized (chipConnected=" << radio->isChipConnected() << ") with datarate=" << radio->getDataRate() << ", PAlvl=" << radio->getPALevel() << ", crcLength=" << radio->getCRCLength() << endl;
        }
        else
        {
            ui.logError() << "RF24 radio: chip is not responding" << endl;
        }
    }
```

2bndy5, 2021-04-11:
@makerMcl good show! What logging lib are you using? (I assume it works on ESP32)

I'm curious about the log lib you're calling to (the `ui.log*()`) because we're looking to unify the RF24* libs' stdout usage (which currently employs `printf()`) on all supported platforms.

makerMcl, 2021-05-02:
Sorry for the late response. It is something self-built, I have out-factored some common tasks there and for logging it's main benefit is a rolling buffer for log-output via webserver. You can find it on https://github.com/makerMcl/universalUi, sample usage in https://github.com/makerMcl/calibrationServer.

If you are thinking about logging, I can only share my humble experience:  for now I found the most simple solution (KISS principle) with macros, see line 90-101 of https://github.com/makerMcl/universalUi/blob/master/hc12tool.h for it.
There are some logging frameworks, and also ESP - for my needs it was too many bytes used for program code. For NRF24 lib requirements might differ.

2bndy5, 2021-05-02:
Thanks for the lead! Yes our RF24Log lib will use a singleton with macros API as well. But, extending it to all possible supported platforms (x86 included), while implementing our own `printf()` specifiers & log level filtering, for use with all possible output streams... Well, that's what's taking up all the program space. We hope to encompass more use cases than just RF24 libs... LED matrices, file output, multiple output streams, ncurses, etc...

kshitijgarg2609, 2021-05-17:
SPIClass *sp;
RF24 radio(2,15);

void setup()
{
  Serial.begin(115200);
  initRF();
}

void initRF()
{
  sp = new SPIClass(HSPI);
  sp->begin();
  if(radio.begin(sp))
  {
    Serial.println("Radio Started !!!");
    radio.openWritingPipe(addr2);
    radio.openReadingPipe(1, addr1);
    radio.startListening();
  }
  else
  {
    Serial.println("Radio couldn't start !!!");
  }
}

I have been trying to use HSPI of ESP32 but this library is also not working with VSPI of ESP32. Ihave shared the snippet of the code I have been trying. 

I connected pins :- 
nrf          esp32
mosi          13
miso          12
sck          14
csn          15
ce          2

2bndy5, 2021-05-17:
@kshitijgarg2609 what exact board are you using (ESP32 is a chip used on several MCU boards)? The fact that `VSPI` doesn't work for you (assuming you use the normal `RF24::begin()` without passing a parameter to use `VSPI`) tells me you may be facing a wiring problem. 

Try changing the wires (shortest wires possible) and make sure the connections are tight. Please note that sometimes the MCU boards' have very high impedance on the SPI bus pins used which makes long wires actual corrupt the data going to the radio module (thus `begin()` will return false).

If finding short wires is a problem, try lowering the SPI frequency by specifying it in the c'tor:
```cpp
RF24 radio(2, 15, 4000000); // uses 4 MHz instead of default 10 MHz
```

kshitijgarg2609, 2021-05-17:
I have resolved myself, I was going according to the documentation and trying HSPI on ESP32 CAM, RF24(_SPI) doesn't work but the SPI.begin(SCK,MISO,MOSI,SS); worked.

Thank you

Frenzyritz13, 2021-08-13:
Hey, I have been trying to use SPI.begin to run the HSPI bus with my nrf24l01+ module but I am unabl to do  so. I am also using i2c on pins 21 and 22. My module's CE pin is being handled by the MCP27013 GPIO expander
Any suggestions on how I should go about troubleshooting this?

2bndy5, 2021-08-13:
@Frenzyritz13 Have you tried following [the docs](https://github.com/nRF24/RF24/blob/master/docs/arduino.md#esp32-example)? I only ask because you didn't mention it.

kshitijgarg2609, 2021-08-13:
@Frenzyritz13 I had also tried HSPI but later on I used VSPI with variable pins. For using different pins of VSPI you have to call SPI.begin(sck_pin,miso_pin,mosi_pin,ss_pin) before RF24.begin(). HSPI is needed when you are using some other SPI device along with rf24 device. One unique feature of esp32 is that pins of i2c and spi can be mapped to any pins.

https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h this link contains the class and functions of SPI ESP32.

2bndy5, 2021-08-13:
Just to clarify, mapping the SPI bus or I2C bus to any other pins is specific to the ESP32 Arduino core (not the hardware). Technically the Arduino core invokes bit-banged (software-driven) solutions when manually mapping SPI/I2C pins to GPIO pins that weren't designed for hardware-driven serialized buses. The core's docs neglect to adequately warn users that these bit-banged solutions are much slower than using the default hardware capable pins.

kshitijgarg2609, 2021-08-13:
@2bndy5 can you send me that docs or link where it is written that pin mapping of esp32 is slow. Because I have experimented esp32 cam streaming over rf24 which has less pins and vspi is occupied by the sd card reader, so I just have to map spi pins, now on the receiever side I have used esp8266 and rf24 which receives data frame from rf24 of esp32 and combines packet as the mtu size is just 32 bytes, and sends over to wifi.

The result I found is that I achieved 5-6 fps on the resolution of QQVGA which is just enough and not that much real time.

kshitijgarg2609, 2021-08-13:
[image]
I am facing this issue that SPIClass pointer or function I am passing in 
[image]
gets compiled but doesn't work.

2bndy5, 2021-08-13:
@kshitijgarg2609 The docs are seriously lacking about the SPI interface, but I managed to find [this table](https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/io_mux.html#peripherals) that hints to my clarification.

> gets compiled but doesn't work.

This very vague. How is it not working? Please don't post pictures of code; I can't copy-n-paste adjustments if its a picture.

kshitijgarg2609, 2021-08-13:
@2bndy5 I will write this code again, need some time and send here. The pictures of code is from specific functions of .h files given on https://github.com/nRF24/RF24/blob/master/RF24.h and https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h

2bndy5, 2021-08-13:
I don't see anything wrong with this code. The only difference I see is that you passed the SPI_SS pin to the RF24 c'tor's CSN parameter (this is not wrong). Although, the RF24 lib assumes it can handle the CSN pin independently from the other SPI pins. 

It is common for SPI slave device drivers to handle the CSN pin independently. In fact the SPI_SS pin is usually still manipulated by the Arduino core while the actual CSN pin is manipulated by the RF24 lib. This better suites the cross-platform compatibility we hold dearly.

Frenzyritz13, 2021-08-16:
> @Frenzyritz13 I had also tried HSPI but later on I used VSPI with variable pins. For using different pins of VSPI you have to call SPI.begin(sck_pin,miso_pin,mosi_pin,ss_pin) before RF24.begin(). [...]

Hey, this makes sense, I will try this.

Frenzyritz13, 2021-08-16:
> Hey, this makes sense, I will try this.

Update: I tried remapping the VSPI pins but couldn't do so. I used HSPI and it seemed to start the communication between my 2 esps. 
Here is the code I ended up using:
```
SPIClass * hspi = NULL; //Before void setup

//In void setup==>

 hspi = new SPIClass(HSPI); // by default VSPI is used
  hspi->begin();
  if (!radio.begin(hspi)) {
    Serial.println(F("radio hardware is not responding!!"));
    while (1) {} // hold in infinite loop
  }
```

This code runs successfully with the MCP23017 expander. I called the adafruit library in the RF24 library itself. Here is the modification I have made to the library.==>

If anyone has better ways of going about doing this, do le me know.

```
#include <Adafruit_MCP23X17.h>
Adafruit_MCP23X17 mcp;

void RF24::ce(bool level)
{
    bool checkBegin;

    //Allow for 3-pin use on ATTiny
    checkBegin = mcp.begin_I2C();

    if (ce_pin != csn_pin)
    {
        if (!checkBegin)
        {
            digitalWrite(ce_pin, level);
            Serial.println("Error.");
        }
        else if (checkBegin)
        {
            mcp.pinMode(4, OUTPUT);
            mcp.digitalWrite(4, level);
        }
    }
}
```

2bndy5, 2021-08-16:
If that's all the modification that was needed to make the MCP23017 work, then you might be able to reduce the compile size with a new macro defined in _RF24_config.h_

```cpp
#else // everything else for Arduino platforms

    // This macro assumes that MCP23017::begin() was called by the application prior to RF24::begin().
    #define USE_MCP23XXX_AS_CE mcp // the global MCP23017 object declared in main application's code space
```
 Then alter the `ce()` function in _RF24.cpp_
```cpp
void RF24::ce(bool level)
{
#if defined USE_MCP23XXX_AS_CE
    USE_MCP23XXX_AS_CE.digitalWrite(4, level);
#else // !defined(USE_MCP23XXX_AS_CE)
    if (ce_pin != csn_pin)
    {
        digitalWrite(ce_pin, level);
    }
#endif // !defined(USE_MCP23XXX_AS_CE)
}
```

Lastly, I still think `_init_pins()` function (in _RF24.cpp_) should be altered to set `pinmode()` for a pin number that's designated for a MCP23017 (modification is toward the bottom and would only apply to Arduino platforms).

```cpp
// This modification assumes you passed the MCP23017 pin number to the RF24 constructor's _ce_pin parameter

bool RF24::_init_pins()
{
    if (!isValid()) {
        // didn't specify the CSN & CE pins to c'tor nor begin()
        return false;
    }

    #if defined (RF24_LINUX)

        #if defined (MRAA)
    GPIO();
    gpio.begin(ce_pin, csn_pin);
        #endif

    pinMode(ce_pin, OUTPUT);
    ce(LOW);
    delay(100);

    #elif defined (LITTLEWIRE)
    pinMode(csn_pin, OUTPUT);
    csn(HIGH);

    #elif defined (XMEGA_D3)
    if (ce_pin != csn_pin) {
        pinMode(ce_pin, OUTPUT);
    };
    ce(LOW);
    csn(HIGH);
    delay(200);

    #else // using an Arduino platform

    // Initialize pins
    if (ce_pin != csn_pin) {
        #if !defined (USE_MCP23XXX_AS_CE)
        pinMode(ce_pin, OUTPUT);
        #else // USE_MCP23XXX_AS_CE is defined
        USE_MCP23XXX_AS_CE.pinMode(csn_pin, OUTPUT);
        #endif // defined(USE_MCP23XXX_AS_CE)
    }

    ce(LOW);
    csn(HIGH);

        #if defined (__ARDUINO_X86__)
    delay(100);
        #endif
    #endif // !defined(XMEGA_D3) && !defined(LITTLEWIRE) && !defined(RF24_LINUX)

    return true; // assuming pins are connected properly
}
```

Please note that my code suggestions are contribution compatible. So, if you find that this solution works, then you can make a Pull Request to add the new featured macro for others to benefit from your hard work/frustration.

Frenzyritz13, 2021-08-16:
> If that's all the modification that was needed to make the MCP23017 work, then you might be able to reduce the compile size with a new macro defined in _RF24_config.h_
> 
> ```c++
> #else // everything else for Arduino platforms
> 
>     // This macro assumes that MCP23017::begin() was called by the application prior to RF24::begin().
>     #define USE_MCP23XXX_AS_CE mcp // the global MCP23017 object declared in main application's code space
> ```
> 
> Then alter the `ce()` function in _RF24.cpp_
> 
> ```c++
> void RF24::ce(bool level)
> {
> #if defined USE_MCP23XXX_AS_CE
>     USE_MCP23XXX_AS_CE.digitalWrite(4, level);
> #else // !defined(USE_MCP23XXX_AS_CE)
>     if (ce_pin != csn_pin)
>     {
>         digitalWrite(ce_pin, level);
>     }
> #endif // !defined(USE_MCP23XXX_AS_CE)
> }
> ```
> 
> Lastly, I still think `_init_pins()` function (in _RF24.cpp_) should be altered to set `pinmode()` for a pin number that's designated for a MCP23017 (modification is toward the bottom and would only apply to Arduino platforms).
> 
> ```c++
> // This modification assumes you passed the MCP23017 pin number to the RF24 constructor's _ce_pin parameter
> 
> bool RF24::_init_pins()
> {
>     if (!isValid()) {
>         // didn't specify the CSN & CE pins to c'tor nor begin()
>         return false;
>     }
> 
>     #if defined (RF24_LINUX)
> 
>         #if defined (MRAA)
>     GPIO();
>     gpio.begin(ce_pin, csn_pin);
>         #endif
> 
>     pinMode(ce_pin, OUTPUT);
>     ce(LOW);
>     delay(100);
> 
>     #elif defined (LITTLEWIRE)
>     pinMode(csn_pin, OUTPUT);
>     csn(HIGH);
> 
>     #elif defined (XMEGA_D3)
>     if (ce_pin != csn_pin) {
>         pinMode(ce_pin, OUTPUT);
>     };
>     ce(LOW);
>     csn(HIGH);
>     delay(200);
> 
>     #else // using an Arduino platform
> 
>     // Initialize pins
>     if (ce_pin != csn_pin) {
>         #if !defined (USE_MCP23XXX_AS_CE)
>         pinMode(ce_pin, OUTPUT);
>         #else // USE_MCP23XXX_AS_CE is defined
>         USE_MCP23XXX_AS_CE.pinMode(csn_pin, OUTPUT);
>         #endif // defined(USE_MCP23XXX_AS_CE)
>     }
> 
>     ce(LOW);
>     csn(HIGH);
> 
>         #if defined (__ARDUINO_X86__)
>     delay(100);
>         #endif
>     #endif // !defined(XMEGA_D3) && !defined(LITTLEWIRE) && !defined(RF24_LINUX)
> 
>     return true; // assuming pins are connected properly
> }
> ```
> 
> Please note that my code suggestions are contribution compatible. So, if you find that this solution works, then you can make a Pull Request to add the new featured macro for others to benefit from your hard work/frustration.

This looks good Brendon! Will test it and get back to you. Also, I am going to be using 2 I2Cs and an Expander based interrupt, I am expecting some issues there as well, will keep this thread updated.

2bndy5, 2021-08-16:
@Frenzyritz13 I just modified my post (found/fixed an error in the logic). Please be sure to use the updated snippet about `_ini_pins()`