nRF24/RF24 #725: [Question] How to use Digispark Attiny85?

rogerfernandes, 2021-01-27:
It's possible to use NRF24L01+ with Digispark Attiny85? I've tried the example [rf24ping85.ino](https://github.com/nRF24/RF24/blob/master/examples/rf24_ATTiny/rf24ping85/rf24ping85.ino) but It's not compiling.

I uncommented the define of CSN_PIN to use pin 3.

This is the compiler log: [compiling-error.log](https://github.com/nRF24/RF24/files/5881423/compiling-error.log)

2bndy5, 2021-01-27:
The official Digispark core from digistump has not been maintained for a few years now. The RF24 library now depends on SpenceKonde's ATTinyCore which includes Digispark support now (a rather recent addition). You should read the comments at the top of rf24ping85.ino; the necessary links are all right there.

rogerfernandes, 2021-01-29:
Thanks @2bndy5! I'll read the comments that you mentioned. And close this issue ;)

closed 2021-01-29 by rogerfernandes