nRF24/RF24 #726: [Question] RF24Mesh for MQTT

TaixMiguel, 2021-02-07:
Good afternoon, since I got the ["Breakout Adapter On-board 3.3V Regulator"](https:///images-na.ssl-images-amazon.com/images/I/61qxk%2BC3XBL._SX342_.jpg) and I was able to make the first connection between a RPi and an ESP32 by means of nrf24 I do nothing but think about going deeper into this path, but for more information I read I can not solve several doubts.

- I do not understand what library RF24, or combination, I need to implement (RF24Mesh, NRF24Network, NRF24Ethernet), my idea is to make a mesh network like Zigbee and that the communication is by MQTT. I have seen the examples of [MQTT in RF24Ethernet](https:///github.com/nRF24/RF24Ethernet/tree/master/examples/MQTT) but I have unresolved doubts... Would there be any kind of security on the network? I mean, could someone from outside connect to this network?(to avoid it) Security would only be at the MQTT level?
- All RPi/ESP32 must have [nRF24L01+PA+LNA](https:///images-na.ssl-images-amazon.com/images/I/61uzOedjg0L._AC_SY355_.jpg) or can be inserted with [nRF24L01](https://www.tiendatec.es/4840-large_default/transceptor-nrf24l01-24ghz.jpg). That is, to have in the RPi the nRF24L01+PA+LNA and in the rest of ESP32 of the nRF24L01 network. Would it work or would they all have to be of the same type?
- I have also read in the RF24Network documentation that you can implement a "sleep mode using interrupts", but it is only in "AVR devices". I tried looking for it but I couldn’t find it, could the ESP32 use this mode? I mean, is it an "AVR device"? Thank you in advance for the libraries and the support you give them.

Thank you in advance for the libraries and the support you give them.

TMRh20, 2021-02-07:
> I do not understand what library RF24, or combination, I need

You would need the RF24Ethernet and MQTT libraries as per the RF24Ethernet examples you linked.

> Would there be any kind of security on the network? I mean, could someone from outside connect to this network?(to avoid it) Security would only be at the MQTT level?

No. Yes. Yes.

> That is, to have in the RPi the nRF24L01+PA+LNA and in the rest of ESP32 of the nRF24L01 network. Would it work or would they all have to be of the same type?

They don't have to be the same type, but the PA+LNA versions typically require a separate power supply due to the large power consumption, so it is a bit of a pain.

> I have also read in the RF24Network documentation that you can implement a "sleep mode using interrupts", but it is only in "AVR devices".

AVR device refers to 8-bit Atmel Microcontrollers.   ESP32 is a different beast. The sleep mode simply lets the MCU go into low power mode and uses the interrupt pin of the radio to wake it up on RX. This can be implemented with ESP32 low power modes, but is not supported.

TaixMiguel, 2021-02-07:
Thank you very much @TMRh20.

I think I have the last questions:
-  The IP address of the server, can I invent it or does it have to be a specific one that matches a contracted Internet service or something similar?
- Do I have to define a specific IP address for each node? Don't have a DHCP service that gives me an IP dynamically?
- I have tried to search for ESP32 low power modes but I cannot find how I can leave the RX active so that it wakes up when it receives information about a topic that I have subscribed to, could you indicate a link where I can consult it?

Again, thank you very much.

TMRh20, 2021-02-08:
1. The IP can be any private IP/IP range for the mesh.
2. Yes. In RF24Mesh, 'dhcp' type stuff refers to dynamic RF24Network address. The nodeID (IP) is static.
3. You need to use interrupts and a low power mode for ESP32, I'm not too familiar so you're on your own.

TaixMiguel, 2021-02-12:
Ok, thank you @TMRh20

I close the issue

closed 2021-02-12 by TaixMiguel