nRF24/RF24 #728: radio.available()

zKingFabzz, 2021-02-10:
[image]
I am trying to use this library with a ATmega32. Sending data is working well, but receiving data is not very well. One problem is that the function .available() is not working and .read(...) sometimes works and sometimes not.

Yours sincerely
zKingFabzz

TMRh20, 2021-02-10:
It should be no different than running on any ATmega chip. Since you have it working partly, I'm gonna guess this would be hardware &/or power supply related. There is not much we can do library-wise to make it work, if you are not receiving anything, available will not return true. I'm not sure what you mean that read sometimes works and sometimes not... if available() returns true, there has to be something to read. If not, again, I would suggest hardware issues.

Beyond a solid power supply, (batteries work best for testing) filter capacitors are recommended as well. (10-100uF from vcc-gnd of radio device)

2bndy5, 2021-02-10:
If this `switch()` is called inside an Interrupt Request handler, then there's a chance that `available()` would return `false` when the IRQ pin is still transitioning HIGH to LOW (that's what I found during testing the IRQ example). Example: say this is the first code called and `first_f` was `false`, `available()` might be getting called too quickly. But, this is most likely to happen when called inside an ISR callback function. I added some [notes about this behavior to the docs for `available()`](https://nrf24.github.io/RF24/classRF24.html#a127105eb7a3b351cfe777c1cec50627a)

zKingFabzz, 2021-02-10:
Thank you for trying your best, but I distanced me to all Interrupts, that means that the switch is placed at the beginning of loop section. I am trying to communicate an ATmega32 with an Arduino Nano,  where the ATmega32 will answer, if he gets a request. I also tried it with an Arduino Uno instead of the ATmega32, where that was not a problem.
You mean available() is called to quickly??

2bndy5, 2021-02-10:
>there's a chance that `available()` would return false when the IRQ pin is still transitioning HIGH to LOW.

Its in the datasheet and in the docs. Instead of a timing problem, your problem sounds more like a hardware problem if you don't use `attachInterrupt()` in your `setup()`.
1. Have tried running the Nano code on the Uno that works?
2. Are you using an adapter or PA+LNA radio module?

zKingFabzz, 2021-02-10:
The nano code on the uno doesn't work :(
And i use the "normal" nRF24L01 modul.

zKingFabzz, 2021-02-10:
I think it is so weird that writing is possible without problems but available and reading is not working...

2bndy5, 2021-02-11:
Library works fine we just have to figure out why it won't work with your device.
- Have you considered [TMRh20 comment about power supply](https://github.com/nRF24/RF24/issues/728#issuecomment-776989711)?
- Also, did you read through the [COMMON_ISSUES.md file](https://nrf24.github.io/RF24/md_COMMON_ISSUES.html) (especially the parts about matching settings and using a capacitor)?

zKingFabzz, 2021-02-11:
It is interesting, because I added a 10µF capacitor and it works maybe all 10 to 20 seconds one time.  What do you mean, what my next step is? :(

2bndy5, 2021-02-11:
Mayb, try a larger capacitor. I usually use 100uF

closed 2021-03-31 by 2bndy5