nRF24/RF24 #729: Works fine with STM32 on platform.io but not listed as supporting STM32 in the platform.io library repo tool

chrisco484, 2021-02-13:
I tried a few different RF24 libraries and this was the only one I could get working on a Blue Pill STM32F103x.

I selected this library even though the platform.io library manager did not have STM32 listed in this libraries set of supported boards - and yet it works wonderfully well.

How would we get platform.io to list STM32 as supported by this library??

2bndy5, 2021-02-13:
Great news!

I believe that supported platforms are listed in [the library.json file](https://github.com/nRF24/RF24/blob/0281ea2048f5295d382257b51980f0004a0f4d1f/library.json#L11).  Best to [read the PlatformIO docs](https://docs.platformio.org/en/latest/librarymanager/config.html#platforms) to be sure. For the Arduino platform, this library assumes all boards are supported using [`*` (in library.properties file)](https://github.com/nRF24/RF24/blob/0281ea2048f5295d382257b51980f0004a0f4d1f/library.properties#L9). I don't see why this assumption wasn't made for platformIO.

TMRh20, 2021-02-13:
Yeah the Pio docs list the different platforms somewhere, it’s just a matter of adding to the json file. Easy enough we can do this right away but won’t affect Pio until we do a release I believe.

2bndy5, 2021-02-19:
closing this as it has been addressed by TMRh20's above commit. @chrisco484 thanks again for bringing this to our attention 👍🏻

closed 2021-02-19 by 2bndy5