nRF24/RF24 #730: Call stopListening() Before openWritingPipe()

gfvalvo, 2021-02-19:
The documentation instructs to call stopListening() before calling openWritingPipe().

Makes sense. But, is it necessary if the radio is only used as a Master Transmitter to send commands (and receive ACK Payloads) from multiple slaves? Meaning, must I call stopListening() every time before I change the slave address with openWritingPipe()?

This Master Transmitter code never calls startListening().

Thanks.

2bndy5, 2021-02-19:
I think you should be ok. The docs say that because of the way the library has to manage the pipe 0 address (and RX state for ACK packets). If your code never calls `startListening()`, then you shouldn't have any problems. Are you asking as part of troubleshooting a problem?

gfvalvo, 2021-02-19:
Thanks for the reply. No real problem. I have a working project and updated RF24 to v1.3.11 from v1.3.3. I knew from a couple threads in the Arduino forum that this might be an issue. Turns out that I do have to call stopListening() **ONCE** …. right after I configure the radio. It's not needed again after that.

I'm closing this issue.

closed 2021-02-19 by gfvalvo