nRF24/RF24 #732: hard code role in MulticeiverDemo.ino as a node does not transmit

i164j9, 2021-02-19:
when the UN-modified MulticeiverDemo.ino for the arduino when loaded onto two key wish rf-nano's everything works as expected.
however when hard coding the role to any value other than R ie: 0 - 5 it does not transmit. the nano's are based on the atmega328p using the new bootloader.

2bndy5, 2021-02-19:
We need to see exactly what code you changed/modified.

i164j9, 2021-02-19:
lines 21, 41, 61
/*
* See documentation at https://nRF24.github.io/RF24
* See License information at root directory of this library
* Author: Brendan Doherty 2bndy5
*/

/**
* A simple example of sending data from as many as 6 nRF24L01 transceivers to
* 1 receiving transceiver. This technique is trademarked by
* Nordic Semiconductors as "MultiCeiver".
*
* This example was written to be used on up to 6 devices acting as TX nodes &
* only 1 device acting as the RX node (that's a maximum of 7 devices).
* Use the Serial Monitor to change each node's behavior.
*/
#include <SPI.h>
#include "printf.h"
#include "RF24.h"

// instantiate an object for the nRF24L01 transceiver
RF24 radio(10, 9); // using pin 7 for the CE pin, and pin 8 for the CSN pin

// For this example, we'll be using 6 addresses; 1 for each TX node
// It is very helpful to think of an address as a path instead of as
// an identifying device destination
// Notice that the last byte is the only byte that changes in the last 5
// addresses. This is a limitation of the nRF24L01 transceiver for pipes 2-5
// because they use the same first 4 bytes from pipe 1.
uint64_t address[6] = {0x7878787878LL,
0xB3B4B5B6F1LL,
0xB3B4B5B6CDLL,
0xB3B4B5B6A3LL,
0xB3B4B5B60FLL,
0xB3B4B5B605LL
};

// Because this example allow up to 6 nodes (specified by numbers 0-5) to
// transmit and only 1 node to receive, we will use a negative value in our
// role variable to signify this node is a receiver.
// role variable is used to control whether this node is sending or receiving
char role = '0'; // 0-5 = TX node; any negative number = RX node

// For this example, we'll be using a payload containing
// a node ID number and a single integer number that will be incremented
// on every successful transmission.
// Make a data structure to use as a payload.
struct PayloadStruct
{
unsigned long nodeID;
unsigned long payloadID;
};
PayloadStruct payload;

// This example uses all 6 pipes to receive while TX nodes only use 2 pipes
// To make this easier we'll use a function to manage the addresses, and the
// payload's nodeID
void setRole(); // declare a prototype; definition is found after the loop()

void setup() {

Serial.begin(9600);
while (!Serial) {
// some boards need to wait to ensure access to serial over USB
}

// initialize the transceiver on the SPI bus
if (!radio.begin()) {
Serial.println(F("radio hardware is not responding!!"));
while (1) {} // hold in infinite loop
[... 104 lines cut ...]
// For the RX node

// Set the addresses for all pipes to TX nodes
for (uint8_t i = 0; i < 6; ++i)
radio.openReadingPipe(i, address[i]);

radio.startListening(); // put radio in RX mode

} else {
// For the TX node

// set the payload's nodeID & reset the payload's identifying number
payload.nodeID = role;
payload.payloadID = 0;

// Set the address on pipe 0 to the RX node.
radio.stopListening(); // put radio in TX mode
radio.openWritingPipe(address[role]);

// According to the datasheet, the auto-retry features's delay value should
// be "skewed" to allow the RX node to receive 1 transmission at a time.
// So, use varying delay between retry attempts and 15 (at most) retry attempts
radio.setRetries(((role * 3) % 12) + 3, 15); // maximum value is 15 for both args
}
} // setRole

2bndy5, 2021-02-20:
`role` is not a textual variable (don't let the `char` datatype mislead you). Instead try using the actual integer.
```c++
char role = 0;
```

It helps to draw attention to this line in the `if (Serail.available())` block:
```c++
        else if (c >= 48 && c <= 53 && c != role)
        {
            // Become a TX node with identifier 'c'

            role = c - 48;
            // ...
```

2bndy5, 2021-02-20:
I'm just now realizing that the comment `// 0-5 = TX node; any negative number = RX node` is inaccurate. it should say 
```c++
// integers 0-5 = TX node; character 'R' or integer 82 = RX node
```
This may be a bit of an artifact from when I was developing the example... I should change this comment in the code.

closed 2021-02-22 by i164j9