nRF24/RF24 #736: Are the docs user friendly?

2bndy5, 2021-02-26:
This issue is long overdue, and I've hinted at ideas to improve Doxygen's aged aesthetics (visual theme). I encourage input from anyone with objections/ideas about the current docs even if you're unaware that this library has docs.

## using doxygen's XML output in Sphinx
Several tools exist to utilize doygen's incomparable C++ parser in Sphinx output.
1. `Sphinx` is a 3rd-party python library that builds static docs from RST (reStructuredText) syntax. RST is a syntax used by python's standard library `docutils`.
2. `breathe` is a Sphinx extension that will parse doxygen's XML output into RST-compatible directives; meaning the docs are written in source code comments but referenced/organized in RST files that are specifically used for building the docs.
3. `recommonmark` is another useful Sphinx extension that allows using MD (MarkDown) files instead of RST files (with this extension both file types can be used). This extension is absolutely optional because it tends to introduce some minor problems.
### Cons
Doxygen uses MD tables and `recommonmark` (and `breathe`) does not natively translate MD tables into RST tables (there is a~n easy~ workaround for this). Since `breathe` uses XML output, some XML specific characters are not allowed (like "&", "<", ">"); code snippets within the docs are treated _differently_. Also, admonitions like "Note" and "Warning" paragraphs need tweaking (with `breathe`) to properly grab the reader's attention (otherwise they look like yet another paragraph). However, RST supports way more admonitions than Doxygen offers; namely "Hint", "Tip", "Important", and custom admonitions.
### Pros
Using Sphinx allows maintainers to select from many modern style themes. RST-based docs can be served from readthedocs.org (a VERY common practice) which allows for users to read previous versions of the docs (based on tagged commits or separate branches). There are also 3rd-party linguistic translation services for Sphinx-based docs, but they aren't easy to setup and may require a monetary donation.

2bndy5, 2021-03-15:
I went ahead and abstracted the 15 "Related Pages" (and the docs main page) from the RF24.h on my overload-begin branch (see the new "docs" folder). this will make the platform support pages "readable" via github (relative hyperlinks specific to doxygen will not work in github viewer). 😅 

This change will have no negative affects on doxygen output.

2bndy5, 2021-03-31:
OK. My crusade to abstract docs' "Related Pages" from _**all**_ RF24* libs (or atleast the ones hosted by the nRF24 org). All hyperlinks were checked for "livelihood" in the event that the abstracted docs get deployed.

@Avamander @TMRh20  This means there should be another merge-n-release crusade when #750 is merged and released. If you want, I can bump the version numbers in the applicable PRs to make this easier.

## Ulterior motive
These docs abstraction PRs are phase 1 of augmenting generated docs (on top of doxygen's XML output like stated in OP).

### current choices I'm seriously conidering
- Sphinx + breathe [+ recommonmark] (see OP)
- mkdocs
   - which would require the 3rd party tool (called [doxybook2](https://github.com/matusnovak/doxybook2)) to extract mkdoc formatted md files from doxygen's XML files.
   - This option renders rather ugly (IMHO) concerning the API docs
   - the [doxybook2](https://github.com/matusnovak/doxybook2) is a still a bit "finicky" and requires advanced knowledge/insight about Jinja syntax for templating the generated mkdoc md files.
   - The current abstracted md files should be avoided when used directly in mkdoc because  doxygen adds hyperlinks we'd have to otherwise add ourselves, then doxybook2 adjusts those relative hyperlinks for use in mkdoc md files.

2bndy5, 2021-10-18:
I've been kinda quiet lately on this org because I've been contributing to other projects. Now I'm ready to start using the results from those contributions in this org.

## Get ready for a docs facelift!
Since I last tried, it seems that the `breathe` sphinx extension now works better regarding md tables embedded in the docs. I'm going to use the newly developed (hint-hint) [`sphinx-immaterial` theme](https://jbms.github.io/sphinx-immaterial/) that was ported from the popular mkdocs-material theme. No other extensions are needed since all the _.md_ files in the docs folder are exported through doxygen then imported through `breathe` to sphinx.

Local testing is very promising. Meaning, the docs can still be developed as they always have been. However, since we'd be moving away from doxygen's html output, I think there's a better way to document RF24 portability...

2bndy5, 2021-10-31:
Mesh & network docs are now hosted live at RTD
- https://rf24network.rtfd.io/
- https://rf24mesh.rtfd.io/

closed 2021-12-01 by 2bndy5

2bndy5, 2021-12-01:
[image] 👀 
@TMRh20 when's the last time the issue count was this low? Hopefully, all my work on the docs help keep this count below 25. 🥂

Avamander, 2021-12-01:
> When's the last time the issue count was this low? Hopefully, all my work on the docs help keep this count below 25. :clinking_glasses:

Not in a long long time. It's great, for a while the repo was just kept steady and oiled, but not actively improved like you have.

TMRh20, 2021-12-02:
Yeah, all the recent contributions really add a lot to the library. I couldn’t hope to find better contributors.