nRF24/RF24 #739: Incompatibility with Arduino NANO 33 BLE.

TonioChingon, 2021-02-28:
Hi, 

this is a great library!
There is however some incompatibility with the Arduino Nano 33 BLE (based on mbed OS).

I have the following setup:

1. NRF24L01+ as transmitter with an Arduino Nano 33 BLE Sense
2. NRF24L01+ as receiever with an Arduino Nano 33 BLE
3. NRF24L01+ as receiever with an Arduino Uno

I am not able to receive any data on the Radio 2, transmission works well as data is received in Radio 3.

In order to debug the code, I tried to use **printDetails** and **printf_P**. Unfortunately they did not print any information.

After some research, I was able to make it work by inserting at **line 27** of **printf.h** the following (see **line 44** in https://github.com/arduino/ArduinoCore-mbed/blob/8564c3a97f9a2f74a54f3952ddd9a0516da22add/cores/arduino/macros.h):

```
#elif defined(ARDUINO_ARCH_MBED)
REDIRECT_STDOUT_TO(Serial);
#endif
```
Original code of **printf.h**:
https://github.com/nRF24/RF24/blob/2af44fcac5b0022780a18fddde593359194c634d/printf.h#L20-L28

Unfortunately, I did not figure it out how to insert it inside **printf_begin()**.

I was able to perform the debugging, but with no success, i.e. I am still not able to receive data with Radio 2.
The Radios 2 and 3 seem to behave **exactly** the same during setup commands, however at the time of reading the data, there is no available data in Radio 2. The tests were carried out while Radio 2 and 3 were both on and only either of them was on.

Any thoughts? I will keep trying.

PS. I was debugging library v.1.3.11 but today I made the update to 1.3.12 also with no success (but no debugging).

TMRh20, 2021-02-28:
As far as I know the library is working with the Nano 33, so it would likely be a hardware or coding issue, especially since it is working as transmitter, just not receiving.

Can you show the output of printDetails? Are you using the included examples to test? (same code on radio 2 and 3?)

2bndy5, 2021-02-28:
This is going to sound ironic, but the Nano 33 BLE uses a nRF52840 as its CPU which internally employs a cortex M4 (maybe a SAMD -- going off of memory here) for user-code execution. I'm not too surprised that printf.h isn't working on this platform given #414 , but all that only addresses why you can't get debugging info printed to Serial.

The real problem is why your project isn't working, but it would be easier to address this if we see your code. If posting your code publicly is prohibited, then I would go with TMRh20's suggestion and load up a library example (gettingStarted.ino seems the most applicable) and test only a pair of radios at a time (the examples aren't written for transmitting to 2 RX nodes).

TonioChingon, 2021-03-01:
Hi @TMRh20 and @2bndy5, 

first of all, thanks a lot for you reply and the good news, that the library should work well with the Nano 33 BLE.
There must be a problem on my side.

I have discarted HW problem, as a I switched the NRFL2401+ boards among the Arduinos and the problem persists.
I started with the example "gettingStarted.ino" and then I simplify it to test it.

The Arduino Uno can send/receive with no problem.
But the Arduino Nano 33 BLE seems to only be able to send data.

These are the codes:

**Sender (Arduino Nano 33 BLE)**
```
//SENDER
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include "printf.h"

RF24 radio(9, 10);                // CE, CSN         
const byte address[6] = "fffff";  //Address. This should be same on the receiving side.
bool flag = false;

void setup() {
  Serial.begin(9600);
  while(!Serial);
  delay(1000);
  Serial.println("Lets go Sender");
  printf_begin();                       //needed only for Arduino Uno
  
  flag = radio.begin();
  Serial.print("radio.begin(): ");
  Serial.println(flag);
  radio.printDetails();          
  radio.printPrettyDetails(); 

  radio.setDataRate(RF24_250KBPS);
  Serial.print("radio.setDataRate(RF24_250KBPS): ");
  Serial.println(flag);
  radio.printDetails();          
  radio.printPrettyDetails();

  radio.openWritingPipe(address);       //Setting the address where we will send the data
  Serial.println("radio.setDataRate(RF24_250KBPS): ");
  radio.printDetails();          
  radio.printPrettyDetails(); 
  
  radio.setPALevel(RF24_PA_MAX);         //You can set it as minimum or maximum depending on the distance between the transmitter and receiver.
  Serial.println("radio.setPALevel(RF24_PA_MAX): ");
  radio.printDetails();          
  radio.printPrettyDetails();

  radio.stopListening();                //This sets the module as transmitter
  Serial.println("radio.stopListening(): ");
  radio.printDetails();          
  radio.printPrettyDetails();
  }

void loop(){
  const char text[] = "Hello there...";
  radio.write(&text, sizeof(text));                  //Sending the message to receiver

  delay(1200);
}
```

**Receiver (Arduino Nano 33 BLE and/or Arduino Uno)**
```
[... 119 lines cut ...]
PS1. Only 2 ouf 3 systems are powered while the test occurs (Nano 33 BLE and Uno, or two Nano 33 BLE's).
PS2. I happened to find a thread where a very similar issue was discussed but with an Arduino MEGA, no solution was found.
PS3. I inserted  lines of code in the method **RF24::available(...)** to print the CONFIG and STATUS registers while sending data. The problem is still there without this modification, also without the modification to **printf.h** of my first post.

```
bool RF24::available(uint8_t* pipe_num)
{
    // get implied RX FIFO empty flag from status byte
	print_byte_register(PSTR("CONFIG\t"), NRF_CONFIG);
	print_status(get_status());
    uint8_t pipe = (get_status() >> RX_P_NO) & 0x07;
    if (pipe > 5)
        return 0;

    // If the caller wants the pipe number, include that
    if (pipe_num)
        *pipe_num = pipe;

    return 1;
}
```

Best regards, 

Tonio

2bndy5, 2021-03-01:
>Only 2 ouf 3 systems are powered while the test occurs (Nano 33 BLE and Uno, or two Nano 33 BLE's).

To power off a radio, do you disconnect the usb cable (and whatever the radio's VCC pin is supplied with)? Or do you call `radio.powerDown()`?

Just trying to eliminate the possibility that the payload is intercepted.

I don't often get to say this, but you've got me stumped. `print*Details()` output looks as it should, RX operation doesn't require the same current consumption (at least when not sending an ACK packet) as TX operation requires, and even the RPD flag is not getting asserted (as it should when detecting _any_ signal on that channel & data rate).

If you run the scanner example (found in "examples/old_backups" folder) and no signals are detected, then I would think the radio module is defective (especially in the presence of WiFi/Bluetooth).

TonioChingon, 2021-03-01:
Hi, 
To disconnect a radio, I am unplugging the Arduino from the usb voltage source.

Sometimes the RPD flag gets active, but very rarely.

This is the output of the **scanner** example. It seems that the default channel (0x4C) is not being affected by noise.
I tried few other channels and it did not work (great Scanner example, btw).

[image]

Btw, I tried the same NRF24L01+ module with an Arduino DUE and it worked perfectly.

Some brainstorming...

- Could it be a timing issue? Since the Uno and the Nano 33 BLE run at different clock speeds, there might be a factor in the .h files that is causing the trouble?
- The Uno has a 5V logic whereas the Nano 33 BLE has a 3V3 logic. However, the DUE has also 3V3 logic aswell, and it worked fine.
- The BLE module of the Nano 33 BLE creates interference, but it is not even activated in the code, and the channel is not suffering interference according to the scanner example.

2bndy5, 2021-03-01:
Love the brainstorming: Your brain is on the same page as mine. Thanks for the scanner output; this not only verifies that the radio is in working order but also the CE pin is behaving as expected.

>Could it be a timing issue?

This is worth reviewing... I would start with source code for `begin()`

TonioChingon, 2021-03-01:
yeah, I will do that, it will take a while though.

Btw, I did the scanner test, while another NRF14L01+ module was transmitting data on the 0x69 channel, which is being reflected in the scanner test, i.e. the module is actually "seeing" data.

2bndy5, 2021-03-01:
BTW, according to BLE specs, any BLE device only focuses on channels 2, 26, and 80.

TonioChingon, 2021-03-02:
I figured it out, well partially. It was a HW problem.

Apparently, the 3V3 output of the Nano 33 BLE was the problem.
I changed the 3V3 source to an external source and now it is working well.

I am not sure why the voltage output of the Nano 33 BLE is not enough. According to the datasheet of the NRF24L01+, more current is drawn on RX mode than in TX mode. There does not seem to be a large difference in the RX and TX modes, though.

[image]

This is kind of disappoting, as I will have to include an external 3V3 voltage source to my setup.

Anyways, it was fun debugging the library and getting to know in detail the NRF24L01+ module.
I think this might be useful to other users of this library in the Nano 33 BLE, specially with the printf.h fix in the first post.

TMRh20, 2021-03-02:
Hmm, most 3.3v devices can handle the Ned, that’s a bit disappointing. Thanks for the info though, I will have to add your modification to printf.h

2bndy5, 2021-03-02:
Good work! It's been so long since I looked at that part of the datasheet (I blame PA/LNA modules for that). I bet the scanner example will show much more ambient signals with a proper power source also.

closed 2021-04-11 by 2bndy5