nRF24/RF24 #740: [Question] Hooking Arduino Nanos with nrf24l01+'s

BatMitio, 2021-02-28:
Hi there,
Your library is great, easy to use and functional. I'm having problems with making connection between two nrfs. I have hooked the transmitter side using a Chinese nano powered from a 12 - 5v step down linear converter LM2956 and uploaded the transmitter code from this tutorial [https://lastminuteengineers.com/nrf24l01-arduino-wireless-communication/](url). On the Rx side I have  the same configuration but powered from my laptop USB also with Rx code from the same tutorial. NOTHING.
On the Buck converter side I have soldered it and double checked the connections and I am 100 percent sure the connections there are OK. I should probably mention that I am using the adapter boards with the regulators on them to step the voltage down from 5 to 3.3 volts for the nrf. 
I made a test to see if they even transmit or receive something so i uploaded the slightly modified version of a frequency scanner for nrf I found online and made the transmitter sweep from channel 30 to 120 and transmit 10 times the same 'Hello World' string at every channel. On the scanner I saw it sweeping so I knew it was able to receive and the sweeper to transmit. So I swapped the modules and did the same test successfully so I know both can transmit and receive. 
But whatever I do I cannot get them to communicate. The radio.available() always returns false and that's it. 
A different test I did was on the receiving side i substituted radio.available() with radio.testCarrier() and saw that when the transmitter is off nothing happens but when I turn it on it starts printing on the console but still it prints an empty string. One little thing there, when I power the transmitter off again the receiver still thinks there is a carrier. 
I tried to put the old 1.1.7 library and uninstall all others related to that but still, nothing happens.
I would be pretty happy if you can help me out with this.

My boards are these ones (ATMEGA168): [https://www.aliexpress.com/item/4000903444456.html?spm=a2g0o.productlist.0.0.20095756zcR21c&algo_pvid=fedfa13e-cdd2-411d-ac80-0efd66662515&algo_expid=fedfa13e-cdd2-411d-ac80-0efd66662515-19&btsid=0b0a556216145241257843892ef3da&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_](url)

The nrfs are these with a 3.3 v regulator board each: [https://www.aliexpress.com/item/1005001621991136.html?spm=2114.12010612.0.0.231446e0zG43TI&gps-id=pcStoreJustForYou&scm=1007.23125.137358.0&scm_id=1007.23125.137358.0&scm-url=1007.23125.137358.0&pvid=98e1fb92-4526-42f3-810e-2d6d6e98b7f9](url)

P. S. I should probably mention that when I first soldered the transmit side there was a solder drop shortening miso and mosi that i didn't saw and the module was getting hot so I removed it and then ran all the tests from above so my conclusion is that I haven't fried the PA because both are emitting, I haven't fried the LNA because both are receiving, and I haven't fried the SPI logic because both can switch channels. It happens to be in my mind that they are just unable to interpret each others data.

TMRh20, 2021-02-28:
> Your library is great, easy to use and functional. I'm having problems with making connection between two nrfs. I have hooked the transmitter side using a Chinese nano powered from a 12 - 5v step down linear converter LM2956 and uploaded the transmitter code from this tutorial https://lastminuteengineers.com/nrf24l01-arduino-wireless-communication/. On the Rx side I have the same configuration but powered from my laptop USB also with Rx code from the same tutorial. NOTHING.

I would suggest testing with the included examples. If you are using the high powered ( pa+lna ) modules try setting the rf power level to low.

The library itself is well tested using Arduino nanos, so there’s not much we can help with, this would be a wiring/power supply/coding issue not an issue with the library.

BatMitio, 2021-02-28:
Sorry for bothering you, I just ordered another two of the modules just in case I have fried them I guess I'm closing it!

closed 2021-02-28 by BatMitio

2bndy5, 2021-02-28:
PA/LNA modules also benefit greatly from adding electromagnetic shielding. I've added some info about these to [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md)