nRF24/RF24 #744: [Question] RF24 not receiving any payload

juampapo546, 2021-03-17:
Hello, I’m trying to communicate 2 Arduinos nano with RF24 antennas. I’m using a A1117 to take the 5v port to 3.3v on both arduinos.

Here is the   radio.printPrettyDetails() and the code of my arduinos:

Receiver

14:06:20.871 -> SPI Frequency		= 10 Mhz
14:06:20.905 -> Channel			= 108 (~ 2508 MHz)
14:06:20.941 -> RF Data Rate		= 1 MBPS
14:06:20.941 -> RF Power Amplifier	= PA_LOW
14:06:20.976 -> RF Low Noise Amplifier	= Enabled
14:06:21.011 -> CRC Length		= 16 bits
14:06:21.049 -> Address Length		= 5 bytes
14:06:21.049 -> Static Payload Length	= 32 bytes
14:06:21.118 -> Auto Retry Delay	= 1500 microseconds
14:06:21.152 -> Auto Retry Attempts	= 15 maximum
14:06:21.152 -> Packets lost on
14:06:21.189 ->     current channel	= 0
14:06:21.222 -> Retry attempts made for
14:06:21.256 ->     last transmission	= 0
14:06:21.256 -> Multicast		= Disabled
14:06:21.289 -> Custom ACK Payload	= Disabled
14:06:21.326 -> Dynamic Payloads	= Disabled
14:06:21.364 -> Auto Acknowledgment	= Disabled
14:06:21.401 -> Primary Mode		= TX
14:06:21.401 -> TX address		= 0xe8e8f0f0e1
14:06:21.434 -> pipe 0 (closed) bound	= 0xe8e8f0f0e1
14:06:21.469 -> pipe 1 ( open ) bound	= 0xe8e8f0f0e1
14:06:21.503 -> pipe 2 (closed) bound	= 0xc3
14:06:21.538 -> pipe 3 (closed) bound	= 0xc4
14:06:21.573 -> pipe 4 (closed) bound	= 0xc5
14:06:21.607 -> pipe 5 (closed) bound	= 0xc6

#include <printf.h>
#include <nRF24L01.h>
#include <RF24_config.h>
#include <RF24.h>
#include <SPI.h>

const uint64_t pipeIn = 0xE8E8F0F0E1LL; 
RF24 radio(9, 10);  
struct Received_data {
  byte pwm;
};

Received_data received_data;

int ch1_value = 0;

void reset_the_Data()  {
  received_data.pwm = 0;
}

void setup() {
  reset_the_Data();
  Serial.begin(9600);
  radio.begin();
  radio.setAutoAck(false);
  radio.setPALevel(RF24_PA_LOW);
  radio.setChannel(108); 
  radio.openReadingPipe(1,pipeIn);
  printf_begin();

  radio.startListening();
  radio.printPrettyDetails();

}

unsigned long lastRecvTime = 0;

void receive_the_data() {
  while ( radio.available() ) {
  radio.read(&received_data, sizeof(Received_data));
  lastRecvTime = millis(); 
}
}

void loop() {

  receive_the_data();

  unsigned long now = millis();
  if ( now - lastRecvTime > 1000 ) {
    reset_the_Data();

  } 

  ch1_value = map(received_data.pwm, 0, 255, 0, 255);
  Serial.print("Radio available: ");
  Serial.println(radio.available());
  Serial.print("Recibo (");
  Serial.print(lastRecvTime);
  Serial.print("): "); 
  Serial.println(ch1_value);
  
}

Transmitter

Start Radio
14:07:46.018 -> SPI Frequency		= 10 Mhz
14:07:46.056 -> Channel			= 108 (~ 2508 MHz)
14:07:46.089 -> RF Data Rate		= 1 MBPS
14:07:46.124 -> RF Power Amplifier	= PA_LOW
14:07:46.124 -> RF Low Noise Amplifier	= Enabled
14:07:46.161 -> CRC Length		= 16 bits
14:07:46.195 -> Address Length		= 5 bytes
14:07:46.230 -> Static Payload Length	= 32 bytes
14:07:46.266 -> Auto Retry Delay	= 4000 microseconds
14:07:46.300 -> Auto Retry Attempts	= 15 maximum
14:07:46.338 -> Packets lost on
14:07:46.338 ->     current channel	= 0
14:07:46.374 -> Retry attempts made for
14:07:46.412 ->     last transmission	= 0
14:07:46.446 -> Multicast		= Disabled
14:07:46.446 -> Custom ACK Payload	= Disabled
14:07:46.481 -> Dynamic Payloads	= Disabled
14:07:46.518 -> Auto Acknowledgment	= Disabled
14:07:46.552 -> Primary Mode		= TX
14:07:46.587 -> TX address		= 0xe8e8f0f0e1
14:07:46.621 -> pipe 0 (closed) bound	= 0xe8e8f0f0e1
14:07:46.621 -> pipe 1 (closed) bound	= 0xe8e8f0f0e1
14:07:46.691 -> pipe 2 (closed) bound	= 0xc3
14:07:46.725 -> pipe 3 (closed) bound	= 0xc4
14:07:46.763 -> pipe 4 (closed) bound	= 0xc5
14:07:46.800 -> pipe 5 (closed) bound	= 0xc6

#include <printf.h>
#include <nRF24L01.h>
#include <RF24_config.h>
#include <RF24.h>
#include <SPI.h>

const uint64_t my_radio_pipe = 0xE8E8F0F0E1LL;
RF24 radio(9, 10);   

struct canales_radio {
  byte pwm;
  
};

canales_radio enviar;

void setup() {
  
  Serial.begin(9600);
  radio.begin();
  printf_begin();
  radio.setAutoAck(false);
  radio.setPALevel(RF24_PA_LOW);
  radio.setChannel(108); 
  radio.setRetries(15, 15);
  radio.openWritingPipe(my_radio_pipe);
  enviar.pwm = 0;
  Serial.println("Start Radio");
  radio.printPrettyDetails();
}

void loop() {

  enviar.pwm = map(analogRead(A0), 0, 1024, 0, 255);

  if(radio.write(&enviar, sizeof(canales_radio))){
    Serial.print("Sending: ");
  }else {
   Serial.print("Not sent: ") ;
   }
  Serial.println(map(analogRead(A0), 0, 1024, 0, 255));

}

2bndy5, 2021-03-17:
You should call `stopListening()` at the end of you transmitter's `setup()` function (preferably right before you call `printPrettyDetails()`). There is a difference between a transmitter not sending and receiver not receiving. Does your transmitter code ever print `Not sent: ...` (I doubt it should because you disable auto-ack)?

## Additional advice
Please be cautious of following unofficial example code (the code you posted feels like it didn't come from any of the official RF24 library examples).

This
```c++
#include <printf.h>
#include <nRF24L01.h>
#include <RF24_config.h>
#include <RF24.h>
#include <SPI.h>
```
 could be reduced to 
```c++
#include <printf.h>
#include <SPI.h>
#include <RF24.h>
```

Its better (in my experience) to pass an instantiated object to `sizeof()` (not the datatype of the instantiated object). So
```c++
sizeof(canales_radio)
// and 
sizeof(Received_data)
```
should be 
```c++
sizeof(enviar)
// and
sizeof(received_data)
```

It might also be better to `setPayloadSize(sizeof(data_variable))` for various reasons; most importantly `radio.read()` can leave the top RX FIFO level occupied with the received payload that was appended with `0`s if `setPayloadSize()` wasn't called on both receiver and transmitter. In your case, I suspect the code you posted would transmit 32 bytes when the last 31 bytes would just be all `0`s

juampapo546, 2021-03-18:
Thanks for the answer. I did all the changes you recommended, I had auto-ack disabled for debugging purposes since it wasn't working and several forums said to disable it if I had any troubles. However there is still no connection, I called   `radio.stopListening()` on my setup as you instructed just before `printPrettyDetails()` and set the payload size, on the transmitter code I did
 ```  to_send.pwm = map(analogRead(A0), 0, 1024, 0, 255);
  radio.setPayloadSize(sizeof(to_send.pwm));
  Serial.println(sizeof(to_send.pwm));
```
And on the receiver I just wrote `  radio.setPayloadSize(1); `

**Notes**
 * @LucasAlda is on this project with me if he wants to add anything
 *  We did try the GettingStarted example and I'm getting `Transmission failed or timed out` on the transmitter
 * I'm about to push the code to a repo in https://github.com/JLCA-Boats if you want to review again

stanekTM, 2021-03-18:
@juampapo546
Hi, you may want to be inspired by my TX, RX code using this great library.
I'm using v1.3.9, so far I have no reason to update for absolutely no problems.
[https://github.com/stanekTM/RC_RX_nRF24L01_Telemetry_Motor_Driver_Servo](url)

2bndy5, 2021-03-18:
@stanekTM I'm not sure your input is relevant (it actually feels like spam).

@juampapo546 it sounds like your project is suffering from power supply noise. Are you using a capacitor on the nRF24 VCC & GND? Lots of people recommend turning off auto-ack because most of the clone/counterfeit chips don't work with auto-ack enabled (and partly because people don't understand what the feature does or how it works). 

Since you enabled auto-ack, the other thing to pay attention to (on the transmitter) is the "Retry attempts made for last transmission" output of `printPrettyDetails()`. This will tell you how many times the radio tried to send for the last call to `write()`. Again this info is only relevant when auto-ack is enabled.

stanekTM, 2021-03-18:
I'm sorry, I accidentally wrote the post next to it, I'll delete it, thanks

juampapo546, 2021-03-18:
No, we're not using a capacitor just the voltage regulator. Which capacitor you recommend us to get (how much µF)? Afaik both arduinos are original, unless we've been scammed (hope not haha). Btw, auto-ack makes sure connection is established before sending any packages right?

It makes 15 attempts :
```
18:22:43.441 -> Retry attempts made for
18:22:43.441 ->     last transmission	= 15
```

> Just copies `received_data.pwm` to the variable `ch1_value` 

We know that, we were previously mapping to another values but left it that way temporarily until we can get the antennas to communicate

> @stanekTM I'm not sure your input is relevant (it actually feels like spam).

Lol it kinda does

2bndy5, 2021-03-18:
>No, we're not using a capacitor just the voltage regulator

Wait, what?! The capacitor question would've been my first, but linear voltage regulators usually require these  (as specified by the datasheet). If you're using just the 3V output from the Arduino, then a 100uF is a safe bet to use. However, you should look at the datasheet for the A1117 regulator; it should tell you what capacitance to use (usually need at least 2 capacitors - 1 on V<sub>in</sub> and GND & 1 on V<sub>out</sub> & GND) Ironically, the A1117 is similar to the voltage regulator used on the adapter boards specifically made for the nRF24L01+PA+LNA modules (which has builtin capacitors). If your're trying to say that you're using that adapter board, then try adding a small (I've used 2uF in the past) capacitor on the adapter board's voltage input.

juampapo546, 2021-03-18:
Using the 3v output (with or without capacitor) won't deliver enough power (I'm guessing) since it doesn't power up the antenna, I'm searching for the A1117 datasheet to see what capacitor I should get and I'll report

2bndy5, 2021-03-18:
>Btw, auto-ack makes sure connection is established before sending any packages right?

Actually, auto-ack makes the transmitter automatically enter RX mode and wait for an Acknowledgement (ACK for short) packet from the receiver (which is only transmitted if the receiver actually received a payload and the auto-ack feature is enabled for both transmitter and receiver). After waiting for the ACK, the transmitter goes back to TX mode (successful or not).

>I'm searching for the A1117 datasheet to see what capacitor I should get and I'll report

Do you have a store link? Websites like alibaba, amazon, ebay, etc don't always provide a link to the datasheet. ☹️

2bndy5, 2021-03-18:
This may not be the same exact part, but just as a sample:
- [a datasheet from mouser.com](https://www.mouser.com/datasheet/2/348/ba1117fp-e-1873964.pdf) (digikey has the same product/datasheet)

juampapo546, 2021-03-18:
>Do you have a store link? Websites like alibaba, amazon, ebay, etc don't always provide a link to the datasheet. ☹️

Ok so it did have a datasheet, looking at it rn

2bndy5, 2021-03-18:
Performance on those PA/LNA modules could also be improved by adding Electromagnetic shielding (I've added instructions with pictures to the [COMMON_ISSUES.md file](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#my-palna-module-fails-to-transmit)).

2bndy5, 2021-03-31:
@juampapo546 did you get those capacitors? more importantly, did you get the radio working reliably? (I'm looking to close this)

juampapo546, 2021-03-31:
@2bndy5 my partner got them but due to timing and work we couldn't test them, we will probably get to it on the next four-day holiday

juampapo546, 2021-04-01:
I got the capacitors and it's still not working, I'll return the antennas

closed 2021-04-01 by juampapo546

2bndy5, 2021-04-01:
That's dissappointing.

Just curious, can you post the store link to the module you bought? (so others can avoid going through this)