nRF24/RF24 #745: NRF24L01 non funziona su arduino due

davide-1999, 2021-03-18:
Buongiorno sto cercando di far comunicare il mio arduino due con un nrf24l01 + pa+lna.
Sto usando il pin 9 come CE e il pin 10 come CSN ma quando uso" radio.printDetails () ", ci sono 0 dappertutto e so che c'è qualcosa che non va.
Sto usando l'alimentazione 3.3V di arduino due che dovrebbe fornire la corrente necessaria al modulo, inoltre ho ricontrollato i collegamenti mille volte
Sto usando la libreria RF24 di TMRh20
Ho provato a seguire tantissime guide e provato più di un modulo nrf24l01 ma nessuno ha funzionato
Grazie mille per l'aiuto

davide-1999, 2021-03-18:
Come pin SPI sto usando quelli consigliati sul sito di arduino ossia SCK 52,MOSI 51, MISO 50

TMRh20, 2021-03-18:
è necessario un alimentatore separato per gestire i moduli lna + pa. facile da testare con batterie 3.3v. Penso che potrebbe essere necessario utilizzare l'intestazione icsp per i pin spi. vedi pinout per il Due

> On Mar 18, 2021, at 7:37 AM, davide-1999 ***@***.***> wrote:
> 
> Buongiorno sto cercando di far comunicare il mio arduino due con un nrf24l01 + pa+lna.
> Sto usando il pin 9 come CE e il pin 10 come CSN ma quando uso" radio.printDetails () ", ci sono 0 dappertutto e so che c'è qualcosa che non va.
> Sto usando l'alimentazione 3.3V di arduino due che dovrebbe fornire la corrente necessaria al modulo, inoltre ho ricontrollato i collegamenti mille volte
> Sto usando la libreria RF2

davide-1999, 2021-03-18:
Ma i 3.3V di arduino due non dovrebbero fornire 800mA che sono molto maggiori di 125mA massimi richiesti dal modulo lna+pa?
Bisogna usare l'intestazione ICSP o SPI? perchè ci sono pareri differenti in rete

TMRh20, 2021-03-18:
Il Due dovrebbe avere abbastanza corrente ma è un problema noto con i moduli PA + LNA. Puoi provare a impostare radio.setPALevel (RF24_PA_LOW, 0); anche. Sì, se hai un'intestazione SPI, usala al posto di ICSP.

davide-1999, 2021-03-18:
Perfetto grazie mille
Problema risolto bastava usare l'intestazione SPI invece dei pin consigliati normalmente

closed 2021-03-18 by davide-1999