nRF24/RF24 #748: [Question] What's up with the "tests" folder?

2bndy5, 2021-03-20:
I've left this alone for a while (I still can leave it as is), but I was thinking of writing some special test programs that check if `SPI_UART` and `SoftSPI` implementations (& possibly more implementations not used in the examples) compile properly -- _almost_ like code coverage analysis. I'd probably use an addition GitHub action workflow based on ArduinoCLI action just for these tests.

Currently, it looks like the existing tests are relics from ManiacBug days, and the pde/jamfile formats are outdated (& printf.h is included with the lib now). I haven't looked too closely at what scenarios they're testing, but I've been relying on the GitHub actions when compiling for hardware I don't actually have access to. I know that's no substitute for actual hardware testing, but it's close (and cheaper in both time & money).

TMRh20, 2021-03-21:
All I know is that is all from the ManiacBug days. The lib could still use some cleanup, despite all the work thats been done so far.

2bndy5, 2021-03-21:
I looked closer, and the tests are just a copy of maniacBug'a examples with the Serial output modified so that a test script can capture the STDOUT and compare with predefined successful output. These tests still exist on maniacBug's repo (which isn't marked as archive only, so someone must care about maintaining that source), so I'm not worried about replacing them with a more updated set of programs.

closed 2021-03-22 by 2bndy5