nRF24/RF24 #751: Pulling my hair out!

budnikasr, 2021-03-21:
Hi Everyone.

I have spent a week trying to get this to work following 4 or 5 video tutorials, plus reading all the different threads everywhere about errors etc. and literally pulling my hair out. I figured after a week trying I would ask for help. It's just so frustrating. I apologize in advance this is a long thread.

I have 4 NRF24l01 units and they work perfectly on Arduino, so I know they are not defective. My units are the PA and LNA versions if that makes a difference. I've read that attaching a capacitor is critical, but if that is the case why does it work with Arduino, which by the way, I was able to achieve nearly 750m line of sight tx/rx. So I did try it on one of the units that I know work. Attached a 10uf capacitor. Tested on arduino, it worked. Tested on rasberry pi, did not work.

Here are the pins I have configured, but first want to point out something strange on the VCC. Attaching to VCC 3.3v on the actual raspberry pi 4 doesn't seem to work. No understanding why so I attached the VCC and GND to an external pwr board and ran it off a 9v battery. 

VCC - external power 3.3v 
GND - external power
CE -     Pin 11 (GPIO17) -(tried GPIO22 as well)
CSN -  Pin 24 (GPIO8)
SCK -   Pin 23 (GPIO11)
MISO - Pin 21 (GPIO 9)
MOSI - Pin 19 (GPIO 10)

As I mentioned I followed many different tutorials, two in particular from this site:

http://thezanshow.com/electronics-tutorials/raspberry-pi/tutorial-32-33
http://thezanshow.com/electronics-tutorials/raspberry-pi/tutorial-34-35

I could not get pi to pi communication or Arduino to pi communication. It would continuously time out. Here is the output from the receiveArduino.py file when it was run. 

`pi@raspberrypi:~/Desktop/NRF24L01 $ sudo python3 receiveArduino.py
/home/pi/Desktop/NRF24L01/lib_nrf24.py:378: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self.GPIO.setup(self.ce_pin, self.GPIO.OUT)
STATUS   = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xe8e8f0f0e1 0xf0f0f0f0e1
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xe8e8f0f0e1
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x03
RF_CH            = 0x76
RF_SETUP         = 0x01
CONFIG           = 0x0c
DYNPD/FEATURE    = 0x3f 0x06
Data Rate        = 1MBPS
Model            = nRF24l01+
CRC Length       = 16 bits
PA Power         = PA_MIN

Sent the message: ['G', 'E', 'T', 'S', 'T', 'R', 'I', 'N', 'G', 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Timed out.
Received: []
Translating our received Message into unicode characters...
`

So not receiving or transmitting, I am stumped at this point. I then went to this page https://nrf24.github.io/RF24/pages.html
and read through everything and instead of trying the Linux General/Raspberry Pi, I looked in Linux Installation and found another setup and followed it. I got so excited when the first time I actually saw the transmission occur, but with errors.

`pi@raspberrypi:~/rf24libs/RF24/examples_linux $ sudo ./gettingstarted
radio hardware is not responding!!
pi@raspberrypi:~/rf24libs/RF24/examples_linux $ sudo ./gettingstarted
./gettingstarted
Which radio is this? Enter '0' or '1'. Defaults to '0'
*** PRESS 'T' to begin transmitting to the other node
*** PRESS 'R' to begin receiving from the other node
*** PRESS 'Q' to exit
t
Transmission failed or timed out
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
Transmission failed or timed out
Transmission failed or timed out
Transmission failed or timed out
Transmission successful! Time to transmit = 230 us. Sent: 0.02
Transmission successful! Time to transmit = 201 us. Sent: 0.03
Transmission successful! Time to transmit = 184 us. Sent: 0.04
Transmission successful! Time to transmit = 206 us. Sent: 0.05
Transmission successful! Time to transmit = 206 us. Sent: 0.06
Transmission failed or timed out
Transmission failed or timed out
Transmission successful! Time to transmit = 27995 us. Sent: 0.07
Transmission failed or timed out
Transmission failed or timed out
Transmission failed or timed out
6 failures detected. Leaving TX role.
*** PRESS 'T' to begin transmitting to the other node
*** PRESS 'R' to begin receiving from the other node
*** PRESS 'Q' to exit`

So I am at a loss. I think I have literally tried everything. 

Any help would be much appreciated.

Robert

budnikasr, 2021-03-21:
Thanks Mate, Just wanted to let people know how frustrated I was.

Robert

2bndy5, 2021-03-21:
Your problem is still a hardware problem because you're using the PA/LNA modules. Have a look at [the COMMON_ISSUES.md concerning the PA/LNA modules](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#my-palna-module-fails-to-transmit). The advice about electromagnetic shielding may be essential in your case.

How are you regulating the 9v power source down to 3v? I'm mostly concerned about the amperage here.

I'm sorry you got so frustrated, but I'm not interested in the unofficial tutorials/videos you followed because I get frustrated when people don't take the time to read our library's docs (betting you didn't either).

budnikasr, 2021-03-21:
Hi 2bndy5,

> I have spent a week trying to get this to work following 4 or 5 video tutorials, plus reading all the different threads everywhere about errors etc. and literally pulling my hair out.

When I moved from the Arduino to the pi, initially, no I didn't, but as I said before I've spent a week reading through all the threads on errors including https://nrf24.github.io/RF24/md_COMMON_ISSUES.html. I did read the entire document and did try the suggestion by wrapping the board with electrical tape and then with foil. I received the same identical results.

Regarding the power supply, since I did read the common issues, I had already been running these boards with these type of power supplies - https://www.amazon.ca/Power-Supply-Module-Solderless-Breadboard/dp/B00BXWV2F6  My arduino setup is 2 ESP32 boards on breadboards, so these power supplies are attached to the breadboard.

This is why I am stumped. Would you have a recommendation on a better board and not the clones?

Thanks,

Robert

Avamander, 2021-03-21:
What type of wires are you using for the SPI connection? What distro is running on the RPi?

2bndy5, 2021-03-21:
Thank you for the store links. Are you connecting the GND of the power supply with the GND of the RPi?

>I've read that attaching a capacitor is critical, but if that is the case why does it work with Arduino

Capacitors help reduce the electric noise in the power supply; nRF24 modules are sensetive to this noise (especially nRF24 clones/counterfeits that are missing capacitors needed on the module - to make manufacturing cheaper). Capacitance values are largely dependent on the voltage regulating implementation (which is why suggestions vary from different advice) start with small capacitance value (like you did), then try larger values.

budnikasr, 2021-03-21:
Hi Avamander,

> What type of wires are you using for the SPI connection? What distro is running on the RPi?

I'm using the same breakout wires I use on Arduino and yes, I have changed them out. The distro I am running is:
Raspberry Pi OS with desktop
Release date: January 11th 2021
Kernel version: 5.4

2bndy5, 2021-03-21:
>Would you have a recommendation on a better board and not the clones?

I usually recommend [these adapters](https://www.amazon.ca/Wireless-Module-NRF24L01-Pinboard-Adapter/dp/B07YQXPCYN/ref=sr_1_7?dchild=1&keywords=nrf24l01+adapter&qid=1616365920&s=electronics&sr=1-7) (designed for the PA/LNA modules specifically). However, Its likely using the same linear 3v regulator (AMS1117) that your breadboard power supply module is using.

budnikasr, 2021-03-21:
Hi Brendan,

> Thank you for the store links. Are you connecting the GND of the power supply with the GND of the RPi?
I'm connecting the GND to the power supply.

And, you've been a big help, Thank you.

Robert

budnikasr, 2021-03-21:
Hi Again,

I have tried both with the power regulator and without. The Arduinos both have the regulators attached.

2bndy5, 2021-03-21:
General rule of electronics: Connect all isolated GND lines together. This is more important when using BJTs (like a linear voltage regulator).

budnikasr, 2021-03-21:
> General rule of electronics: Connect all isolated GND lines together. This is more important when using BJTs (like a linear voltage regulator).
> 
> EDIT just don't mix AC GND with DC GND (not a problem in this scenario).

Will Do! thanks for the tip!

2bndy5, 2021-03-31:
@budnikasr did you solve this? can we close this?

closed 2021-06-12 by 2bndy5