nRF24/RF24 #752: [Question] utility/Due/RF24_arch_config.h not needed?

2bndy5, 2021-03-22:
I couldn't find where the Arduino Due support file _/utility/Due/RF24_arch_config.h_ file is/was getting included. There's no following snippet in the _/RF24_arch_config.h_
```CPP
#if defined (ARDUINO_SAM_DUE)
    // ARDUINO_SAM_DUE is defined using CLI args (-DARDUINO_SAM_DUE) at compile time
    #include "utility/Due/RF24_arch_config.h"
#endif
```
So, against my better nature, I went out of my way to break Due support by commenting out everything in _/utility/Due/RF24_arch_config.h_ and un-commenting  `SERIAL_DEBUG` in _/RF24_arch_config.h_... It still compiles the same as before (I can't break Due support if I tried?! 😆 ) I suspect _printf.h_ doesn't work for the Due either (judging from the [ArduinoCore-SAM repo](https://github.com/arduino/ArduinoCore-sam)), but it might (using [`sprintf()` from _avr/pgmspace.h_](https://github.com/arduino/ArduinoCore-sam/blob/790ff2c852bf159787a9966bddee4d9f55352d15/cores/arduino/avr/pgmspace.h#L99)).

@TMRh20 I have a feeling I hit yet another relic from maniacBug days.

TMRh20, 2021-03-22:
At one point I supported the extended SPI API for the due, but that was removed due to unnecessary complexity. I think that is why there was a folder for it in utility. Printf works out of the box, but no pgmspace.

2bndy5, 2021-03-22:
Actually pgmspace was added back into the core in 2013. It now gets included in RF24_config.h (using `ARDUINO_SAM_DUE`). Since I overloaded `begin()`,  I may have reintroduced the extended CSN functionality, but `csn()` (and any calls to `SPI.transfer()`) needs extra intuition about this. I thought I left the Due SPI support alone (using the useless arch_config file), but clearly I need to single out the Due in RF24_config.h and in RF24.cpp. Currently, it should work as usual without calling the overloaded `begin()`.

2bndy5, 2021-03-23:
I started re-implementing the Due's extended SPI feature, but it made the code really messy and grew the compile size quite a bit. It's not like the Due doesn't have the capacity, but we're going for "optimization" here.

I also excluded #743 from Due support (properly this time) in _RF24_config.h_ because the Due core's SPI library only supports 1 SPI bus (opposed to what the pinout may suggest).

I'm going to remove the Due arch_config file to avoid any further confusion in the future.

closed 2021-04-11 by 2bndy5