nRF24/RF24 #753: Support for RFM73/RFM75S and other bk242x-style nrf24-clones

microtronics, 2021-03-26:
Currently, the library only works with genuine nRF24 and "good" clones. I bought RFM73/RFM75 when I didn't know they are similar to nRF24, because they were available in a very small form factor. Since these modules are "dumber" they require additional effort to work with the current code.

I already made a couple of changes to the original code that I'm going to pack into DEFINES and IFDEFs so it can be set to work with the cheap clones. I even made sure, that genuine (prob. only the better clones) and the silly RFM7X-modules work together. Since the RFMs need additional initialization I will add an example that makes use of https://github.com/jnk0le/RFM7x-lib. A couple of comments in the config.h is also required.

Contrary to lots of other peoples statements I have no issues with dynamic payload lengths and auto acknoledgements on these modules.

Another set of tiny changes is required for RF24Network.

2bndy5, 2021-03-26:
>Currently, the library only works with genuine nRF24 and "good" clones

There is also integrated support for the Si24R1 clones. Which (as noted in the [RFM7x lib's readme](https://github.com/jnk0le/RFM7x-lib#no_ack-bit)) are not "good" clones.

2bndy5, 2021-03-27:
I've been reading up on the RFM73 datasheet provided by the RMF7x lib's reference folder. Since the `init_*()` functions seem to be the only differentiating behavior, I've decided to focus on understanding what they do.
1. `spi_init()` is designed to support everything that the ManiacBug library supports. However it appropriates the default SPI bus (no secondary SPI buses seem to be supported) to perform the extra `rfm7x_init()`
2. `rfm7x_io_init()` is almost identical to what RF24 lib does (though more centralized because its an old lib) for the CE and CSN pins.
3. `rfm7x_init()` is the secret sauce. In order it:
   - switches to a separate register bank (called BANK1) in which the default register bank (BANK0) is almost identical to the nRF24 register map (at least what was documented by Nordic Semi in their datasheet). 
      - Bit 7 in the STATUS register indicates which register bank is exposed (because their access shares the same register offset addresses). The RFM7x lib uses the resulting value of this bit (toggled or not) to determine if the RF7x radio module is present. This is rather reminiscent of what I did in `RF24::begin()` to detect if a non-plus variant of the nRF24L01 is present.
      - Toggling between register banks is done using the `ACTIVATE` SPI command followed by the data byte value `0x53`. It should be noted that the data byte value `0x73` (as used in the `RF24::toggle_features()`) is implemented.
   - The registers in BANK1 mostly only accept certain values. Specific attention should be paid to a register in `BANK1` called RAMP whose value (a bytearray) must be set based on the specific variant of RFM7x being used. There is also a read-only register called "Chip ID" stores the model of the RFM7x module being used.
   - when init of BANK1 is finished, switch back to using BANK0 for the app's remaining life cycle. However, manipulating register 4 in BANK1 (bit 21 controls the threshold of RPD or CD bits in register 9 of BANK0) can produce better results in RF scanner apps.

closed 2021-03-27 by microtronics