nRF24/RF24 #755: Connecting an NRF24L01 module to a 2nd SPI on a samd21g18

ex-caliper, 2021-03-28:
Hi,  

Having previously created a small network of NRF24L01 modules with Arduino Uno’s, I now have a need increase capacity in the master device.  Consequently,  I now have a custom breakout board with a samd21g18 containing an Arduino Zero bootloader. 

With the default SPI occupied, I am trying to connect the NRF24L01 module to a 2nd SPI and like most I have studied the Adafruit and Sparkfun tutorials on how to create a 2nd SPI.  https://learn.sparkfun.com/tutorials/adding-more-sercom-ports-for-samd-boards/introduction

Hence, I created an SPIClass mySPI (&sercom1, 11, 12, 10, SPI_PAD_2_SCK_3, SERCOM_RX_PAD_0); 

However, where I am lost is how to communicate with the NRF24L01 device on this 2nd SPI.  I cannot even get the device to work on the default SPI.  

How do I initialise the device, is this done by using the mySPI.beginTransaction(SPISettings(10000000, MSBFIRST, SPI_MODE0)) and further commands entered through SPI.transfer? 

Has anyone successfully connected the module to a samd21g18, either by the default SPI or more interestingly via a 2nd SPI?

Any information would be greatly appreciated, thanks in advance.

2bndy5, 2021-03-28:
I've looking to test my solution (links in comment above), but it depends on the original arduino core. I cannot test the 2nd spi bus on my Feather M0 because it already uses SERCOM2 for the default SPI and uses another SERCOM for the built-in SPI flash storage. I have other M0 based boards,  but they're tiny form factors (qt py & trinket M0), so they don't have the necessary pins broken out on board.

@ex-caliper Are you still trying to achieve this? I'd like to have my solution tested before merging into main for release.

ex-caliper, 2021-03-28:
@2bndy5
FYI, I connected the default SPI to the nRF24 and used the 2nd SPI to drive an ILI9341 display with the samd21g18.  This satisfied my needs but never managed to wire it the other way round.  However, one observation, it was important to check the order of arguments in the target device. e.g. for the ILI9341, see below:
//tft defines..
#define tft_DC 0 //PA11
#define tft_CS 3 //PA09
#define tft_RESET 1 //PA10
//2nd SPI
SPIClass mySPI (&sercom1, 11, 12, 10, SPI_PAD_2_SCK_3, SERCOM_RX_PAD_0);
//instantiate objects ---------------------------------------------------------------------------------------------------------------------------------------
Adafruit_ILI9341 tft = Adafruit_ILI9341(&mySPI, tft_DC,tft_CS,tft_RESET);
/*
 * Note the order of arguments in the above is important,
 * normally the CS goes before the DC but this 2nd SPI requires
 * it to go the other way round. See line 124 of Adafruit_ILI9341.cpp
 */
//rf24..
#define rf24_CE 6      //PA20=D6, chip enable..
#define rf24_CS 7   //PA21=D7, chip select..
RF24 radio(rf24_CE, rf24_CS);

Thanks for your work on this.  
Regards..

2bndy5, 2021-03-28:
ok I'm glad you work things out.

I will comment out the example snippet I put together for the docs until I can get it verified.

closed 2021-04-11 by 2bndy5