nRF24/RF24 #756: RPI make install failed on compatibility.o

sh4rp0, 2021-04-05:
RPI b+

[image]

pi@rpi_atom:~/RF24 $ sudo make install
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/RPi/spi.cpp
arm-linux-gnueabihf-gcc -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/RPi/bcm2835.c
arm-linux-gnueabihf-g++ -fPIC -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/RPi/interrupt.c
arm-linux-gnueabihf-gcc -fPIC  -marm -march=armv6zk -mtune=arm1176jzf-s -mfpu=vfp -mfloat-abi=hard -Ofast -Wall -pthread  -c utility/RPi/compatibility.cpp
In file included from /usr/include/c++/4.9/chrono:35:0,
                 from utility/RPi/compatibility.cpp:2:
/usr/include/c++/4.9/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
utility/RPi/compatibility.cpp:4:1: warning: ‘auto’ changes meaning in C++11; please remove it [-Wc++0x-compat]
 auto start = std::chrono::steady_clock::now();
 ^
utility/RPi/compatibility.cpp:4:6: error: ‘start’ does not name a type
 auto start = std::chrono::steady_clock::now();
      ^
utility/RPi/compatibility.cpp: In function ‘uint32_t millis()’:
utility/RPi/compatibility.cpp:8:5: warning: ‘auto’ changes meaning in C++11; please remove it [-Wc++0x-compat]
     auto end = std::chrono::steady_clock::now();
     ^
utility/RPi/compatibility.cpp:8:10: error: ‘end’ does not name a type
     auto end = std::chrono::steady_clock::now();
          ^
utility/RPi/compatibility.cpp:10:14: error: ‘std::chrono’ has not been declared
  return std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
              ^
utility/RPi/compatibility.cpp:10:41: error: ‘std::chrono’ has not been declared
  return std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
                                         ^
utility/RPi/compatibility.cpp:10:63: error: ‘end’ was not declared in this scope
  return std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
                                                               ^
utility/RPi/compatibility.cpp:10:69: error: ‘start’ was not declared in this scope
  return std::chrono::duration_cast<std::chrono::milliseconds>(end - start).count();
                                                                     ^
utility/RPi/compatibility.cpp:11:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Makefile:51: recipe for target 'compatibility.o' failed
make: *** [compatibility.o] Error 1

Fixed by appending - "-std=c++11" to CFLAGS in Makefile.inc

2bndy5, 2021-04-05:
I can't seem to reproduce this on my RPi2 or RPi4. Made sure to use BCM driver (`--driver=RPi`) like demonstrated in OP. I did confirm that the `-std=c++11` was not used in my _Makefile.inc_ CFLAGS

2bndy5, 2021-04-05:
@sh4rp0
1. what OS are you using?
2. what's the exact version of the RPi ("b+" doesn't much help)?
3. can you also post the output from `./configure`?

sh4rp0, 2021-04-05:
**1.	what OS are you using?**
Raspbian GNU/Linux 8 (jessie)
Linux rpi_atom 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux

**2.	what's the exact version of the RPi ("b+" doesn't much help)?**
Raspberry Pi 1 B+

**3.	can you also post the output from ./configure?**
pi@rpi_atom:~/RF24 $ . configure
[SECTION] Detecting arm compilation environment.
  [OK] arm-linux-gnueabihf-gcc detected.
  [OK] arm-linux-gnueabihf-g++ detected.
[SECTION] Detecting target machine.
[OK] machine detected: SoC=BCM2835, Type=RPi, CPU=armv6l.
[SECTION] Detecting DRIVER
  [OK] DRIVER detected:RPi.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.

2bndy5, 2021-04-05:
I dusted off my old RPi1 B+ and still couldn't reproduce this. Its worth noting that I didn't have a copy of Raspbian (Jessie), so I downloaded a fresh copy of Raspberry Pi OS (Buster) and updated it after first boot.

2bndy5, 2021-04-22:
In switching to the CMake build system, the `-std=c++11` will be automatically used. Please stay tuned...