nRF24/RF24 #757: Occasional slow ~100ms writes

matt-walker0, 2021-04-05:
**Describe the bug**
While pushing the NRF24 radio I've encoutered a strange bug which seems to delay write's significantly (~100ms). Not sure if this is some sort of race condition.
I define retries as two attempts with a gap of 500us.
     radio.setRetries(1,1); //delay(microseconds (n+1)*250, count)

To test I've assigned two timers, one before the write loop, one after. Loop serves as a rough signal quality indication.
      test1 = millis();
      while(ack_rec == false && count < 10)
      {
        ack_rec = radio.write(serialIN, 5);
        count += 1;
      }
      test2 = millis();

Highlighted in blue is the time difference (ms) from before writing radio data to sending data back across USB [test1], highlighted in red is the time difference (ms) from after writing to sending data back across USB [test2]. Hence ~1 second is used in the write here. More testing showed that a loop of 1 iteration had ~100ms delay, hence delay is cumulative with iterations. Additional testing also seemed to suggest that more retries in the .setRetries() did not increase delay. Also worth mentioning is that every "slow" response failed to receive any ack.

[image]

However in what should be a worst case scenario with receiver unplugged gives a ~35ms delay.
[image]

**1. Code to reproduce**
      test1 = millis();
      while(ack_rec == false && count < 10)
      {
        ack_rec = radio.write(serialIN, 5);
        count += 1;
      }
      test2 = millis();

**2. Expected behaviour**
Maximum writetime steady, even if something strange happening at RX end I'd expect the maximum time to never stray over the time taken when the RX is turned off.

**3. What device(es) are you using? Please specify make and model.**
Pro Micro on the TX side.
RPi Pico on the RX side (running Micropython).
Both using power amplified version of the NRF24 module.

2bndy5, 2021-04-05:
>Also worth mentioning is that every "slow" response failed to receive any ack.

A connection will only be as fast as it's weakest link. The TX node is using the full amount of retries you configured.

>that more retries in the .setRetries() did not increase delay

The TX node will only use the minimal amount of retries that it takes until an ACK packet is received.

>radio.setRetries(1,1)

Using 500 us delay is actually what the datasheet recommends as the minimum delay (for a 5 byte payload on 1Mbps data rate without an ACK payload). This is because of the timing it takes to actually transmit data OTA (including an empty ACK packet) and transfer said data over SPI (which varies due to SPI frequency).

>RPi Pico on the RX side (running Micropython)

It's best to test this library with itself. MicroPython may have a slower execution time. I know this for a fact because I rewrote that MicroPython driver script for CircuitPython. When testing my CircuitPython lib OTA with this lib, the transmission times were slower. BTW, we are working on Pico SDK support 😉

>Both using power amplified version of the NRF24 module.

These modules are known to struggle with the power supply when transmitting anything (including empty ACK packets). They also are sensitive to electromagnetic interference. Furthermore, the fact that they have added circuitry to implement the PA/LNA features means they aren't good test subjects (unless you're also trying to test the performance of these PA/LNA modules). When forming a hypothesis, it's best to eliminate all possible variables from the experiment. In your case the control variable seemed to be using `setAutoRetries()`.

<br>
It's also worth considering the execution time of `millis()` as it may differ on different boards.

matt-walker0, 2021-04-05:
Many thanks for the reply.

> A connection will only be as fast as it's weakest link. The TX node is using the full amount of retries you configured.
> The TX node will only use the minimal amount of retries that it takes until an ACK packet is received.

Noted. That last screenshot you can see the delay is ~35ms as the Pro Micro tries to send out signal x10 retries (x2), to a offline receiver, the latter red number is time ~11ms taken right after the .write loop and measures time to receive the floating point number back (or not in case of turned off!) and send back via USB to the host PC.

As for retries I'll check data sheet and delay more as needed, it is indeed set to a 5 byte payload.

What I am still alittle confused about is how it can go from taking ~3.5ms per write (35ms/10) to 100ms, is there something that the Pico could be doing with the ACK packet to delay this? Running without the Pico (rx) online never seems to cause these "long" writes.

> It's also worth considering the execution time of millis() as it may differ on different boards.
Millis() here is only used a reference for identifying how long it takes to transmit, you can see on first screenshot that this varies I guess dependent upon retires used.

> BTW, we are working on Pico SDK support 😉
Excellent! :)

2bndy5, 2021-04-05:
I think `micros()` is more accurate/faster than `millis()`

>What I am still a little confused about is how it can go from taking ~3.5ms per write (35ms/10) to 100ms, is there something that the Pico could be doing with the ACK packet to delay this?

Indeed, this is unusually long for a single transmission (even when averaged with rather consistent measurements on ACK success). The Pico wouldn't be doing anything to construct and send an ACK packet (that's automatic using the nRF24's auto-ack feature -- all performed by the radio's firmware). 

That MicroPython driver is probably the worst driver I've seen yet (for various reasons). For instance, the [`any()` method](https://github.com/micropython/micropython/blob/d87f42b0e53829052f17955ba1e17f938ae486fb/drivers/nrf24l01/nrf24l01.py#L201-L203) (which is supposed to be equivalent to the `RF24::available()`):
```python
        return not bool(self.reg_read(FIFO_STATUS) & RX_EMPTY)
```
This is actually slower because #631

The RF24 library waits for a max of 95 ms when it takes a long time to receive an ACK or the SPI bus simply isn't responding properly. If it supasses this hardcoded timeout, then you should see a warning like "HARDWARE NOT RESPONDING; check wiring", but that may not be showing up on a Arduino pro micro because of needed `printf()` support on the ATmega32U4.

2bndy5, 2021-04-05:
>because of needed printf() support on the ATmega32U4.

you can verify `printf()` support by trying to get the output of `printDetails()` but you [need to read the docs on that first](https://nrf24.github.io/RF24/classRF24.html#adc95213ed4c8569a90eb33122e16cea6).

closed 2021-04-07 by matt-walker0

matt-walker0, 2021-04-07:
> The RF24 library waits for a max of 95 ms when it takes a long time to receive an ACK or the SPI bus simply isn't responding properly. 

Thats it for sure, not a library issue then. I've double check connection (connection between powerboard and NRF24 wasn't great) and increased retry time delay. Seems to _mostly_ sorted it. More testing required.