nRF24/RF24 #760: Freezes during a library build pi Zero - W

jwilts, 2021-04-13:
**The following are the steps I have been following to get the Library built on a pi Zero-W

Fresh Raspian Build**

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install python3-dev python3-pip
sudo pip3 install spidev
sudo -H pip3 install --upgrade --ignore-installed pip setuptools
sudo usermod -a -G spi,gpio pi
sudo apt-get install ntp

**Setup for virtual environment**
sudo pip3 install virtualenv virtualenvwrapper
sudo rm -rf ~/get-pip.py ~/.cache/pip
nano ~/.bashrc
 
**ADD**
# virtualenv and virtualenvwrapper
export WORKON_HOME=$HOME/.virtualenvs
export VIRTUALENVWRAPPER_PYTHON=/usr/bin/python3
source /usr/local/bin/virtualenvwrapper.sh
 
**(Close and save .bashrc then “source it”)**
source ~/.bashrc
mkvirtualenv wideG -p python3  

**Setup and build the library**
workon wideG  **Using a virtual environment**
mkdir ~/rf24libs
cd ~/rf24libs
git clone https://github.com/TMRh20/RF24
cd ~/rf24libs/RF24
./configure
sudo make install
sudo apt-get install python3-dev libboost-python-dev
sudo ln -s $(ls /usr/lib/arm-linux-gnueabihf/libboost_python3-py3*.so | tail -1) /usr/lib/arm-linux-gnueabihf/libboost_python3.so
sudo apt-get install python3-setuptools
cd pyRF24/
python3 setup.py build  **<-- THIS IS WHEN IT FREEZES.**

**(Has been running for 2 hours on building 'RF24' extension)

Even clock on Pi stops.**

**Remainder of install steps I never get to**
sudo python3 setup.py install
sudo apt-get install python3-dev python3-rpi.gpio
cd ~/rf24libs/RF24/pyRF24
python3 setup.py build
sudo python3 setup.py install

2bndy5, 2021-04-13:
>Even clock on Pi stops

I assume this means your running the desktop environment. The RPi0 has only 512 Mb of RAM (iirc), and building anything with boost.python takes excessive resources. Have tried building from only the terminal (no desktop/X server occupying RAM)?

The docs do note that
>Build takes several minutes on arm-based machines. Machines with RAM less than 1GB may need to increase amount of swap for build.

I've never messed with the swap memory on a RPi, so maybe someone wiser (or google) could help with that.

>python3 setup.py build <-- THIS IS WHEN IT FREEZES.
**(Has been running for 2 hours on building 'RF24' extension)

My RPi2 took about 5 minutes to build the python wrapper (just for reference).

>Remainder of install steps I never get to
sudo python3 setup.py install
sudo apt-get install python3-dev python3-rpi.gpio
cd ~/rf24libs/RF24/pyRF24
python3 setup.py build
sudo python3 setup.py install

the only step left should be `sudo python3 setup.py install`. `sudo apt-get install python3-rpi.gpio` is required to run the _interrupt_configure.py_ example (located in _/RF24/examples_linux_).

----------
You are right to install this in a venv, because uninstalling the pyRF24 lib takes some extra doing (in my past experience) since it installs the lib in a way that pip3 doesn't know (or is unsure) how to uninstall the RF24 python module.

jwilts, 2021-04-14:
You were right! I was able to get the library to install when I disabled the GUI.  Good call!  Thanks for the super fast response.

closed 2021-04-14 by jwilts

Temirlan504, 2024-03-16:
I still have the freeze on my Rpi0 2w while creating env with python3 -m venv venv.

I don't have gui, only pi os 32 bit lite

2bndy5, 2024-03-16:
We now have a pip installable package called [pyrf24](https://nrf24.github.io/pyRF24/). I highly recommend you use that instead as it is [drop-in compatible](https://nrf24.github.io/pyRF24/#migrating-to-pyrf24) and has [its own API docs](https://nrf24.github.io/pyRF24/rf24_api.html#module-pyrf24.rf24).