nRF24/RF24 #762: Pi 3 and Pi Zero Errors

jwilts, 2021-04-15:
Please forgive the long post but I thought I would provide as much detail as possible on this particular issue.  (Hopefully you can help with this one as quickly as you did with the last)
Attachments with this post:

Filename                                         | Purpose
ExampleProgram.py                       | Revised sample   program for testing multicever functionality.
multiceiver_demo.py                      | https://github.com/nRF24/RF24/blob/master/examples_linux/multiceiver_demo.py
lib_nrf24.py                                     | Alternative   library (used to confirm not wiring / power)
RCV-Triage.py                                 | Testing receiving   program uses alternative library
XMIT-Triage.py                               | Testing transmit   program uses alternative library

When using the python wrapper on Pi zeros and Pi 3’s there is some sort of issue which causes errant behaviour and timeouts.  
I can confirm that it is NOT wiring, power or a build issue as I have been testing with an alternate library which does not have the same behaviour.
All of the Pi’s are running the latest Rasbian with a clean build and the RF library installed in virtual environment.
Pi 3s and Pi zeros are not running the desktop, PI 4’s are running through the desktop.
When I am running the enclosed ExampleProgram.py in Receive mode on the Pi Zero with a Pi4 transmitting I get duplicate data. As you can see payloads 7, 15,27 31 repeats numerous times and others are never received.  

ExampleProgram.py
*** Enter 'R' for receiver role.
*** Enter a number in range [0, 5] to use a specific node ID for transmitter role.
*** Enter 'Q' to quit example.
R
Received 8 bytes on pipe 3 from node 3. PayloadID: 1
Received 8 bytes on pipe 3 from node 3. PayloadID: 3
Received 8 bytes on pipe 3 from node 3. PayloadID: 3
Received 8 bytes on pipe 3 from node 3. PayloadID: 6
Received 8 bytes on pipe 3 from node 3. PayloadID: **7**
Received 8 bytes on pipe 3 from node 3. PayloadID: **7**
Received 8 bytes on pipe 3 from node 3. PayloadID: **7**
Received 8 bytes on pipe 3 from node 3. PayloadID: 12
Received 8 bytes on pipe 3 from node 3. PayloadID: 13
Received 8 bytes on pipe 3 from node 3. PayloadID: **15**
Received 8 bytes on pipe 3 from node 3. PayloadID: **15**
Received 8 bytes on pipe 3 from node 3. PayloadID: 14
Received 8 bytes on pipe 3 from node 3. PayloadID: **15**
Received 8 bytes on pipe 3 from node 3. PayloadID: **15**
Received 8 bytes on pipe 3 from node 3. PayloadID: **15**
Received 8 bytes on pipe 3 from node 3. PayloadID: 24
Received 8 bytes on pipe 3 from node 3. PayloadID: 25
Received 8 bytes on pipe 3 from node 3. PayloadID: 27
Received 8 bytes on pipe 3 from node 3. PayloadID: 27
Received 8 bytes on pipe 3 from node 3. PayloadID: 30
Received 8 bytes on pipe 3 from node 3. PayloadID: **31**
Received 8 bytes on pipe 3 from node 3. PayloadID: **31**
Received 8 bytes on pipe 3 from node 3. PayloadID: **31**
Received 8 bytes on pipe 3 from node 3. PayloadID: 28
Received 8 bytes on pipe 3 from node 3. PayloadID: 29
Received 8 bytes on pipe 3 from node 3. PayloadID: **31**
Received 8 bytes on pipe 3 from node 3. PayloadID: **31**
Received 8 bytes on pipe 3 from node 3. PayloadID: 30
Received 8 bytes on pipe 3 from node 3. PayloadID: **31**
Received 8 bytes on pipe 3 from node 3. PayloadID: **31**
Received 8 bytes on pipe 3 from node 3. PayloadID: **31**

When I run pi zero in Transmit mode and a Pi4 receiving I get a lot of transmission failures:
Transmission of payloadID 20 as node 2 successfull! Time to transmit = 20118.834 us
Transmission of payloadID 21 as node 2 successfull! Time to transmit = 11341.907 us
Transmission of payloadID 22 as node 2 successfull! Time to transmit = 2399.98 us
**Transmission of payloadID 23 as node 2 failed or timed out**
Transmission of payloadID 24 as node 2 successfull! Time to transmit = 641.995 us
Transmission of payloadID 25 as node 2 successfull! Time to transmit = 18345.849 us
Transmission of payloadID 26 as node 2 successfull! Time to transmit = 9510.921 us
Transmission of payloadID 27 as node 2 successfull! Time to transmit = 2407.98 us
Transmission of payloadID 28 as node 2 successfull! Time to transmit = 25427.791 us
Transmission of payloadID 29 as node 2 successfull! Time to transmit = 21889.819 us
Transmission of payloadID 30 as node 2 successfull! Time to transmit = 14867.877 us
Transmission of payloadID 31 as node 2 successfull! Time to transmit = 4180.966 us
Transmission of payloadID 32 as node 2 successfull! Time to transmit = 649.995 us
Transmission of payloadID 33 as node 2 successfull! Time to transmit = 7718.937 us
Transmission of payloadID 34 as node 2 successfull! Time to transmit = 9498.922 us
Transmission of payloadID 35 as node 2 successfull! Time to transmit = 655.995 us
Transmission of payloadID 36 as node 2 successfull! Time to transmit = 648.994 us
Transmission of payloadID 37 as node 2 successfull! Time to transmit = 661.995 us
Transmission of payloadID 38 as node 2 successfull! Time to transmit = 5951.951 us
Transmission of payloadID 39 as node 2 successfull! Time to transmit = 2412.98 us
Transmission of payloadID 40 as node 2 successfull! Time to transmit = 9497.922 us
**Transmission of payloadID 41 as node 2 failed or timed out**
Transmission of payloadID 42 as node 2 successfull! Time to transmit = 7721.936 us
Transmission of payloadID 43 as node 2 successfull! Time to transmit = 6029.95 us
Transmission of payloadID 44 as node 2 successfull! Time to transmit = 4186.966 us
**Transmission of payloadID 45 as node 2 failed or timed out**
Transmission of payloadID 46 as node 2 successfull! Time to transmit = 2410.98 us
Transmission of payloadID 47 as node 2 successfull! Time to transmit = 646.994 us
Transmission of payloadID 48 as node 2 successfull! Time to transmit = 5955.951 us
Transmission of payloadID 49 as node 2 successfull! Time to transmit = 642.995 us

6 failed or timeouts within ~75-80 transmissions is what I am seeing on average.  When I run Pi4s for both transmitter and receiver I will get to ~4300-4500 packets before I hit the 6 failures.  I have run the tests 15 times to confirm.
When I run this example the results are even further degraded. [(https://github.com/nRF24/RF24/blob/master/examples_linux/multiceiver_demo.py  )]

I mentioned the other library which I have also been testing with. 
https://github.com/ChallengerTECH/Lib_nrf24_pi
Using this Library (Albeit with only 2 pipes not all 6) the same errors rates are not there.  (I ran 63K transactions back and forth before I got to 6 errors)  I have only been running the 2 pipes as I cannot figure out from that library how to get multiceiver functionality to work. 
Although not an exact apples to apples comparison the other library is how I confirmed that it was not power / wiring etc. 

[Triage.zip](https://github.com/nRF24/RF24/files/6320244/Triage.zip)  <-- All the code I have been using to triage and the alternative library. 

Thanks for any guidance on how to further narrow this problem down.

2bndy5, 2021-04-15:
I should've mentioned this before: If what you want is python library for the RPi family my CircuitPython_nRF24L01 lib should also work for you. I can provide support for RF24 & CircuitPython_nRF24L01 libs. Any other libs I tend to defer to their respective owners. I also won't sift through another lib's src code.

Your problem might simply be interference. Try a higher channel (WiFi and Bluetooth tend to occupy the lower half of the supported spectrum). The scanner example (not written in python but I did write one for my CircuitPython_nRF24L01 lib) should show any signal that might pose as interference.

You might also want to increase the PA level (both receiver and transmitter when using ACK packets/payloads). The examples use a low PA level to try and avoid power issues with PA/LNA modules.

2bndy5, 2021-04-15:
I should note that any pure python module (written entirely in python - no C/C++) will be slower in execution as compared to a C/C++ equivalent. The "other lib" you referenced is based on https://github.com/jpbarraca/pynrf24 (which also [says this in its readme](https://github.com/jpbarraca/pynrf24#caveats)).

The python wrapper we provide is actually 2x-4x faster than any pure python module for the nRF24L01. So, any inconsistent behavior may actually be related to behavioral timing. i.e. A slower lib will show less errors because its taking extra time that allows counterpart radio nodes to do what they need to (concerning applications' circumstantial behavior).

jwilts, 2021-04-15:
I appreciate you looking into this.  I was not expecting you to review / dig into the other library, I was just passing it along as a means to show that the problem is not wiring or power-supply related.  (As I know those are often the biggest issues with these NRF boards)  I would certainly expect that the compiled C/C++ code would be faster and more efficient then the interpreted Python code.  I am limited to using Python as there are a few others helping with my larger project who only know Python and a lot of the back end is already written.  I will be looking at the Circuitpython as an option based on your recommendation.   Unfortunately increasing the PA level has not resulted in any appreciable difference.  
I am going to keep troubleshooting until I get multiceiver working on the stack of pi 3's that I have.

2bndy5, 2021-04-15:
have you looked into interference? I advise you run the scanner example for arduino platform (it tends to run faster thus picking up on more apparent interference).

My circuitPython lib's dev branch makes use of the same `SpiDev` & `RPi.GPIO` modules used by the other lib you referenced (for improved speed/performance). For easier installation, you can 
```
pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@dev
```
but be warned its more of a "bleeding edge" branch (the RF24Network module & its example are still broken). You'd also want the [modified examples from the dev branch](https://github.com/2bndy5/CircuitPython_nRF24L01/tree/dev/examples).

p.s. The other lib you reference doesn't properly allow sharing the same SPI bus for additional SPI Slave devices.

jwilts, 2021-04-15:
I am looking at interference as my next item working through the checklist.  
I did pickup on the issue with SPI sharing.  I spent a lot of time trying to get an RC522 RFID tag reader and NRF2401 to live nicely together and was never fully successful.  My inglorious retreat was to just switch to an i2c RFID tag reader (not ideal but is workable).  
Again I do appreciate all your help with this!  
BTW - This is part of a larger project to build an open source "gaming platform" that can be used to play a whole bunch of socially distance, but still competitive and fun wide games.  We have designed and built arduino UNO game pads that use RFID tags instead of pushing buttons so that they are touchless.  The pads are essentially dumb and all the smarts will be in the Pi which tells the pads what colors to light up RGB, what to display on screen, what tone to play, etc.   (Target is scouts, but will work with many groups)   In our range tests we were able to get well over 800 meters between the base stations and a central controller. (875 Yards)

Here is one of the games that we are building (18 in the plan):
Imagine the game “whack a mole” with 6 moles.
But the “moles” are spread across a very large field.
Replace the “buttons” on the moles with RFID tags so that you don’t need to physically touch anything.
Instead of only 1 player put 3 out at time. 
Each player is only chasing 1 specific color of mole.
As each player whacks a mole another lights up on a pad that is not currently in use. 
As soon as a player “whacks” 10 moles they are done.

Fastest time wins.

2bndy5, 2021-04-16:
ooo, that project sounds interesting. do you have a link? I also teach (in the summers) kids (4th-8th grades) how to build a robot (using microbits 😞 ). And your project sounds like a good influence.

jwilts, 2021-04-16:
Not a properly organized document, (Have had every intention of creating some proper documentation, just always lacking time to follow up on it with the youth who was supposed to take that on)  Here is a doc that generally explain the project.  
[Project Summary v3.docx](https://github.com/nRF24/RF24/files/6322004/Project.Summary.v3.docx)

2bndy5, 2021-04-16:
Timing seems extremely important in your application. There is probably going to be packet collision:
### Say 2 players wack the mole at the same time
When 2 packets are sent (& they collide in air), the RX node will only receive and decode 1 packet at a time. 
   - The nRF24 modules (in TX mode) will automatically retry to send a failed packet (this feature is controlled with [`RF24::setRetries()`](https://nrf24.github.io/RF24/classRF24.html#a4c6d3959c8320e64568395f4ef507aef)). To avoid on-air collision, the datasheet recommends staggering the retry delay for each TX node. This is also [demonstrated in the multiceiverDemo.ino example](https://github.com/nRF24/RF24/blob/afa3e41509cd56896363ca35f81f7e4a6b389b4f/examples/MulticeiverDemo/MulticeiverDemo.ino#L193-L196). Although I'm just now realizing I forgot to add this to the python equivalent example...
   - proximity to the RX node may favor the player/controller in closer proximity. 

It may be fairer to isolate the player's response time at the controller, then have the controller simply report the response time for that round. You could also have the payload include a bool describing if they hit the right mole or not. 
#### Are you using an IRQ pin for the RFID reader module?
ISR callbacks on Arduino don't allow the use of `millis()` or `micros()` inside the callback function (or they aren't accurate when you do call them inside the ISR). For this reason, calling `RF24::write()` inside an ISR is also prohibited (because the function uses `millis()` to detect failures or the timeout sentinel of 95 ms).

jwilts, 2021-04-16:
Packet collisions are an inevitability.  I have on the backlog list to put in the space between the retries to be based on the radio # so that there is different durations between the retries.  I should clarify though when the scout hits a "mole" another game pad "mole" that is not currently in use will light up that they will need to run to and "whack", so they will need to have the central controller managing the time. (vs the pad doing the math)  The latency from the distance to the base will get compensated for by kids having to visit different stations.  The two IRQ's available on the UNO are already tied up with the rotary encoder so I am not using them with the RFID or the NRF (for now, I do intend to do some more testing with them to make things simpler).  I would love to be able to get IRQ working for the NRF so that I can do a heartbeat to confirm that a pad is still operating.   
What is being programmed (for now) is 3 modes on the UNO. 

Mode 1:
Loop(
Wait for RFID or Button Push or Rotary Encoder
Transmit VIA NRF
Receive NRF data 
process  [update display, switch RGB, play tone etc]
)

Mode 2:
Loop(
Listen on NRF for Data
Process [update display, switch RGB, play tone etc]
Wait for RFID or Button Push or Rotary Encoder
)

Mode 3:
Loop(
Listen on NRF for Data
Process Data  (Populate arrays etc.)
)

Mode 3 is for initial data loading. (A bit of future visioning)  Rather than moving a whole lot of data back and forth between the pad and the Pi during the games, would like to move the data at Pad startup.  (Team names, people lists etc.)  and then just reference them by index #.  The mode is set 2 different ways.  When the pad is first started the rotary encoder is used to pick the Radio # and the initial mode, and a bit in the response struct from the Pi will tell the arduino which mode they should be operating in, so as each response is processed the pad will know what to do.  (So in a worst case as soon as the first RFID tag is read the pad will get set into the correct mode)

jwilts, 2021-04-19:
I downloaded and ran the scanner program from [https://github.com/nRF24/RF24/blob/master/examples/old_backups/scanner/scanner.ino](url)
Is there a link that explains how to interpret the results?

20:37:18.236 -> 
RF24/examples/scanner/
20:37:18.236 -> SPI Speedz	= 10 Mhz
20:37:18.283 -> STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
20:37:18.329 -> RX_ADDR_P0-1	= 0x7878787878 0xb3b4b5b6f1
20:37:18.376 -> RX_ADDR_P2-5	= 0xcd 0xa3 0x0f 0x05
20:37:18.421 -> TX_ADDR		= 0xe7e7e7e7e7
20:37:18.467 -> RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
20:37:18.467 -> EN_AA		= 0x00
20:37:18.515 -> EN_RXADDR	= 0x01
20:37:18.515 -> RF_CH		= 0x4c
20:37:18.562 -> RF_SETUP	= 0x03
20:37:18.562 -> CONFIG		= 0x0e
20:37:18.562 -> DYNPD/FEATURE	= 0x00 0x00
20:37:18.608 -> Data Rate	= 1 MBPS
20:37:18.608 -> Model		= nRF24L01+
20:37:18.654 -> CRC Length	= 16 bits
20:37:18.654 -> PA Power	= PA_LOW
20:37:18.701 -> ARC		= 0
20:37:18.701 -> 000000000000000011111111111111112222222222222222333333333333333344444444444444445555555555555555666666666666666677777777777777
20:37:18.840 -> 0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF0123456789ABCD
20:37:24.558 -> 010112176666250000001000333135460710022120000017890CBB988AB9751002125577800883002000000000000000000000000000000000000000000000
20:37:30.313 -> 013000124666656544311000131697A821321104000000020000012469AADDCDFE966432200010002000000000010000000000000000000000000000000000
20:37:36.065 -> 0011010756531200013350003345677B2464333430000039A409987656512222255678AAB00440031100000000000000000000000010000000000000000000
20:37:41.831 -> 0010356777776732100010029BAFCFCA4BB9B9C8910000011001445779CEFFCFEE754311000000011000100000000000000000000000000000000000000000
20:37:47.590 -> 000000143210001245766000439BBE972565423440000028850464320111133456579ABBB00110000100000000000000000000000000000000000000000000
20:37:53.358 -> 001146666677470000000001234898641213533430000001100478AACFFEFCAB96100000100240001100000000000000000000000000000000000000000000
20:37:59.074 -> 0111000200023456667650107567AA7500334635200000056003100011335577A7CABA88801000021000000000000000100000000000000000000000000000
20:38:04.835 -> 021045476777360010000000033634250432233311000001400899ABAFEDC93430101235700050010000000000000000000000000000000000000000000000
20:38:10.593 -> 0100000012356666654330005466769A34267432200000011000013236989AACDDBA9775400000012000000000000000000000000000000000000000000000
20:38:16.316 -> 0011321366662400100120001222234512000000000000077209BA986B98540110223467800480012000000000000000000000000000000000000000000000
20:38:22.058 -> 0111022467777653210000004676A88A1657557580000000200244566789BDCFFF875443200000010000000000000000000000000000000000000000000000
20:38:27.814 -> 0110001777531311234540000004656523120302301110069206765436433234867779CCC00242112100000000000000000000000000000000000000000000
20:38:33.568 -> 020236566665462000000001867CBCCD6775163862000001010256789BCFFFDEBC844111101030010200000000000000000000000000000000000000000000
20:38:39.328 -> 001100142210124455675001046B7758334200545100000690034321001013466789AACBA00022022000000000010000000000000000000000000000000000
20:38:45.093 -> 001256655566071000000000328AB696142333342100011351067CBBCDEDECBB96200001100131001000000000000000000000000000000000000000000000
20:38:50.814 -> 001100131124576666654000145867874845122120000002500200001345779ABBCCCA98700010000000000000000000000000000000000000000000000000
20:38:56.585 -> 0000444666876710000010000027322225310010000000026008ABC9ADCCA94422001234500233000200000000000000000000000000000000000000000000
20:39:02.348 -> 001000123466777532100000021746982A11021320000000100021245567ABBCFEF98765300000001000000000010000000000000000000000000000000000
20:39:08.065 -> 021001066677340000134000020313373564212100000007A20AA9865A85320022545689901230021000000000000000000000000000000000000000000000
20:39:13.836 -> 010013566676553200000000022677AB281021131000000000024445899ABCDEDE766322000012000100000000000000000000000000000000000000000000
20:39:19.563 -> 010000066543001234666010554677754503212230000015B507654213210002345689ABA10653030000000000000000000000000000000000000000010000
20:39:25.334 -> 01022577986656110111212212697879253111121100000132048788BBBDDDABBA420000100120010100000000000000000000000000000000000000000000
20:39:31.100 -> 0200000431013457767750003146987534421212100000016202321000114466788AABBB900140000000000000000000000000000000000000000000000000
20:39:36.817 -> 0001346444342610000000002059976432231025200000047107ACBCCEDECA8984201123500141041100000000000000000000000000000000000000000000
20:39:42.579 -> 000200123334566666431000545BB84737583322400000044101111123468899DCECCB96500000010100000000000000000000000000000111111100000000
20:39:48.341 -> 010012235676270000023001022621121531010001000003410BBBA98A99884320012345700141011100000000000000000000000000000000000000000000

2bndy5, 2021-04-19:
The first 2 lines are the channel value in hex (typed vertically). There's a link at the top of the _scanner.ino_ file that leads the influential code which has color coded explanations.

Looks like channel `0x55` and above are safe bets (at least for your testing environment - actually deployment environment might show different results).

jwilts, 2021-04-19:
Thanks - unfortunately the link in the code brings me to a"Oops! That page doesn’t exist or is private." message.   Again, thank you for the super fast response.

jwilts, 2021-04-21:
Brendan:
After doing a whole bunch of testing over the past few days I can say with certainty that the circuitpython library does not have the same problem on a Pi Zero of Pi3 that I was seeing with the wrapper.  I even added an arduino into the testing mix (multiceiver test) to confirm that inter-species communications works bidirectionally!  (VERY EXCITED to be finally past this problem with the project)

With the revised library I have been able to get thousands of payloads through without any transmission errors.  FANTASTIC!!

For reference these are the steps that I went through to install all the libraries / pre-requisites. 

1.  sudo   apt-get install python3-pip
2.  cd ~
3.  sudo   pip3 install --upgrade adafruit-python-shell
4.  wget   https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py
5.  sudo   python3 raspi-blinka.py
6.  (REBOOT)
7.  sudo   pip3 install adafruit-circuitpython-lis3dh
8.  mkdir   circuit && cd circuit
9.  sudo   apt-get install python3-venv
10.  sudo   python3 -m venv .env
11.  source   .env/bin/activate
12.  sudo   pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@dev
 
Thanks for all your help with this!  Will let you know when we get a full project site created.
(I will submit a few tweaks to multiceiver for your consideration that I found useful for my circumstance)

closed 2021-04-21 by jwilts

2bndy5, 2021-04-21:
> I will submit a few tweaks to multiceiver for your consideration that I found useful for my circumstance

I look forward to this. Heads up though, I've already added the missing `setRetries()` call to the _multiceiverDemo.py_ example. I also slowed down the TX role for that example. Additionally, I've adjusted all the _/examples_linux/*.py_ examples' "Leaving RX role" prompts. All of these changes were committed to the rp2xxx branch (which is slated for PR very soon).