nRF24/RF24 #763: Arduino MKRZero receives wrong values from Pi4 (while Pi4 receives the correct values when they are trasmitted from the MKRZero)

dkts2000, 2021-04-19:
I started using the gettingstarted example with the Pi4 as the receiver and the MKRZero as the trasmitter. After sending and receiving successfully ints and floats alone i successfully sent and received the following struct:

`struct mypayload
{
        float FloatData1;
        float FloatData2; 
        uint32_t Duration1;
        uint32_t Duration2;
        uint16_t Id;  
        uint16_t Data1;
        uint16_t Data2;
        uint16_t Data3;
        uint16_t Data4;
        uint16_t Data5;
        uint32_t Data6;       
};`

I tried it with various random values and i always received the correct values (with the ack enabled).

Then i tried the opposite. Pi4 trasmits the same struct to Arduino MKRZero. I always receive the message but some values are wrong.
Examples: if everything is 0 and the FloatData1 is 1.50 i receive all zeros and FloatData1=1.75; Always.
All zeros and the id is between 0 to 259 i receive it right. if it is 260, i receive 262. And from 260 and over i receive a bigger value.
Also the uint32_t variables. If everything else is zero and i trasmit a value eg. 75000 i receive ~76200 (more or less. this is not constant. but its a multiplier of ~1.016 if i remember right )

Does anyone have an idea what might be wrong?
I should mention that i use the simple NRF24L01+ (not the one with the antenna) and a Socket Adapter Module Board For 8 Pin NRF24L01+ Wireless Transceiver.

TMRh20, 2021-04-19:
Are you using the SPIDEV driver on the RPi4? Are you able to test with A different device other than mkrzero?  I would be tempted to think it’s something with your code, did you just modify gettingstarted to send the struct? Can u post your code?

2bndy5, 2021-04-19:
This might be related to how much memory is allocated for different datatypes on the MKRZero. The MKRZero uses the ATSAMD21 and my experience told me to use `unsigned long` for `uint8_t` on my ATSAMD21 based boards; this is why the multiceiverDemo example uses `unsigned long` to transmit 2 numbers that could easily be `uint8_t`.

Try printing to serial the `sizeof()` some variable and you should see how many bytes are allocated to it. I noticed that a `uint8_t` occupied 4 bytes on my ATSAMD21 boards, but it only occupied 1 byte on the RPi.

dkts2000, 2021-04-19:
Yes SPIDEV.
This is the output of Details on Pi4 (on the MKRZero i can't make it serial print):

`================ SPI Configuration ================
CSN Pin		= /dev/spidev0.0
CE Pin		= Custom GPIO25
SPI Speedz	= 10 Mhz
================ NRF Configuration ================
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x65646f4e32
RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3f
EN_RXADDR	= 0x03
RF_CH		= 0x64
RF_SETUP	= 0x23
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 250 KBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 0
================ SPI Configuration ================
CSN Pin			= /dev/spidev0.0
CE Pin			= Custom GPIO25
SPI Frequency		= 10 Mhz
================ NRF Configuration ================
Channel			= 100 (~ 2500 MHz)
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1500 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Disabled
Auto Acknowledgment	= Enabled
Primary Mode		= TX
TX address		= 0x65646f4e32
pipe 0 ( open ) bound	= 0x65646f4e32
pipe 1 ( open ) bound	= 0x65646f4e31
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6`

**the Pi4 code:** 
`
#include <ctime>       // time()
#include <iostream>    // cin, cout, endl
#include <time.h>      // CLOCK_MONOTONIC_RAW, timespec, clock_gettime()

#include <RF24/RF24.h> // RF24, RF24_PA_LOW, delay()

// custom defined timer for evaluating transmission time in microseconds
struct timespec startTimer, endTimer;
uint32_t getMicros(); // prototype to get ellapsed time in microseconds

void master();

struct mypayload
[... 254 lines cut ...]
23:01:27.538 -> FloatData2=ovf
23:01:27.538 -> Duration1=79612
23:01:27.538 -> Duration2=1834951
23:01:27.538 -> Id=262
23:01:27.538 -> Data1=0
23:01:27.538 -> Data2=15
23:01:27.538 -> Data3=118
23:01:27.538 -> Data4=236
23:01:27.538 -> Data5=255
23:01:27.538 -> Data6=65532
23:01:28.533 -> Received 32 bytes on pipe 1: size=32
23:01:28.533 -> FloatData1=1.75
23:01:28.533 -> FloatData2=ovf
23:01:28.533 -> Duration1=79612
23:01:28.533 -> Duration2=1834951
23:01:28.533 -> Id=262
23:01:28.533 -> Data1=0
23:01:28.533 -> Data2=15
23:01:28.533 -> Data3=118
23:01:28.533 -> Data4=236
23:01:28.533 -> Data5=255
23:01:28.533 -> Data6=65532
`

(For some reason i can't get the code formatting right. I hope you understand it.)

dkts2000, 2021-04-19:
**sizeof on MKRZero:**
`23:28:00.283 -> uint8_t=1
23:28:00.283 -> uint16_t=2
23:28:00.283 -> uint32_t=4
23:28:00.283 -> uint64_t=8
23:28:00.283 -> float=4
23:28:00.283 -> char=1
23:28:00.283 -> unsigned long=4
23:28:00.283 -> mypayload=32
`

**and on Pi4:**
`
uint8_t = 1
uint16_t = 2
uint32_t = 4
uint64_t = 8
float = 4
char = 1
unsigned long = 4
mypayload = 32
`

2bndy5, 2021-04-19:
>```cpp
>  // CAN'T MAKE IT SERIAL PRINT ON MKRZERO
>  // For debugging info
>  //printf_begin();             // needed only once for printing details
>  //radio.printDetails();       // (smaller) function that prints raw register values
>```

This is because the ArduinoCore-samd doesn't support the `printf()` calls that we use internally for debugging. The adafruit fork of ArduinoCore-samd does however.

>For some reason i can't get the code formatting right. I hope you understand it.

Please use "fenced" code-blocks. For example:
>\```cpp
// your code here
\```

The "cpp" is optional and only does syntax highlighting.
The "insert code" button only works for inline \`code-blocks\`.

>I should mention that i use the simple NRF24L01+ (not the one with the antenna) and a Socket Adapter Module Board For 8 Pin NRF24L01+ Wireless Transceiver.

About this adapter board you're using: If the jumper wires are long (by "long"  I mean anything more than 1 inch) or the connections are somehow loose, then lower the SPI speed by specifying it in the c'tor.
```cpp
RF24 radio(6, 7, 4000000); // uses 4MHz for SPI speed (default is 10MHz)
```
This advice is a bit of a hack though. It would be best to use short jumpers and connections that are as tight as possible.

2bndy5, 2021-04-19:
BTW, your `float` is outputting like "ovf" because [this snippet is the code called to output a float](https://github.com/arduino/ArduinoCore-API/blob/e2d2f205094ffa46b7dd31e38a93785960de8604/api/Print.cpp#L335-L345). But that only confirms your received data is somehow getting corrupted.

TMRh20, 2021-04-20:
Yeah, the code works fine between RPi4 and Arduino Due, so this is an odd one. 
Since the radios check the CRC, the data should be good there, its either getting corrupted at the RPi4 before sending or after receiving. I would be very interested to know if this code can be tested with RPi4 to another device other than MKRZero. This would point out where the data is getting corrupted.

dkts2000, 2021-04-20:
Well, the same code on a Leonardo works fine. Receives the values correctly. So what's wrong with MKRZero? Could it be something regarding bit order?

2bndy5, 2021-04-20:
>Could it be something regarding bit order?

I think we would've run into endianess problems on the ATSAMD21 a long time ago. I have several ATSAMD21 based boards, and the only problem I ever had with them is with using long jumper wires. However, with my code I was able to look at the whole payload in hexadecimal and binary representation.

I agree with TMRh20, this has to be a corruption at either of the radio modules' SPI connections (SCK, MOSI, MISO). It is strange that the erroneous data is only observed on the MKRZero board.

>Well, the same code on a Leonardo works fine. Receives the values correctly

Are you using the same wires and the same adapter module?

dkts2000, 2021-04-21:
No. Different wires. Different 8 pin Socket Adapter Module but the same NRF24L01+ module (and also a different NRF24L01+ which worked fine as well).
I'll try using the same wires and modules from the MKRZero and also a different pair on it.
The wires are the 10cm ones. I don't have smaller.

Also (if this helps) both on Leonardo and the MKRZero i connect the power to the 5v pin (not directly from the NRF24L01+ but from the socket adapter which i read has a 3.3V regulator). If i connect it to arduino's 3.3V pin they both fail receiving (and transmitting) any data (Leonardo and MKRZero).
On the RPi4 side the power goes to a 3.3V pin with no problems (also from a 8 pin socket adapter).

The Leonardo is powered from a 9V external power supply and the MKRzero from a breadboard power supply module (which outputs 5V) to Vin which (breadboard power supply) is powered from the same 9V external power supply. Only one arduino runs at a time.

dkts2000, 2021-04-21:
You were right. It was the wires after all! I replaced all the wires and now it works, although the connections are very sensitive. With a slight move it starts receiving weird values again. If i leave it in a state that receives right values, it's steady.

Thank you both for your interest and time.

closed 2021-04-21 by dkts2000

2bndy5, 2021-04-21:
@dkts2000 you could also use a lower SPI speed for more reliability
```cpp
​RF24 ​radio​(​6​, ​7​, ​4000000​); ​//​ uses 4MHz for SPI speed (default is 10MHz)
```