nRF24/RF24 #764: Sending data from multiple

rbp9802, 2021-04-25:
Hello, i'm developing an acceletometer sensor network (Arduino) and i'm having problem receiving data on the master node (Raspberry Pi), slaves are identified sensing an ID byte on the message and send data at the same time to the master and data is being lossed (from what i have research this is not ideal [](https://forum.arduino.cc/t/nrf24l01-multiple-pipes-and-addressing-questions/515084/4 )) but i don't have other options since saving data on SD (to send it later) is too slow and power consuming for my porpose. I also tried sending data from different pipes but data is recieved at any pipe, not specifically at the one i send it to.

Are ther any way to avoid data losses? Hope you can help me, thank you for your time. 

Here is part of my code:

```
// SPI/I2C Library
#include "SPI.h"
#include <Wire.h>

// NRF24L01+ Library
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"
const int CS_RF = 8;  // NRF24L01+ CS pin
const int CE_RF = 7;  // NRF24L01+ CE pin
RF24 radio(CE_RF, CS_RF); // CE, CSN
byte const ID = 1; 

typedef struct{     // Paquete de datos que envía RF
  const int8_t ID1 = ID; // ID sensor
  const int8_t ID2 = 1; // ID datos
  uint16_t c = 1;
  int16_t AX;       // X Acceleration
  int16_t AY;       // Y Acceleration
  int16_t AZ;       // Z Acceleration
  unsigned long t_acc; // time in millis
  }
myData;
myData data;

//times
unsigned long t;
unsigned long t_refAcc;

////////////////////////////////////////////////////////////////////////////////////

void setup() {
  Serial.begin(115200); 
  Wire.begin();

  RadioSetUp();
  Acc_SetUp(); // Accelerometer Set Up
  Acc_config();
  Acc_PrintDetails();
  radio.stopListening();
  t_refAcc = millis();
}

void loop() {
  t = millis();          // Current Time
  if (t - t_refAcc > 19){      //50 Hz readings T = 20 ms
    Acc_Read();
    data.t_acc = millis();
    t_refAcc = t_refAcc + 20;
    Acc_RFSend();
    }
}
void RadioSetUp(){ 
  const uint64_t pipes[2] = { 0xE8E8F0F0E1LL, 0xE0F0E0F0E1LL };
  if (radio.begin()){
    Serial.println("RF started");
  }
  else {
    Serial.println("fail to start RF");
  }
  radio.setPALevel(RF24_PA_HIGH);
  radio.setChannel(0x76);
  radio.openWritingPipe(pipes[1]);
  radio.openReadingPipe(1, pipes[0]);
  radio.enableDynamicPayloads();
  //radio.setPayloadSize(14);   // don't now why setting payload size to 14 on slave and master makes fail writing data
  radio.setCRCLength(RF24_CRC_8);
  radio.setDataRate(RF24_1MBPS);
  radio.setAutoAck(1);
  radio.setRetries(0, 15);
  radio.stopListening();

  printf_begin();
  Serial.println(" ");
  radio.printDetails();
  }

void Acc_RFSend(){
  // Option 1
    if (radio.write(&data, sizeof(data))){
    }
    else{
       Serial.println("RF fail to send data.");
    }

// Option 2: work same as option 1
//  radio.writeFast(&data, sizeof(data));
//  if (!radio.txStandBy(20)){
//    Serial.println("RF fail to send data.");
//  }
}

```

Raspberry Pi

```
import numpy as np
import RPi.GPIO as GPIO
from lin_nrf24 import NRF24
import os
import time
import spidev
import datetime

GPIO.setmode(GPIO.bcm)
radio.begin(0,17)
radio.setPayloadSize(32)
radio.enableDynamicPayloads()
radio.setCRCLength(NRF21.CRC_8)
radio.setChannel(0x76)
radio.setRetries(0,0)
radio.setDataRate(NRF24.BR_1MBPS)
radio.setAutoAck(1)
radio.enableAckPayload()
radio.printDetails()

A1 = np.empty((0,7), int)
A2 = np.empty((0,7), int)

t = int(time.time()*1000 )  ##t=millis()
t_ref = t
while t - t_ref < 30000: ##read for 30 seconds
    t = int(time.time()*1000 )  ##t=millis()
    if radio.available():
        r=[]
        radio.read(r, radio.getDynamicPayloadSize())
        ##           ID1   ID2   c         AX uint_16t   AY uint_16t   AZ uint_16t,   t_acc unsigned long
        b = np.array(r[0], r[1], r[2]+r[3]*256, r[4]+r[5]*256, r[6]+r[7]*256, r[8]+r[9]*256, r[10]+r[11]*256+r[12]*2**16+r[13]*2**32)
        if b[0] == 1: ##separate IDs
            A1 = np.vstack(A1,b)
        elif: b[0] == 2:
            A2 = np.vstack(A2,b)
        else:
            pass
   else:
       pass

```

As i'm sending data every 20ms (50Hz) i should receive 1500 messages from each sensor every 30 seconds on raspberry but i'm receveing from 1400 to 1500 which affects data processing

2bndy5, 2021-04-25:
You could be suffering from ambient interference (use _scanner.ino_ to find the interference to avoid). You could also have a look at how the [_multiceiverDemo.ino_ example uses the `radio.setRetries()`](https://github.com/nRF24/RF24/blob/0b60c0b457ac75c60fbe9f9a508223cab703509a/examples/MulticeiverDemo/MulticeiverDemo.ino#L196) to avoid packet collision in air.

rbp9802, 2021-04-26:
Thanks Brendan. I set `radio.setRetries(ID, 5);` and it improve very much, unfortunatly i'm still missing some data. Also, receiver is getting some messages twice. 

I set retry delay to ID so i can put more transmitters. is it better to increase delay between transmitter? like in the example ``radio.setRetries(ID*3, 5);`. I found as i increase the delay i started missing more data as i only have 20 ms for the hole process. 

For the moment i have not used differents pipes for each transmitter as it did not work well with the same retries set. Will this improve data reception?

2bndy5, 2021-04-26:
> I set retry delay to ID so i can put more transmitters. is it better to increase delay between transmitter?

Sounds like you should read [the docs about the `setRetries()` function](https://nrf24.github.io/RF24/classRF24.html#a4c6d3959c8320e64568395f4ef507aef). Using that function is meant to stagger transmissions among multiple transmitters.

> For the moment i have not used differents pipes for each transmitter as it did not work well with the same retries set. Will this improve data reception?

Think of pipes as parking spots. You don't get any speed benefit from choosing a different pipe. Any reception affect from changing pipes would only be directly related to the address set for that pipe.

> My plan is to have 10 sensor (at least) sendind data

This radio can only listen up to 6 devices at once.

rbp9802, 2021-05-15:
Hi Brendan, thank you for your reply. I've try to configure the radio without success (testing on 4 sensors). This is whats i've done:
 - First, I set address an retries as in multiciever example . 
 - I also change `radio.setCRCLength(NRF21.CRC_8)` to `radio.setCRCLength(NRF21.CRC_16)` and improve a lot
 - I double struct size so i can put 2 samples per struct and write data in half of frequency, so at each instant i have 2 sensor sending data instead of 4.

As result, i'm still missing some samples depending on the writing frecuency and sensing time, for example sensing data from 4 sensors at 60Hz works pefectly for the first 2 minutes and then start missing data (transmission began to be coupled?), same with sending data at 100Hz, works only for 30 seconds an then started missing data.

This only work when setting data rate to 2mbps and failed at 1mbps and 250kbps, i would appreciate if you could tell me why as my sending/receiving data rate is far below (i think).

On the other hand, i´ve been thinking in adding an external flash memory to the nano board so i can send the data after sensing it which lead me to become interested in the new Raspberry Pi Pico that has 2MB flash (enough for sensing 20 to 30 minutes in this case). I've found a [driver](https://github.com/micropython/micropython/blob/master/drivers/nrf24l01/nrf24l01.py) for the module  and recently Arduino IDE include the board to their list. I've also read you are adjusting the library to work with RPi Pico, which i would like to use. 

So, is this library available to use with RPi Pico on Arduino IDE? I would be grateful to read your comments (sorry if i use the wrong words to explain myself, hope you understand)

2bndy5, 2021-05-15:
Have you looked into my suggestion about avoiding ambient interference (using [`radio.setChannel()`](https://nrf24.github.io/RF24/classRF24.html#a5e6e5a5f6c85d2638381cab2c0f3702e))?

The fact that 1Mbps and 250kbps fail is strange as hell. Usually 250kbps would fail if your radio module is not a plus variant ([radio.isPVariant()`](https://nrf24.github.io/RF24/classRF24.html#a62846750b82682beb7593719eb60ed60)). But 1Mbps has always been reliable for me, so I have no idea what to say to that.

>I've found a driver for the module and recently Arduino IDE include the board to their list.

Yes. They added the RPi Pico to MBED OS support and then extended/ported that MBED support to the ArduinoCore-mbed repo (it has to be installed via board manager in the IDE). I personally don't actually own a board that the ArduinoCore-mbed supports, so I can't testify as to weather or not the RF24 libs work for those boards yet. I have a couple Adafruit variants that employ the rp2040 chip (all Adafruit boards using the rp2040 have a 8MB flash storage), but to test the RF24 libs for those boards under the Arduino IDE, I'd have to use a 3rd party Arduino core (though it wouldn't be official because Adafruit hasn't released their own Arduino IDE support for their rp2040 based boards).

The driver you hyperlinked to is for MicroPython, and it is only meant as a proof of concept (not a deployable library). Personal experience tells me, "you can't beat the speed of C++" (which directly affects transmission performance).

>On the other hand, i´ve been thinking in adding an external flash memory

I like your idea to workaround the data degradation using a cache on flash memory. It would pollute the RF spectrum less for scenarios like using the RF24Network lib. 

Flash storage usually uses a form of SPI called QSPI (which is faster than normal SPI because it uses multiple data lines), but QSPI may occupy an SPI bus just for the flash storage; this really depends on the board manufacturer.

2bndy5, 2021-05-15:
Forgot to mention that the rp2xxx branch on this RF24 repo is _mostly_ stable, but we're still working out the CMake integration (for Linux and the PicoSDK). I say "mostly" because the manualAcknowledgement.cpp example spots out a false positive result for the RX role in the rp2040 based board. Although, this experience may be specific to my radio modules and/or wiring/setup.

RF24Network and RF24Mesh also require a patch to integrate those libs into the PicoSDK build system (for CMake compatibility).

rbp9802, 2021-05-16:
Thanks again for your reply!

> my suggestion about avoiding ambient interference

I've change channel once but i'll try with more options

> Usually 250kbps would fail if your radio module is not a plus variant

> "you can't beat the speed of C++" 

i'm using [E01-ML01DP5 module from ebyte](https://www.ebyte.com/en/product-view-news.html?id=109) which should work at 250kbps perhaps i´m programming RPi (master) on python 

> but QSPI may occupy an SPI bus just for the flash storage

QSPI is optional? If so it will not work on arduino nano as i intended😢 unless i use softSPI on RF24 but this might be too slow for my porpose i guess. I had in mind using [W25Q128FV 128Mb flash module](https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/?__locale=en&partNo=W25Q128FV) with this [library](https://github.com/Marzogh/SPIMemory).

> all Adafruit boards using the rp2040 have a 8MB flash storage

Nice! found [Adafruit Feather RP2040](https://learn.adafruit.com/adafruit-feather-rp2040-pico), looks promising for my porpose. Once again, I would be grateful to read your comments. I'll keep you informed any progress.

2bndy5, 2021-05-16:
@rbp9802 Thank you for the fine details!

>i'm using E01-ML01DP5 module from ebyte which should work at 250kbps

Coincidentally, someone else recently asked me about using similar PA/LNA module from ebyte with my CircuitPython lib. I posted some preliminary research on [that discussion's thread](https://github.com/2bndy5/CircuitPython_nRF24L01/discussions/25).

**Important**: Note that the power requirements for those ebyte modules tend to be higher than the usual open source PA/LNA modules' requirements (likely because they use a different PA/LNA muxing chip). From the "User Manual" provided by ebyte:
[image]

>QSPI is optional? ... I had in mind using W25Q128FV 128Mb flash module with this library.

It would seem that QSPI commands are optional using that particular flash storage module (I skimmed the datasheet provided by the site you linked). From what I've heard, flash storage modules are _**not**_ drop-in compatible with each other. This is (one reason) why Lady Ada went with an 8MB flash storage chip on all her/Adafruit RP2040 based boards (she usually only uses 2MB) - I also suspect that buying those things in bulk didn't affect the price point either.

>found Adafruit Feather RP2040, looks promising for my purpose

You're in luck! That's the exact board I've been testing the rp2xxx branch with!

2bndy5, 2021-05-16:
@rbp9802 I was reading further into the ebyte user manual, and I found this peculiar info:
>If the communication line uses a 5V level, a 1k-5.1k resistor must be connected in series (not recommended, there is still a risk of damage)

This actually does apply to the Arduino Nano (which uses 5V logic) concerning the SCK, MOSI, MISO, CSN, & CE pins.

Additionally, the user manual specifically says:
>CE pin can be high level for long-term, but it needs to set as POWER DOWN mode when the module write registers, and it is recommended that CE is controlled by MCU pin

which basically means you should call `radio.powerDown()` before your application does anything with the transceiver that isn't transmitting data (like changing channel or setting up the auto-retry feature). The RF24 lib does what it can to satisfy this stipulation, but ultimately the user needs to be aware of it.

rbp9802, 2021-05-16:
Thanks for your prompt response!

> flash storage modules are not drop-in compatible with each other

I lost the thread with that, if it's not a bother, could you explain?

> This actually does apply to the Arduino Nano (which uses 5V logic) concerning the SCK, MOSI, MISO, CSN, & CE pins.

Found [this](https://forum.arduino.cc/t/3-3v-spi-bus-with-arduino/154407/8) for resistors, perhaps, are logic level converter recommended or too slow/power consuming? 

> There is a lot of places where the user manual insists the power supply be "well grounded" and "stable" (which begs the use of capacitors)

Found 10uF a common value for nRF24L01, this will work with E01-ML01DP5 module power consumption? With this adaptations it will be spected to work ar 250kbps or is a separate issue?

2bndy5, 2021-05-16:
>>flash storage modules are not drop-in compatible with each other
>
>I lost the thread with that, if it's not a bother, could you explain?

SPI commands used for 1 flash storage module aren't guaranteed to work on a different flash storage module. Also, the pin arrangements could be different depending on what the manufacturer decided.

>Found this for resistors, perhaps, are logic level converter recommended or too slow/power consuming?

First, I would try what the manual says and attach a resistor in series with the radio module. I can explain what "in series" means if you want me to. Personally, I'd use a 1.5 k-ohm on each line because that's what I have lying around and it falls within the suggested range of resistance values. The resistors are just there to consume excess voltage, so this shouldn't slow anything down. Same goes for logic level converters: they won't slow anything down, and would likely consume some amperage to operate (you need as much amperage as you can get for the PA/LNA modules). 

>Found 10uF a common value for nRF24L01, this will work with E01-ML01DP5 module power consumption?

This really depends on the power supply being used. 2 identical boards can have the same voltage regulator on them, but the nature of transistors dictates that the efficiency can vary - meaning 1 board's power supply could be noiser (this is often also called efficiency or ripple) than the other identical board's power supply. It won't hurt to try using more capacitance if you don't have a oscilloscope to see how much the voltage varies from the power supply.

rbp9802, 2021-05-16:
Thanks for your explanation! 

> voltage varies from the power supply.

Ooopss. i've been using 3v3 output pin from arduino following common nRF24L01 tutorials 😆. until now it has worked with 2mbps and PA level max using this pin (which supports 50mA(?)). I've just switch to [18650 UPS module](https://www.amazon.com/-/es/Hvlystory-ESP32S-ESP32-cargadores-Arduino/dp/B07M9419Y5) and started working with multiple sensors at 250kbps. I have and [INA219 adafruit module](https://learn.adafruit.com/adafruit-ina219-current-sensor-breakout) but i can only measure RX current (24mA) and not instant TX current so i can verify current consumption, sleep current was 2mA (don't now if this measure is real beacuse even if nothing it's connected INA219 measure 2mA). 

Until now it has work with 4 sensors for now, but still missing data as acumulated time from retry attempt is more than sampling time. Anyway i'll try resistors for SPI and capacitor to make it more "healthy". Perhaps, i'm still planning to move to RPi Pico or Adafruit Feather RP2040.... or external flash memory.  

I'll keep you update if you're interested. Thanks again for you help, I really appreciate it!!

rbp9802, 2021-05-20:
Hello Brendan, just got a pair of RPi Pico which i'm trying to program with arduino IDE. I have installed Arduino IDE official support and [3rd party support](https://github.com/earlephilhower/arduino-pico) and test your [getting started code](https://github.com/nRF24/RF24/blob/rp2xxx/examples_pico/gettingStarted.cpp)  without success . I imported [RF24-rp2xx](https://github.com/nRF24/RF24/tree/rp2xxx)  as a new library but i don´t really know what to do with the files and how to use them (I don't really undestand the different between `.cpp` files and `.ino` files.  `defaultPins.h` uses pin number or GPIO pin number?) What would be a general procedure to use this library on a RPi Pico board?

I connected RF24 module like following;
```
| RPi Pico pin | GPIO | RF24 |
| 6            |  4   | MISO |
| 10           |  7   | MOSI |
| 9            |  6   | SCK  |
| 19           |  14  | CE   |
| 20           |  15  | CSN  |
| 38           |  GND | GND  |
| 36           |  3V3 | VCC  |
```

Hope you can help me with this (again) 🙏

2bndy5, 2021-05-20:
I should've been more clear. The cpp files are specifically for using only the PicoSDK - not an easy task for beginners because of the absolute granular control over everything including how the device enumerates when plugged into the PC. If you have the actual RPi Pico board, then you can install Arduino-MBED support (from the Arduino IDE's board manager), and then run the RF24 examples as you would on any other board. The same should go for the 3rd-party Arduino-pico core: After you get it installed in the Arduino IDE, you should be able to run the RF24 examples from the Arduino IDE's File->Examples->RF24 menu.

To run the cpp files in the _examples_pico_ folder (which are not designed for Arduino IDE), you need to build them with CMake and have the Pico SDK saved in a directory adjacent to the RF24 lib directory. To hasten this, you can download the [RP2xxx build's workflow artifact for the pico board](https://github.com/nRF24/RF24/suites/2737767927/artifacts/60470233), and just upload the UF2 file to your pico board while it is in bootloader mode, but be warned that the pins are assumed to be the default `spi0` pins defined fdor that board (I think this is stated in the _default_pins.h_ file you mentioned). More info about using the Pico SDK with RF24 lib can be found in the [/docs/pico_sdk.md file (on the rp2xxx branch)](https://github.com/nRF24/RF24/blob/rp2xxx/docs/pico_sdk.md).

### about what pins to use
This seems to have changed recently with the 3rd-party arduino-pico core. Thankfully, Earl also updated the docs [about using different pins for peripheral interfaces](https://arduino-pico.readthedocs.io/en/latest/pins.html#spi-spi0-spi1-spi1). See also his [docs about using the SPI or SPI1 classes](https://arduino-pico.readthedocs.io/en/latest/spi.html) As for the default SPI pins used, these are defined in the [arduino-pico core's variant directory](https://github.com/earlephilhower/arduino-pico/tree/master/variants) (select the board you have and look at the `PIN_SPI0_*` pins defined). I'm not sure if these numbers are supposed to be the numbers printed on the board, but my guess is that the numbers defined in the respective _pins_arduino.h_ files are supposed to be the GPx pins displayed on the [RPi Pico's pinout chart](https://www.raspberrypi-spy.co.uk/2021/01/pi-pico-pinout-and-power-pins/#prettyPhoto/1/).

I have no knowledge about using the Arduino-MBED core because it is so much more limited than the 3rd-party arduino-pico core (this support was literally released by Arduino folks about a month ago and it is rather incomplete at this time).

rbp9802, 2021-05-20:
I tried uploading the [getting starded example](https://github.com/nRF24/RF24/blob/master/examples/GettingStarted/GettingStarted.ino) on both, Arduino-MBED support and 3rd-party Arduino-pico core and got compilation erros in both cases (i'm using [Arduino IE 1.8.15](https://www.arduino.cc/en/software)): 

for the  Arduino-MBED support:
```
In file included from C:\Users\X\Documents\Arduino\libraries\RF24\gettingStarted.cpp:13:0:
C:\Users\X\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.1.0\cores\arduino
   /mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: 
   fatal error: pico/stdio.h: No such file or directory
 #include "pico/stdio.h"
          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compilando para la tarjeta Raspberry Pi Pico.
```

for earlephilhower support:
```
:\Users\X\Documents\Arduino\libraries\RF24\interruptConfigure.cpp:45:6: error: ambiguating new declaration of 'bool setup()'
   45 | bool setup()
      |      ^~~~~
In file included from C:\Users\X\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.5.1\cores\rp2040/api/Interrupts.h:8,
                 from C:\Users\X\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.5.1\cores\rp2040/api/ArduinoAPI.h:29,
                 from C:\Users\X\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.5.1\cores\rp2040/Arduino.h:30,
                 from C:\Users\X\Documents\Arduino\libraries\RF24/RF24_config.h:69,
                 from C:\Users\X\Documents\Arduino\libraries\RF24/RF24.h:18,
                 from C:\Users\X\Documents\Arduino\libraries\RF24\interruptConfigure.cpp:19:
C:\Users\X\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.5.1\cores\rp2040/api/Common.h:115:6: note: old declaration 'void setup()'
  115 | void setup(void);
      |      ^~~~~
C:\Users\X\Documents\Arduino\libraries\RF24\interruptConfigure.cpp: In function 'int main()':
C:\Users\X\Documents\Arduino\libraries\RF24\interruptConfigure.cpp:360:18: error: could not convert 'setup()' from 'void' to 'bool'
  360 |     while (!setup()) { // if radio.begin() failed
      |             ~~~~~^~
      |                  |
      |                  void
C:\Users\X\Documents\Arduino\libraries\RF24\interruptConfigure.cpp:360:18: error: in argument to unary !
```

Same goes to other examples. I also include this lines in the setup code to set up SPI0 pin:
```
RF24 radio(6, 5);
void setup(): {
  ...
  SPI.setRX(4);
  SPI.setTX(3);
  SPI.setSCK(2);
  SPI.setCS(5);
  }
```

Also try `.uf2` with default pins but i didn't found out which CE pin it's considered. Looks like it would be easier (for the moment) to program with c++ and build them as you mentioned. Thank you!

2bndy5, 2021-05-20:
Stop using the cpp files in the arduino IDE. Use the ino files in the examples folder for the arduino IDE.

rbp9802, 2021-05-20:
yes, i used [GettingStarted.ino](https://github.com/nRF24/RF24/blob/master/examples/GettingStarted/GettingStarted.ino)

2bndy5, 2021-05-20:
> C:\Users\X\Documents\Arduino\libraries\RF24\gettingStarted.cpp:13:0 

This means you are somehow using the cpp files. The should be no cpp files in the directory where the ino files are. The arduino IDE will automatically include any other c++ compatible files it finds with the ino file.

rbp9802, 2021-05-24:
Thanks for your explanation Brendan! Finally, I moved all `example.cpp` files from RF24 library folder and upload the sketch succesfully. Although i was not able to print details.  I used the following pins, include `printf.h` and call `printf_begin();` I found  [this related issue](https://github.com/nRF24/RF24/issues/414). Does this function works on RPi Pico (or RP2040) at the moment? Are there other ways to 'manually' get details?

Pins
```
GP16 SPI0 RX  --> MISO
GP17 SPI0 CSn --> CSN
GP18 SPI0 SCK --> SCK
GP19 SPI0 TX  --> MOSI
GP20          --> CE
```
Radio
```
RF24 radio(20, 17);
```

2bndy5, 2021-05-24:
> Does this function works on RPi Pico (or RP2040) at the moment?

From what you just described, no - probably not. The `printDetails()`, `printPrettyDetails()`, and `errNotify()` functions don't work on all platforms. The current solution implemented was only meant to work on AVR and Linux platforms (but it has been patched for a few other capable platforms). We are currently working on a logging library that should address that problem, but it isn't ready for release yet. 

> Are there other ways to 'manually' get details?

There are some getter functions in the RF24 API, but they won't provide all the data output (like the STATUS byte and addresses assigned to the pipes) by `printDetails()`.

For a more immediate fix, you'd have to look at the Arduino core's code and find out what is needed to use _pgmspace.h_ and `printf()`. Instead of doing this for each platform, we have chosen to start a new RF24Log lib to achieve this for all platforms.

These functions do work when used with only the Pico SDK (and not the Arduino IDE).

TMRh20, 2021-05-25:
Also per pins_Arduino.h the default pins are:
Miso: 4
Mosi: 3
Sck: 2

Can and ce are user specified.

2bndy5, 2021-05-25:
@TMRh20 There are different _pins_Arduino.h_ definitions for different rp2040 board variants (depending on the arduino core being used). The SPI lib allows optionally specifying the CSN pin for either the official ArduinoCore-MBED (for RPi Pico - personally not recommended) and the 3rd party arduino-pico core (by EarlePhilhower for all RP2040 based boards - personally recommended). The current implementation in RF24 does not make use of this feature, so you are right - they are user-specified. I doubt there is any performance improvement when using the SPI lib to manage the CSN.

@rbp9802 I recently installed the 3rd-party arduino-pico core and ran GettingStarted.ino (with no modifications) on my Adafruit Feather RP2040, and it worked. I can also confirm that `printf()` does not work on the RP2040 because it uses ArduinoCore-API (which is the same problem for many of the newer arduino cores since 2016). `spintf()` could be used instead, but it would be more like a hack (rather ugly under the hood).

2bndy5, 2021-05-25:
Just an update about trying to implement `sprintf()` in lack of `printf()`: I tried the following code snippet and it partially worked.
```cpp
// in printf.h
#if defined (ARDUINO_API_VERSION)
#include <avr/pgmspace.h>
#include <stdarg.h>
int printf(const char *fmt, ...)
{
    va_list args;
    va_start(args, fmt);
    char printf_buf[80] = {0};
    int n = sprintf_P(printf_buf, fmt, args);
    Serial.print(printf_buf);
    va_end(args);
    return n;
}
#endif
```
it outputs the `const char*` data in the `fmt` parameter fine, but it fails to de-reference the packed arguments parameter.
[image]

I also tried using `#define PRIPSTR "%S"` macro, but it didn't help much. I'm going to abandon this idea and continue developing the RF24Log lib (where we implement our own printf-parsing 🥇 ).

2bndy5, 2021-09-09:
This issue got a bit off track after talking about switching to the RP2040-based boards. Is sending data from multiple radios still experiencing problems?

closed 2021-12-01 by 2bndy5