nRF24/RF24 #766: Raspberry 3B+ receiving nonsense from Arduino Nano (other way around works!) [Question]

Fabzi, 2021-04-27:
Hey Everybody!

I am trying to create a wireless connection between an Arduino Nano and a RPi 3b+, by using RF24L01+PA+LNA modules.
Nearly everthing is working fine, except receiving data on the RPi. I am using the getting_started (ino and py) files on each site.
[image]
As you can see in the image above sending from the Arduino works fine, but the RPi is receiving complete nonsense. I also reversed the connection and let the RPi transmit and the Arduino receive which worked fine. An other thing I tested was the same setup with 2 Arduinos (1 Nano and 1 Uno) and this worked without any issues.  At the moment I am powering the modules via the pins of the micro controllers (also tried adapter andI know it's stupid because the peak current could exceed the current cap of the arduino, but the 2 Arduino setup worked the same way and did well). I also think it's not an elctro-magnetic problem due to the fact that the Arduino-only setup didn't had any shielding as well.

Fabzi, 2021-04-28:
Update!

I found a fix! I reduced the SPI-speed of the RPi (from 10MHz to 4MHz).
`radio = RF24(22, 0, 4000000)`
Now everthing is working as it should. Still don't know how this fixed the problem, but I think I have to accept it.

closed 2021-04-28 by Fabzi

Avamander, 2021-04-28:
Cool, thanks for letting us know that solved it. 

It's very likely some signal integrity issues, long and/or wires? I'm sure there's also guides online about what/if some pull-ups or pull-downs are required on RPi SPI.

Fabzi, 2021-04-28:
My first though about this problem was that the wires are too long or damaged. But I used the same wires in the Arduino setup (even for the same pins), beacause of this I elimated the though.

Due to my foreknowledge of the RPi and some projects with I²C on the RPi I am pretty sure that pull-up / pull-down resistors are provided at the board for any kind of serial interface. Usually the Arduino doesn't provide the required resistors, hence I think that the nRF24 module itself has some pull-ups / pull-downs installed.

Still thanks @Avamander! Giving it a though even if the issue is closed!!