nRF24/RF24 #768: [Question] Get received message Packet Control Field

VasilyRakche, 2021-04-28:
Hey,

I was comparing messages received with nRF and with other device. I see that the Packet Control Field is cut-off from the message. Based on the [nRF24 Datasheet](https://raw.githubusercontent.com/nRF24/RF24/master/datasheets/nRF24L01P_datasheet_v1.pdf) this field is part of Enhanced ShockBurst, 
[image]
but is not present in regular one:
[image]

So is there a way to get Packet Control Field content, or if there is a way of just switching to regular ShockBurst protocol?

2bndy5, 2021-04-28:
Only the payload is accessible to the user. All other parts of the packet aren't exposed through some API. You can calculate the CRC yourself...

The difference between Enhanced ShockBurst and regular old ShockBurst is only the use of the auto-ack feature. Nordic semi. has a history of making hardware as backward compatible as possible, that's the only reason auto-ack feature can be turned off.

VasilyRakche, 2021-04-30:
Got it, but is there still a chance to get **Packet Length**, **Packet ID**, or if the **ACK flag was set** from Packet Control Field through API? 
[image]

2bndy5, 2021-04-30:
Sorry there isn't. You can calculate the packet length based on the radio configuration and payload width. You might be able to estimate the Packet ID from the TX ARC after a transmitting the data (info not available at receiver). The NO_ACK bit is directly controlled by the transmitter (asserted when the payload is written to TX FIFO). The radio as a receiver decodes/disassembles the packet as it is received; it does not actually save any artifact or log info containing the data you seek.

pascallanger, 2021-04-30:
@VasilyRakche What you want is simply to configure the NRF in ShockBurst and not **Enhanced** ShockBurst which is done by disabling dynamic payload and auto ack. Then the NRF gives you the full payload without modification.

VasilyRakche, 2021-04-30:
@pascallanger  Thanks a lot, that was it! That way it retains everything inside the packet!

closed 2021-04-30 by VasilyRakche