nRF24/RF24 #769: Receive data after ACK signal send

BlackX-Treme, 2021-05-16:
Hello,
I'm trying to receive some data after sending acknowledge signal, but my network, after sending ack signal, isn't available.
I use an ESP8266 to control my NRF24.
The used code sequence is:
-set auto ack
-open writing pipe
- Verify ack received
- radio powerdown (to reset somehow the nrf, i tried also by sending esp in deepsleep mode)
- radio powerup
- open reading pipe
- start listening
and after all that i check radio available and it fail here.
All this sequence is used in void loop function.

So, i missed something? i did something wrong? RF24 device can't send ack signal and then receive data?

Thank you for helping me.

2bndy5, 2021-05-16:
I appreciate you giving an outline of your program, but it would still help to see the source code. I don't know what you're missing if I don't know exactly what you're code is doing with the library.

BlackX-Treme, 2021-05-17:
> I appreciate you giving an outline of your program, but it would still help to see the source code. I don't know what you're missing if I don't know exactly what you're code is doing with the library.

please check your email

2bndy5, 2021-05-17:
Ok. Here is what you should know (this info is available in the lib docs)
1. ACK payloads require enabling dynamic payloads. Please review the [acknowledgementPayloads example](https://nrf24.github.io/RF24/examples_2AcknowledgementPayloads_2AcknowledgementPayloads_8ino-example.html).
2. Avoid using pipe 0 to receive data. This is mentioned a couple times in the docs (with the reason why to avoid pipe 0 for RX) - see [`RF24::openReadingPipe()`](https://nrf24.github.io/RF24/classRF24.html#a9edc910ccc1ffcff56814b08faca5535).
3. `powerDown()` then `powerUp()` do not perform a soft reset of the radio module. I'm not sure about your intention of doing a soft reset, but the lib does not provide a user accessible way to do this. You might be able to use `begin()` to do this, but either way `begin()` and `powerUp()` wait 5 milliseconds to execute properly.
4. I see you have `stopListening()` commented out in the end of your `setup()` function. To enter TX mode, you need to call `stopListening()` first (preferably after you have opened the pipes you want to use).

Finally, you only sent the code for 1 RF24 node. If `RF24::available()` returns false, then there might also be a problem with the code running on the other RF24 radio module. Just to be very clear, this radio will not communicate with WiFi.

BlackX-Treme, 2021-05-18:
thank you, 
soon i will rewrite the code following your instructions, i will told you if i will success. not the radio module will communicate with wifi, it will be the ESP device. the esp device control the RF device via  SPI. I have an older version of my softwares without ack and everything works fine, i can transmit data from one nrf to another and from rf receiver to website via esp device. 
here is the other device code, also simplified for this issue only.
void setup() { ProMini_Transmitter.begin(); //Start the nRF24 module ProMini_Transmitter.setAutoAck(1); // Ensure autoACK is enabled so rec sends ack packet to let you know it got the transmit packet payload ProMini_Transmitter.enableAckPayload(); //allows you to include payload on ack packet ProMini_Transmitter.maskIRQ(1,1,0); //mask all IRQ triggers except for receive (1 is mask, 0 is no mask) ProMini_Transmitter.setPALevel(RF24_PA_LOW); //Set power level to low, won't work well at higher levels (interfer with receiver) ProMini_Transmitter.openReadingPipe(1,address); //open pipe o for recieving meassages with pipe address ProMini_Transmitter.startListening(); // Start listening for messages attachInterrupt(1, interruptFunction, FALLING); //Create interrupt: 0 for pin 2 or 1 for pin 3, the name of the interrupt function or ISR, and condition to trigger interrupt}
void loop() {  if(pCount < count)     {       //If this is true it means count was interated and another interrupt occurred      Serial.begin(57600); //start serial to communicate process      Serial.print("Receive packet number ");      Serial.print(count);       Serial.print(" has content: ");      Serial.println(gotByte);      pCount = count;                ProMini_Transmitter.openWritingPipe(address);      ProMini_Transmitter.setPALevel(RF24_PA_MIN);                   ProMini_Transmitter.write( /*   my data */);                  Serial.end(); //have to end serial since it uses interrupts    }}

BlackX-Treme, 2021-05-18:
thank you,  soon i will rewrite the code following your instructions, i will told you if i will success. not the radio module will communicate with wifi, it will be the ESP device. the esp device control the RF device via  SPI. I have an older version of my softwares without ack and everything works fine, i can transmit data from one nrf to another and from rf receiver to website via esp device.

2bndy5, 2021-05-18:
@BlackX-Treme Thank you for sending me the other node's code. It looks fine except that it doesn't call [`RF24::enableDynamicPayloads()`](https://github.com/nRF24/RF24/blob/0b60c0b457ac75c60fbe9f9a508223cab703509a/examples/AcknowledgementPayloads/AcknowledgementPayloads.ino#L79) in the `setup()` function. The code doesn't account for the 5 millisecond delay on the esp board, so it could be that the message is sent while the other node (driven by the esp board) is not listening.

I do like your use of the IRQ pin (I don't often see people using that feature)! However, Arduino Docs strongly suggest using `digitalPinToInterrupt()` like this
```cpp
attachInterrupt(digitalPinToInterrupt(1), interruptFunction, FALLING); 
```
instead of 
```cpp
attachInterrupt(1, interruptFunction, FALLING); 
```
Another thing I should warn you about (because I don't know what is happening when your `interruptFunction()` is called): 
1. You can't use any `RF24::write()` function inside your `interruptFunction()` because they use `millis()` to implement a timeout. According to the Arduino docs, `millis()` and `micros()` internally use a counter that is likely inaccurate when the `interruptFunction()` is triggered.
2. `RF24::available()` should not be called before `RF24::whatHappened()` inside your `interruptFunction()` because the data returned is inaccurate during the instant that the IRQ pin is going from HIGH to LOW (a FALLING transition). Calling `RF24::whatHappened()` before calling `RF24::available()` should help avoid this problem.

Here is a helpful [link to the arduino docs about using `attachInterrupt()`](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/)

closed 2021-09-09 by 2bndy5