nRF24/RF24 #770: [Question] SOFTSPI - how?

bifferos, 2021-06-01:
I read in the docs about [SOFTSPI](https://nrf24.github.io/RF24/md_docs_arduino.html):

However I realised setting build flags for individual sketches [isn't really a thing in the Arduino world](https://forum.arduino.cc/t/how-to-set-compiler-constants-required-for-a-library/578762).

Although I can perhaps hack something for my board, or my system in the appropriate Arduino config file.

I have a Nano + NRF24L01 setup which looks like this:
[image]

So I was hoping a combination of:
```
#define SOFTSPI
#define SOFT_SPI_MISO_PIN 12
#define SOFT_SPI_MOSI_PIN 11
#define SOFT_SPI_SCK_PIN 13

#include <nRF24L01.h>
#include <RF24.h>
#include <SPI.h>
```
etc...

...would give me exactly what I had using SPI hardware, but in bit-banged mode.  It didn't compile.
So I tried editing the config file to make the definitions there. That compiled but didn't work.

I read in the RF24 docs on SOFTSPI that it's necessary to include the line in my sketch:
```
#include <DigitalIO.h>
```
But that left me mighty confused because there are these lines in RF24.h:
```
#elif defined SOFTSPI
	#include <DigitalIO.h>
#endif
```
So it looks like it's trying to do this for me.

Would appreciate any clarification you can offer, thanks!

2bndy5, 2021-06-02:
Maybe I'm mis-reading this, so please correct me. The purpose of using SoftSPI is to be able to use non-hardware SPI driven GPIO pins. For instance using a sensor that hogs the SPI pins may require the radio attached to non-SPI pins. I haven't looked at the DigitalIO library source code in a while, but I don't think it was designed to use the hardware driven SPI pins as if they were Software driven SPI pins. What is the reason for wanting to use SoftSPI in your project?

>But that left me mighty confused because there are these lines in RF24.h
>```cpp
>#elif defined SOFTSPI
>	#include <DigitalIO.h>
>#endif
>```
Don't worry about those lines. They are there to save program space when the DigitalIO library isn't included in your sketch. In fact, the DigitalIO library will define `SoftSPI` (as a class - though using `#define SoftSPI` might help) for you, so all you really need to do is alter the `SOFT_SPI_*_PIN` defines in _RF24_config.h_.

So your sketch should start by including the DigitalIO library **before** including the RF24 library. I think this is the clarification that might be missing from the docs. This might also be applicable to the `SOFT_SPI_*_PIN` defines when defining them in the sketch.
```cpp
// assuming the RF24 library's source code is unchanged from latest release
#include <DigitalIO.h>
#define SOFT_SPI_MISO_PIN 16
#define SOFT_SPI_MOSI_PIN 15
#define SOFT_SPI_SCK_PIN 14
#include <RF24.h>
#include <printf.h> // for printf_begin() and printDetails()

// ...
```
But altering the `SOFT_SPI_*_PIN` values & un-commenting `#define SoftSPI` line in _RF24_config.h_ should be the most reliable way when all else fails.

@bifferos can you post the verbose compilation output from your Arduino IDE? I just want to make sure the DigitalIO library is installed properly (& discoverable by the Arduino IDE).
<br><br>
ps - the `#include <nRF24L01.h>` isn't really needed in the sketch anymore (its automatically included by the RF24 library). Every time I see that line, I think, "someone is following an unofficial tutorial".

2bndy5, 2021-06-02:
Wait a minute!!! I just discovered [this line from _RF24_config.h_](https://github.com/nRF24/RF24/blob/0b60c0b457ac75c60fbe9f9a508223cab703509a/RF24_config.h#L93) that shouldn't be there. Seems like I forgot to reove from testing the new overloaded begin() feature in v1.4.0. @bifferos Please remove [line 93 from RF24_Config.h](https://github.com/nRF24/RF24/blob/0b60c0b457ac75c60fbe9f9a508223cab703509a/RF24_config.h#L93)
```cpp
            #define _SPI SoftSPI<SOFT_SPI_MISO_PIN, SOFT_SPI_MOSI_PIN, SOFT_SPI_SCK_PIN, SPI_MODE>
            #define RF24_SPI_PTR // THIS LINE SHOULDN"T EXIST!!
 ```

bifferos, 2021-06-02:
I wanted to report on my experimentation using your library with an Arduino, but I should perhaps explain where I'm going with this.  Ultimately I want to get this running under Linux with some kind of software SPI running over GPIO pins.  Either I could use an [spi kernel module](https://openwrt.org/packages/pkgdata/kmod-spi-gpio-custom) for this or I could bit-bang using the GPIO file interface directly, in user-space. I just decided softspi with the Nano will be a convenient stepping-stone on the way.

I will try those suggested changes, and of course I will post all compiler output if I don't get it working.  Thanks for the info.

2bndy5, 2021-06-02:
> Ultimately I want to get this running under Linux with some kind of software SPI running over GPIO pins. 

Yeah, RF24 lib doesn't currently support SoftSPI on Linux. My first idea to do this is to use the BCM2xxx driver (dubbed "RPi" in the utility folder) to do this natively, but that wouldn't be a great solution for all Linux devices (only RPi SoC devices).

bifferos, 2021-06-02:
I had only a quick look at this at lunch-time but this is the compiler error I get:
```
/home/biff/Arduino/libraries/RF24/RF24.cpp: In member function 'void RF24::read_register(uint8_t, uint8_t*, uint8_t)':
/home/biff/Arduino/libraries/RF24/RF24.cpp:167:18: error: expected primary-expression before '.' token
     status = _SPI.transfer(R_REGISTER | reg);
```
So maybe _SPI is not defined correctly?

I've changed the config file as suggested, here is my code:

```
/*
 TRANSMITTER
*/

#include <DigitalIO.h>
#include <RF24.h>
#include <SPI.h>

RF24 radio(10, 9);               // nRF24L01 (Chip enable,chil select)

// +----+----+----+----+
// |3.3v|CSN |MOSI|IRQ |
// |GND |CE  |SCK |MISO|
// +----+----+----+----+
 

uint8_t addresses[][6] = {"0writ","1read"};
uint8_t count = 0;

void setup() 
{
  SPI.begin();
  Serial.begin(9600);
  radio.begin();   // begin operation of the chip
  radio.openWritingPipe(addresses[1]);    // writing pipe
  radio.setChannel(24);
  radio.setPALevel(RF24_PA_MIN);    // min power level for receivers close together.
  radio.stopListening();
  Serial.write("TRANSMITTING\n");
}

void loop() {
  delay(1000);
  bool tmp = radio.write(&count, 1);
  if (tmp)
  {
     Serial.print("Send OK ");
  } else {
     Serial.print("Send FAILED!!! "); 
  }
  count++;
  Serial.println(count, DEC);
}
```

bifferos, 2021-06-02:
A very minor point about the docs, I guess you need to add angle brackets around the include?
In:
https://nrf24.github.io/RF24/md_docs_arduino.html
Change:
[image]
To
[image]

2bndy5, 2021-06-02:
There are other changes in the [2bndy5-patch-softspi branch](https://github.com/nRF24/RF24/tree/2bndy5-patch-SoftSPI) that I haven't mentioned here (they can be seen in the [PR's "file changes" tab](https://github.com/nRF24/RF24/pull/771/files) - `_SPI` should be defined as `#define _SPI spi` and some other changes). I suggest you download that branch and use it to overwrite the current installation in your
_C:\Users\\%USERNAME%\Documents\Arduino\libraries\RF24_ folder (assuming your on Windows). I tested the branch this way in the Arduino IDE, and it worked for both SoftSPI & HardSPI.

Alternatively, you can use a previous version of this library available from the Arduino IDE's library manager. Either way, the docs should be correct (about using SoftSPI).

I will adjust the docs as you suggested about the angle brackets. That is a good point (I keep adding the brackets automatically).

2bndy5, 2021-06-05:
Heads up. I deleted the 2bndy5-patch-SoftSPI branch yesterday because all is changes were merged into the master branch.

closed 2021-07-02 by 2bndy5