nRF24/RF24 #773: NRF24L01+ Module - Receiver Node is not able to catch Payload

luckyRajputana, 2021-06-05:
**Modules**  - 2 NRF24L01+ Module and 2 Raspberry Pi
**Problem**  - In the Starting I was not able to Transmit Data from Transmitter Node as i was getting _Failed to send or Timeout_. Then i read on some portal to turn off the setAutoAck(false) in off state. After that I was able to Transmit Data from Transmitter Node but Receiver Node was not able to pick the signal and getting False form **radio.isavailable()** every time.
Can you please provide the resolution if you are aware of this issue ?

================ SPI Configuration Receiver ================
CSN Pin         = /dev/spidev0.0
CE Pin          = Custom GPIO22
SPI Speedz      = 10 Mhz
================ NRF Configuration ================
STATUS          = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1    = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5    = 0xc3 0xc4 0xc5 0xc6
TX_ADDR         = 0x65646f4e32
RX_PW_P0-6      = 0x04 0x04 0x04 0x04 0x04 0x04
EN_AA           = 0x3f
EN_RXADDR       = 0x03
RF_CH           = 0x4c
RF_SETUP        = 0x03
CONFIG          = 0x0e
DYNPD/FEATURE   = 0x00 0x00
Data Rate       = 1 MBPS
Model           = nRF24L01+
CRC Length      = 16 bits
PA Power        = PA_LOW
ARC             = 0

================ SPI Configuration ================
CSN Pin         = /dev/spidev0.0
CE Pin          = Custom GPIO22
SPI Speedz      = 10 Mhz
================ NRF Configuration Sender================
STATUS          = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1    = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5    = 0xc3 0xc4 0xc5 0xc6
TX_ADDR         = 0x65646f4e32
RX_PW_P0-6      = 0x04 0x04 0x04 0x04 0x04 0x04
EN_AA           = 0x3f
EN_RXADDR       = 0x03
RF_CH           = 0x4c
RF_SETUP        = 0x03
CONFIG          = 0x0e
DYNPD/FEATURE   = 0x00 0x00
Data Rate       = 1 MBPS
Model           = nRF24L01+
CRC Length      = 16 bits
PA Power        = PA_LOW
ARC             = 15

2bndy5, 2021-06-05:
What exact radio are you using? This sounds like a power supply issue. Please read the [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md) file. It also explains why you get a false positive when disabling auto-ack feature.

luckyRajputana, 2021-06-06:
**Radio Details : ** 2.4GHz NRF24L01+PA+LNA SMA Wireless Transceiver Antenna
Link : [Click Here](https://robu.in/product/2-4ghz-nrf24l01palna-sma-antenna-wireless-transceiver-communication-module-1km/)

FYI - I am using 100µF Capacitor in both Modules and Powering up my RPi with USB.

2bndy5, 2021-06-06:
I thought so. Those modules are very touchy about their power supply. Did you try any of the examples first? And the Raspberry Pi's are known to be power hogs.

>Powering up my RPi with USB

Please tell me you don't mean a USB port on your computer. That just won't do; the voltage tends to sag and there's usually a current limiter on those PC USB ports. There's a reason Raspberry Pi foundation sells "official" power supplies rated at 5.2v.

Avamander, 2021-06-06:
I suspect the 3.3V pin/supply on the RPi might be insufficient anyways given the power requirement of the PA+LNA modules. I'd try a separate linear regulator with 1000µF of decoupling (the datasheet of the regulator might also mention its recommendations) before spending too much time on debugging the code.

luckyRajputana, 2021-06-06:
Yes. I am using **getting started.cpp** example under **examples_linux folder** to test the connection.
Now, I am providing direct power through 5V/2A power Adapter to both my RPis. Still, not receiving the signal at receiver end.

2bndy5, 2021-06-06:
Ok, if your skill level hasn't involved working directly with linear voltage regulators (and reading their corresponding datasheet), then you might want to get some of [the nRF24L01 adapter boards](https://www.amazon.com/Makerfire-NRF24L01-Breakout-Regulator-Wireless/dp/B01IK9GCPE/ref=mp_s_a_1_1?dchild=1&keywords=nrf24l01+adapter&qid=1622985905&sprefix=nRF24L01&sr=8-1) and use the 5v pin to the adapter board (which has a linear voltage regulator built-in). These adapter boards were designed specifically with the PA/LNA modules in mind.

luckyRajputana, 2021-06-06:
Thanks for your quick responses. Ordered the adapter module and will also use 10uF capacitor as per your suggestion.
I will reach out to you in couple of days.

luckyRajputana, 2021-06-12:
Tried as per you suggestion but still getting the same thing. Receiver node is not receiving anything.

2bndy5, 2021-06-12:
Hmm, you might have to use a separate power supply. Remember the general rule of DC electronics: Always tie the GNDs from isolated circuits together (especially when data is traversing a wired connection between the separate power supplies). I'm curious, do you have any USB devices or other GPIO peripherals drawing power from the RPi? (even HDMI draws from the 5v bus)

luckyRajputana, 2021-06-12:
Please find below the images of the whole setup : 

[https://ibb.co/Fhc31yR](url) : Adapter Image
[https://ibb.co/2t97DZr](url) L Setup Image

Question :  _do you have any USB devices or other GPIO peripherals drawing power from the RPi_
Answer : No

2bndy5, 2021-06-12:
ok. well that's a lot of cable... there may be some voltage drop amongst all that copper. I've never had good results when using long wires (even with the non-PA/LNA modules!) Also I can't tell (& you never mentioned) what version of RPi you have. The 2A wall adapter may not be good enough for the RPi4 or even RPi3.

We're getting into the realm of DIY because there are many ways to supply power to the radio separately from the RPi power supply (& I'm not familiar with the circuitry for those "hats" you have installed). All I can say for sure is that you definitely suffer from insufficient current in the power lines to the radio. It might help if you know someone that has an oscilloscope to see what's really happening in the power supply lines.

closed 2021-09-09 by 2bndy5