nRF24/RF24 #777: STM32Duino printDetails on Serialx

yetoo, 2021-07-12:
Hi! 
As I was doing a project with an STM32 BluePill, the Serial was used for another RF module. So I've got only Serial2 or Serial3 to output data with the function _printDetails_ after a lot of search, and going "deep" into the library, I found (I guess) where it might be implemented.
[image]

I just need your opinion about that.

2bndy5, 2021-07-12:
>  I found (I guess) where it might be implemented.

where what might be implemented? can you post the changes you referring to?

If you're asking about changing the `Serial` object used in `Serial.write()`, then I guess you could change it to `Serial2.write(c)` or whatever UART bus object your application is using. Is this what you're asking? Have you tried this?

yetoo, 2021-07-12:
Yes, I know that this a kind of way to do it, but this would be specific to that project (the project I am working on).
I was thinking of something more generic. Some like this :
[image]
And some modifications on the void _printf_begin(void)_ function.

yetoo, 2021-07-12:
> > 
>  Have you tried this?
> 
Not yet but I can try it tomorrow in the office. I wanted to be sure about my changes.

2bndy5, 2021-07-12:
I would suggest you put the new `serial_putc()` in an `#elif` block. 
```cpp
#if defined (ARDUINO_ARCH_AVR) || defined (__ARDUINO_X86__)
// serial_putc() from lib

#elif defined(STM32_CORE_VERSION)
// new serial_putc() from your modification

#elif defined (ARDUINO_ARCH_MBED)
// redirect for MBED core

#endif
```

closed 2021-09-09 by 2bndy5