nRF24/RF24 #778: MKR WIFI 1010 SAMD

Oa373, 2021-07-14:
PRINTF error is still occuring with the SAMD microcontrollers.
F24-master\RF24_config.h:174:41: error: 'class Serial_' has no member named 'printf'; did you mean 'print'?
                 #define printf_P Serial.printf

Is there any solution for this?

2bndy5, 2021-07-14:
the current `printf()` implementation will work with some SAMD boards, but it only worka with adafruit's fork of the arduino-samd core. The MKR-WIFI board is not an adafruit board, so `printf()` shouldn't work on those boards. This is because Arduino's official samd core uses a unified API repo that doesn't support printf().
> F24-master\RF24_config.h:174:41: error: 'class Serial_' has no member named 'printf'; did you mean 'print'?
#define printf_P Serial.printf

this is the full block that it complains about:
https://github.com/nRF24/RF24/blob/7c3d6ecfc55e52ab1cd968ea71d5c86ab8c7ffd4/RF24_config.h#L172-L175
If you're getting that error, then you're likely not using the official arduino-samd core (which defines `ARDUINO_API_VERSION`). The adafruit fork does not use the unified Arduino API repo, thus it doesn't define `ARDUINO_API_VERSION`.

Oa373, 2021-07-14:
So i will not be able to use it with the MKR 1010 right ?

2bndy5, 2021-07-14:
you can use the RF24 library, but you can't get anything printed out from `printDetails()`, or `printPrettyDetails()`. So there's no point in including `printf.h` and calling `printf_begin()`.

I just test compiled 1 of the RF24 examples (with `printf.h` included and `printf_begin()` called) on Arduino IDE for the MKR WIFI1010 board using the Arduino SAMD core, and I didn't get that error.

Oa373, 2021-07-14:
I have  tried the getting started example multiple times , with and without the printf.h . But im still getting the same error

2bndy5, 2021-07-14:
ok I need more details because I just tried to compile the gettingStarted example in PlatformIO, and I still can't recreate that error. 

@Oa373 What steps are you doing to compile the example for the MKR WIFI 1010 board?

Oa373, 2021-07-14:
I am trying to use the mkrwifi1010 to transmit data to an arduino uno . The receiver file compiles perfectly for the Arduino uno . However the transmitter file always shows this error. I tried first compiling one of the examples "getting started" but the examples wont combine whatsoever
[image]

[New Text Document.txt](https://github.com/nRF24/RF24/files/6813116/New.Text.Document.txt)
[image]
 x

Oa373, 2021-07-14:
i have just created the sketch and chose the board and compiled . straightforward process but would still generate this error

2bndy5, 2021-07-14:
check your Arduino board manager for updates. Please don't post just pictures I need to see the entire compiler output - enable verbose compiler output in the Arduino IDE settings
[image]
you can select the text then use ctrl+c to copy it, then paste it here). Seeing the same error over and over again doesn't help.

Just to be sure you are selecting the MKR WIFI 1010 from the arduino board manager like so:
[image]

2bndy5, 2021-07-14:
@Oa373 this is going to be rather ugly hack, but I suggest modifying line 172 of RF24_config.h from 
```cpp
            #if defined (ARDUINO_ARCH_SAMD) && !defined (ARDUINO_API_VERSION) 
```
to
```cpp
            #if defined (ARDUINO_ARCH_SAMD) && (defined (ADAFRUIT_FEATHER_M0) || defined (ADAFRUIT_FEATHER_M0_EXPRESS) || defined (ADAFRUIT_METRO_M0_EXPRESS) || defined (ADAFRUIT_CIRCUITPLAYGROUND_M0) || defined (ADAFRUIT_GEMMA_M0) || defined (ADAFRUIT_TRINKET_M0) || defined (ADAFRUIT_QTPY_M0) || defined (ADAFRUIT_ITSYBITSY_M0) || defined (ADAFRUIT_HALLOWING) || defined (ADAFRUIT_CRICKIT_M0) || defined (ADAFRUIT_METRO_M4_EXPRESS) || defined (ADAFRUIT_GRAND_CENTRAL_M4) || defined (ADAFRUIT_ITSYBITSY_M4_EXPRESS) || defined (ADAFRUIT_FEATHER_M4_EXPRESS) || defined (ADAFRUIT_FEATHER_M4_CAN) || defined (ADAFRUIT_TRELLIS_M4_EXPRESS) || defined (ADAFRUIT_PYBADGE_M4_EXPRESS) || defined (ADAFRUIT_METRO_M4_AIRLIFT_LITE) || defined (ADAFRUIT_PYGAMER_M4_EXPRESS) || defined (ADAFRUIT_PYGAMER_ADVANCE_M4_EXPRESS) || defined (ADAFRUIT_PYBADGE_AIRLIFT_M4) || defined (ADAFRUIT_MONSTER_M4SK_EXPRESS))
```
This should make the `Serial.printf` line only trigger on boards supported by the adafruit fork of the arduino-samd core, more specifically not the boards supported by any other arduino-samd core.

Oa373, 2021-07-14:
Hello Thank you for you answers , I did manage to compile it however it will not do the job requred. As i mentioned earlier i am trying to send a text from my MKRWIFI 1010 to my uno  with the former being the transmitter and the latter the reciever . For the both boards those are my connections ce=6 csn=7 mosi=8 miso=10 sck=9 . I have changed the pin numbers in the iconfig file accordingly  lines  79-91. But the data i get on my reciever are just empty boxes and spams of unwanted characters.

2bndy5, 2021-07-15:
>But the data i get on my reciever are just empty boxes and spams of unwanted characters.

I don't know if this will help, but c-strings in C++ need to end in a `'\0` (null terminating char). Without this `'\0'`, printing a c-string to the Serial monitor may be misinterpreted.

>I have changed the pin numbers in the iconfig file accordingly lines 79-91

IDK what this means.

2bndy5, 2021-08-23:
I submitted a patch (new macro is called `ARDUINO_SAMD_ADAFRUIT`) to the adafruit fork of the arduinoSAMD-core that lets us determine if the board is indeed using the adafruit samd core. We can now pin to version v1.7.4+ of the adafruit samd core for better compile-time detection of `printf()` on SAMD-based boards.

Note that `printf()` is still not supported by the official arduinoSAMD-core (& probably never will be).