nRF24/RF24 #779: [Question] What's with the duplicate py wrappers for network/mesh libs?

2bndy5, 2021-07-24:
I've been trying to keep the python wrappers' copies of RF24Network and RF24Mesh located in this repo up-to-date with their respective source repos. However, the improved _setup.py_ files for Network & Mesh now use the version number from their respective _library.properties_ files while the copies on this repo default to using "1.0" for their version.

I'm wondering if it is ok to remove the copies on this repo since
1. it would be easier to maintain 1 copy on the repo containing the actual C++ source code
2. the auto-install script on https://github.com/TMRh20/tmrh20.github.io/blob/master/RF24Installer/RPi/install.sh doesn't install the python wrappers (so no harm would be done)
3. version numbers will be more accurate (with patch numbers) in `pip list` output when the py wrappers are installed from the cloned repo that contains the C++ source code
   ```shell
   (env) pi@rpi4:~/github/RF24Network/RPi/pyRF24Network $ pip list
   Package       Version
   ------------- -------
   pip           21.1.3
   pkg-resources 0.0.0
   RF24          1.4.2
   RF24Mesh      1.1.6
   RF24Network   1.0.16
   setuptools    57.4.0
   wheel         0.36.2
   ```

ps - A copy of the LICENSE file will be included in the py wrappers' package metadata upon next release. Additionally, there is a hidden warning in the packages' `long_description` that warns about how the wrappers aren't meant to be uploaded to pypi; this warning would only be seen on pipy.org when a package _is_ uploaded to pypi by someone that doesn't know what they're actually doing.

TMRh20, 2021-08-22:
I think it should be ok to remove them as long as associated documentation is up to date with that.

2bndy5, 2021-10-31:
The python wrapper docs only existed in RF24 repo, and they seemed a little more implicit than they should've been. So, I made them a bit more explicit concerning the separate python wrappers from different RF24* source repos.

Now I'm ready to remove the duplicates from this RF24 repo...

commit 04b103e8 (2022-01-22): resolve #779 and augment docs
closed 2021-10-31 by 2bndy5