nRF24/RF24 #780: uncorrect reading from getting_started.py in examples_linux

robinpdev, 2021-07-24:
**Describe the bug**
When i use the [getting_started.py example ](https://github.com/nRF24/RF24/blob/master/examples_linux/getting_started.py) and read from my pi zero W and write from my arduino pro mini, the readings i get are not entirely correct, the float i receive is very close to the expected behaviour, but is always a bit less then needed.

Example output:
Received 4 bytes on pipe 1: 0.0
Received 4 bytes on pipe 1: 0.009999999776482582
Received 4 bytes on pipe 1: 0.019999999552965164
Received 4 bytes on pipe 1: 0.029999999329447746
Received 4 bytes on pipe 1: 0.03999999910593033
Received 4 bytes on pipe 1: 0.04999999701976776

expected output:
Received 4 bytes on pipe 1: 0.0
Received 4 bytes on pipe 1: 0.01
Received 4 bytes on pipe 1: 0.02
Received 4 bytes on pipe 1: 0.03
Received 4 bytes on pipe 1: 0.04
Received 4 bytes on pipe 1: 0.05

Is there something i might be doing wrong? Or is this expected behaviour and do i need to round the float?

2bndy5, 2021-07-24:
This is normal. I specifically chose to use floats to demonstrate this inaccuracy. It actually has nothing to do with the radio or library, rather it is due to the way floats are stored in memory (and this can be more or less accurate depending on the CPU).

Since you're using the python example, you probably have noticed that it uses python's builtin `struct` module to `pack()`/`unpack()` the float data. Here is an excerpt from the [python docs](https://docs.python.org/3/library/struct.html#format-characters):
>For the 'f', 'd' and 'e' conversion codes, the packed representation uses the IEEE 754 binary32, binary64 or binary16 format (for 'f', 'd' or 'e' respectively), regardless of the floating-point format used by the platform.

If you would like to continue researching this inconsistency, I would recommend looking into IEEE 754 binary32 specification.

Avamander, 2021-07-24:
If you can determine the amount of decimal places you need, you can just use an integer and multiply accordingly. E.g. instead of `5.6789` you can store it as `56789` and multiply by `0.0001` when you need to get it back to "real float". In a lot of cases on embedded systems not using floats yields significant improvements in terms of performance, memory usage and protocol complexity.

robinpdev, 2021-07-24:
@2bndy5 Thanks for the clarification

closed 2021-07-24 by robinpdev