nRF24/RF24 #781: change in PIO lib registry disabled auto-crawl for all libs

2bndy5, 2021-07-25:
ok. it looks like I was right about the change in PIO lib registry behavior. They are working toward disabling "auto-crawl" for "legacy registry" libs. Therefore, we have to automate lib publishing to PIO ourselves from now on. I think this is a recent-ish change upon moving to [PIO-core v5.0.0](https://github.com/platformio/platformio-core/releases/tag/v5.0.0), although I don't see where it says anything about the legacy "auto-crawl" feature going away.

I can add a conditional step to the PIO CI workflow that will publish the lib when triggered by a "published" release event, but I need a new Github secret token set ([`PLATFORMIO_AUTH_TOKEN`](https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_AUTH_TOKEN) for each nRF24 repo) to use as a the PIO account authentication for publishing the libs - I'm going off of the [PIO docs from here](https://docs.platformio.org/en/latest/core/userguide/account/cmd_login.html). I don't have that kind of access to RF24* repos to set the github secret tokens (I would be using my credentials if I had the access).

_Originally posted by @2bndy5 in https://github.com/nRF24/RF24/pull/772#issuecomment-877727201_

TMRh20, 2021-08-22:
I have no idea how to accommodate this request . :/

2bndy5, 2021-08-23:
>I need a new Github secret token set (PLATFORMIO_AUTH_TOKEN for each nRF24 repo)

This can actually be done at the organization level, so that each repo can access the same secret token. I know @TMRh20 is on-board with me becoming an owner, but @Avamander may need to help make this happen. I do appreciate giving me access to all the repos in the org, but that seems like a work-araound to not knowing how to change me to an owner.

[image]

TMRh20, 2021-08-23:
Ok you are set as an owner now. :)

2bndy5, 2021-08-23:
ok token has been added. I'll start going through the PIO CI workflows to add an auto-upload-pkg step that gets triggered on release events. Thanks @TMRh20 !

ps - I'll also be transferring my circuitpython lib over soon...

2bndy5, 2021-09-24:
verified solution works (locally). See https://github.com/nRF24/RF24/pull/772#discussion_r715259970

2bndy5, 2021-11-28:
Good news! I checked all RF24* libs that just got released for subsequent releases on PlatformIO, and it looks like my new secret token worked as expected.