nRF24/RF24 #784: Adding GPIO Expander(23017) Support to the library

Frenzyritz13, 2021-08-16:
@Frenzyritz13 I just modified my post (found/fixed an error in the logic). Please be sure to use the updated snippet about `_ini_pins()`

>This looks good Brendan! Will test it and get back to you. Also, I am going to be using 2 I2Cs and an Expander based interrupt, I am expecting some issues there as well, will keep this thread updated.

This deserves a separate thread/issue because we are no longer discussing just the ESP32's SPI bus.

_Originally posted by @2bndy5 in https://github.com/nRF24/RF24/issues/722#issuecomment-899452928_

2bndy5, 2021-08-16:
For reference, this is the suggested idea here: https://github.com/nRF24/RF24/issues/722#issuecomment-899446116

2bndy5, 2021-08-16:
> Also, I am going to be using 2 I2Cs and an Expander based interrupt, I am expecting some issues there

I should probably mention that using any `RF24::write*()` call in an ISR callback is prohibited by the RF24 library because it uses `millis()` to detect timeouts and `delayMicroseconds()` for every SPI transaction on boards that have a high speed CPU. You can have a look at the _interrupt_configured.ino_ example to see a working hint about using a volatile boolean to get around that.

2bndy5, 2023-01-17:
I'm closing this due to lack of activity. The associated PR (#787) will remain open for anyone that is curious, but we have no plans to support this feature.

closed 2023-01-17 by 2bndy5