nRF24/RF24 #789: Has this been successfully used with a Raspberry Pi 4? [Question]

jpollack, 2021-09-01:
I'm unable to get this library to work with a Raspberry Pi 4.  I specify the BCM2835 driver to `configure`, but when I run any of the examples `SPIException` is thrown, implying that the SPIDEV driver is linked in.  I've verified that I can talk to the NRF24L01+ using my own short program that uses BCM2835.

Before spending any more time trying to fix, I'm curious if this is a known issue?

2bndy5, 2021-09-01:
I can confirm it does work. However, the OS you're running can treat some things differently (there's a CMake solution in the works for that).

I, personally, have noted better reliable SPI connection at lower than default spi_speed. But that may just be specific to my setup or something.

jpollack, 2021-09-02:
Thanks for the info!  It's useful to know there's a light at the end before exploring a tunnel.

I'm using debian bullseye, which I didn't think would be too far from the beaten path, but *shrug*.

After poking at it more (and based on your info), I'm convinced the problem is something going awry in the build process, not anything in code.  I'll keep rolling my own (a habit I'm kinda trying to get out of...), and check out the CMake solution seen in the PR.   Thanks!

closed 2021-09-02 by jpollack

2bndy5, 2021-09-02:
if you're interested in the rp2xxx branch then check out the CMake instructions in the docs folder of the rp2xxx branch. So far, the CMake solution tends to solve compiling 32bit vs 64bit versions (depending on your OS's bitness).

jpollack, 2021-09-02:
Bitness! That's a good clue.  I'm using the Arm64 release, and just noticed that `make` passes `-march=armv6zk` to the compiler, which I don't think is right.  These flags look like they want a `armel` or `armhf` release.

2bndy5, 2021-09-02:
specifically armhf release. The RPi-OS is still using 32 bit for the RPi4