nRF24/RF24 #790: radio power off, and transmit again after power on

flexphperia, 2021-09-08:
I have a problem sending data after turning off the radio's power.
My code is more complex, but here's what the problem is.
My project has very little power resources, so I disconnect the power from the NRF24L01 + when I am not using it (the NRF24 is connected to the power supply via LDO).
When I turn on the power and reinitialize the radio, everything works as it should, but sometimes the first message sent with the `write()` (or other) method does not reach the addressee, even though the method returns `TRUE`.

I found a solution: when I use the radio for the first time after turning on the power, I send two messages one after the other, both of them always come.

Is it an anomaly that turning the power off can have unexpected effects, or am I doing something wrong?

Below sample code:

```cpp
bool isFirstMessage = true;

bool rslt;

void loop() {
  initRadio();
  isFirstMessage = true;
  uint8_t sendData[2] = {1, 2};

  if (isFirstMessage){ 
    //without this next radio.write will not work all the time, 
    //but with this both will always work
    radio.write( &sendData, sizeof(sendData) );
    isFirstMessage = false;
  }
  rslt = radio.write( &sendData, sizeof(sendData) ); //without above this will return true sometimes but not send

  delay(1000);
  rslt = radio.write( &sendData, sizeof(sendData) ); //this will work always no matter what

  delay(1000);
  powerDownRadio();
  delay(1000);
}

void initRadio(){
  pinMode(RADIO_PWR_PIN, OUTPUT);
  digitalWrite(RADIO_PWR_PIN, HIGH);

  delay(30); //wait to LDO power up

  radio.begin();

  radio.setChannel(112);
  radio.setPayloadSize(2); 
  radio.setDataRate( RF24_250KBPS );
  radio.setRetries(15, 15);
  radio.openWritingPipe(transmitterAddress);
}

void powerDownRadio(){
  digitalWrite(RADIO_PWR_PIN, LOW); //turn off radio power
}
```

2bndy5, 2021-09-08:
try adding `radio.stopListening()` to the end of your `initRadio()`. This suggestion is just a guess. I'm not really sure what could be the problem.

You didn't mention anything about the receiver, so I assume that
1. there is only one receiver listening for the `transmitterAddress`
2. the receiver is always powered up and listening

You also didn't mention the type of radio module that you're using (I'm guessing from the config and LDO that you're using a PA/LNA radio).

> NRF24 is connected to the power supply via LDO

Did you add capacitance also? Any linear regulator's datasheet will typically recommend what capacitance to use.

TMRh20, 2021-09-09:
> Is it an anomaly that turning the power off can have unexpected effects, or am I doing something wrong?

This type of thing works fine for me. I would suspect power supply issues. Have you tried with a higher dataRate? It should use slightly less power per transmission at 2mbps and may make a difference.

flexphperia, 2021-09-09:
There is only one receiver that is connected to the power supply all the time.
In the transmitter, I use modules with PA/LNA and without with the same effect.
Its power source always has a 10uF capacitor and of course the LDO also has the appropriate capacitors.
With the power supply, it seems to me that I have no problem because I am able to transmit almost constantly loading the power source and all messages are delivered. After starting the power supply, there is also a waiting time in the code for the voltage to stabilize.
Yes, I know the article "RF24 Addressing Gotchas" and I followed the instructions.
Adding r`adio.stopListening()` changing baud rates, addresses etc doesn't help either.

Of course I can live with it, but maybe something else comes to your mind?

TMRh20, 2021-09-09:
I’ve attempted to recreate the issue using your code and it works fine for me, so I’m assuming it’s not a software issue. Not much for suggestions here…

2bndy5, 2021-09-09:
> In the transmitter, I use modules with PA/LNA and without with the same effect.

I think there is a typo here: "without with the same effect". Could you clarify?

PA/LNA modules are famous for needing excessive power to transmit. Unless you need to transmit farther than 30 ft, I would stick with the less power hungry non-PA/LNA modules.

> After starting the power supply, there is also a waiting time in the code for the voltage to stabilize.

This is good. 30 ms should be enough, but I don't know the capacitance on the power supply circuit. I only mention this because the wait time can be calculated according to the "5 time constants" of capacitors.

---

I'm sorry if what I say seems like old news to you, and I understand this is frustrating enough to open an issue here. It is refreshing to get questions from someone with considerable competence.

flexphperia, 2021-09-10:
As an explanation: Regardless of whether I use the radio with or without an amplifier, I have the same problem.
In a few weeks I will have new PCBs for my project so maybe this will fix the problem. Please do not close this thread. I will come back to the topic.

2bndy5, 2021-09-10:
If you're using a custom PCB, have you tested the code with no PCB (direct connection to GPIOs)? Several issues in the past were solved by eliminating PCBs from the setup.

I usually leave threads like this open for a few months with no response (lots of people don't seem to be considerate enough to close the issue they open).

flexphperia, 2021-09-10:
Now my project is breadboard prototype, but reasonably solidly constructed. I do not suspect connection cables, but I will check with PCBs and write here.

2bndy5, 2021-09-10:
The only connections I would doubt in this case are the PWR & GND

flexphperia, 2021-09-24:
Okay, I have new PCBs, I have done hundreds of tries and still nothing, the problem is still there.
Interruptions are the cause.
My code is a bit more complicated than the one I posted.

The principle of its operation is as follows:
1. The code waits for the button to be pressed for 3 seconds.
2. In case of inactivity, it turns off the radio, sets an interrupt to listen for a button press and puts the arduino to sleep.
3. When the button is pressed, the arduino wakes up and turns on the radio.
4. Now the first data trasnmitting always fails but `write` method returns `TRUE`.

Nothing works: wait some time, wait some loop execution, etc.

All that will work is to rerun the `write` command. For example, I tried to change the broadcast channel so as to send this first message over the air and then return to my channel, but it also doesn't work. The `write` command must be addressed to the same address, on the same channel, at the same speed as next call. Of course, this first command reaches the receiver once every 20 times.

When I run the code without interrupts, I can turn the radio on and off without any problems and it works.

2bndy5, 2021-09-24:
Yep. Don't use `RF24::write()` call within a ISR callback method because `RF24::write()` functions use `millis()` to implement a timeout. More info about why you can't call `millis()` from an ISR callback function is available at [the Arduino docs](https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/&ved=2ahUKEwjMhtSRl5jzAhVUsp4KHWIPCrIQFnoECAoQAQ&usg=AOvVaw1X9H0058Nz7Hck91VIC3bD).

I've been saying this quite a bit lately. I think it's time to put this tidbit in the COMMON_ISSUES.md doc. This is noted in the RF24 docs somewhere but I can't find it right now.

flexphperia, 2021-09-24:
O course I don't use `write` in ISR routine. I use it after.

2bndy5, 2021-09-24:
Hmm. Can we see your code in it's entirety?

flexphperia, 2021-09-24:
Yes, I will prepare a code with this case in days and add it here.

flexphperia, 2021-09-25:
This is a minimalist version of the code in which the radio does not work properly on the first send right after waking up. The code is fairly simple and should be easy to read.

```cpp
#include "Arduino.h"
#include <LowPower.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <SPI.h>

#define BUTTON_PIN 2
#define RADIO_PWR_PIN 3
#define SLEEP_TIMEOUT 3000

unsigned long lastButtonActionTime = 0;
bool firstSendAfterSleep = true;
bool wasPressed = false;
volatile bool wakingUp = false;

const uint8_t address[3] = {'A','B', 'C'};

RF24 radio(9, 10);

void setup()
{
	Serial.begin(9600);
	delay(300); //wait for serial begin
	Serial.println("Started");

	initRadio();

	configureButton();
}

void loop()
{
	//checks that we are in first loop after waking up from ISR and do things
	wakeUp();

	bool nowPressed = digitalRead(BUTTON_PIN) == LOW;

	if (!wasPressed && nowPressed){
		wasPressed = true;

		lastButtonActionTime = millis();

		bool rslt = false;

		//hack to ensure that below this block write will be delivered
		if (firstSendAfterSleep){
			firstSendAfterSleep = false;

			//aditional send something
			uint8_t a = 11;
			rslt = radio.write(  &a, 1 ); //this will reach target sometimes, sometimes not
			Serial.print("First send after wake up, result: ");
			Serial.println(rslt);
		}

		uint8_t sendData[2] = {82, 45};
		rslt = radio.write(  &sendData, 2 );

		Serial.print("Send result: ");
		Serial.println(rslt);
	}
	else if(!nowPressed){
		wasPressed = false;
	}
	shouldSleepCheck();
}

void initRadio(){
	//run LDO to power up radio
	pinMode(RADIO_PWR_PIN, OUTPUT);
	digitalWrite(RADIO_PWR_PIN, HIGH);

	delay(20);

	firstSendAfterSleep = true;

	radio.begin();

	radio.setChannel(120);
	radio.setPayloadSize(2);
	radio.setCRCLength(RF24_CRC_8);
	radio.setAddressWidth(3);
	radio.setPALevel(RF24_PA_MIN, 0);
	radio.setDataRate( RF24_250KBPS );
	radio.setRetries(15, 15);
	radio.openWritingPipe(address);

	Serial.print("Waked up, isChipConnected: ");
	Serial.println(radio.isChipConnected());
}

void powerDownRadio(){
	digitalWrite(RADIO_PWR_PIN, LOW);
}

void configureButton(){
	detachInterrupt(digitalPinToInterrupt(BUTTON_PIN));

	pinMode(BUTTON_PIN, INPUT_PULLUP);
}

void configureISR() {
	pinMode(BUTTON_PIN, INPUT_PULLUP);

	EIFR = bit (INTF0); // clear flag for interrupt 1 (pin 2)
	attachInterrupt(digitalPinToInterrupt(BUTTON_PIN), buttonISR, FALLING);
}

void buttonISR(){
	wakingUp = true;
	configureButton(); //immidiately detach interrupt
}

void wakeUp(){
	if (!wakingUp){ //are we waking up
		return;
	}

	Serial.println("waking up");
	lastButtonActionTime = millis(); //prevent immediately sleep
	initRadio();
	wakingUp = false;
}

void shouldSleepCheck(){
	if (millis() - lastButtonActionTime > SLEEP_TIMEOUT)
	{
		Serial.println("sleeping");
		delay(100);

		powerDownRadio();
		configureISR();

		LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_OFF);
	}
}```

2bndy5, 2021-09-25:
> try adding `radio.stopListening()` to the end of your initRadio(). 

This suggestion wasn't optional. It is required usage of the RF24 API to call `stopListening()` after `openWritingPipe()`.

I personally prefer using the return value of `RF24:: begin()` instead of `isChipConnected()` because it's less error prone and saves on code size.

I see you changed the delay value after enabling the LDO. This value is critical. If anything I would try increasing it. Otherwise you can just use `RF24::powerUp()` and `RF24::powerDown()` to control the radio's power cycle. But that's a speed vs power situation; using the LDO is slower, but using `RF24::power*()` still uses a few micro amps when the radio is powered down.

Lastly, I still think this is a hardware problem. It could be related to the custom PCB or the capacitance on the LDO circuit (vs that aforementioned delay value). I can't really think what else it would be. The interrupt code looks fine - there's nothing that misuses the radio there.

flexphperia, 2021-12-04:
I confirm after dozens of trials and several different prototypes, but my problem was LDO. Problem solved.

closed 2021-12-04 by flexphperia