nRF24/RF24 #791: RPI-4 Hello World Fail

sauravuprety21, 2021-09-09:
Hello, I have been trying to send "HelloWorld!" between two RPi 4 Model B. I also have an Arduino listening to debug.

The "HelloWorld!" is sent from the first RPi. Both the Arduino and the second RPi recieve the message. However, the second RPi recives something else and not hello world. The Arduino recieves helloworld though

I have pasted my codes below and have attached the output images.
[image]

Here is my script on sending py:
```
import RF24
import time
import struct

radio = RF24.RF24(22,0)

def main():

        if not radio.begin():
                raise RuntimeError("Radio hardware not responding")

        address = b"00001"

        radio.openWritingPipe(address)
        radio.setPALevel(RF24.RF24_PA_HIGH)
        radio.setDataRate(RF24.RF24_2MBPS)
        radio.stopListening()

        while(1):

                msg = b'HelloWorld!'
                radio.write(msg)
                print(f"[SENT] - {msg}")
                time.sleep(1)

if __name__=="__main__":
        main()
```

Receiving RPi:
```
import RF24
import time
import struct

radio = RF24.RF24(22,0)

def main():

        if not radio.begin():
                raise RuntimeError("[ERROR] - Radio Hardware not responding!")

        address = b"00001"

        radio.openReadingPipe(0,address)
        radio.setDataRate(RF24.RF24_2MBPS)
        radio.setPALevel(RF24.RF24_PA_HIGH)
        radio.startListening()
        while(1):
                if(radio.available()):
                        msg = radio.read(len(b'HelloWorld!'))
                        print(f"\n[MESSAGE] - {msg}")
                        print(f"[DECODED] - {msg.decode('utf-8')}\n")

if __name__=="__main__":
        main()
```

Recieving Arduino:
```
#include "nRF24L01.h"
#include "RF24.h"
#include "SPI.h"

/*radio(CE,CSN)*/
RF24 radio(7,8);

const uint8_t address[6] = "00001";

void setup() 
{
  radio.begin();
  radio.openReadingPipe(0,address);
  radio.setDataRate(RF24_2MBPS);
  radio.setPALevel(RF24_PA_HIGH);
  radio.startListening();

  Serial.begin(9600);
}

void loop() 
{
  if(radio.available())
  {
    char text[32] = "";
    radio.read(&text, sizeof(text));
    Serial.print("[RECEIVED]:");
    Serial.println(text);
  }
}
```

2bndy5, 2021-09-09:
I have seen this garbled pattern before... Lower the SPI speed on the RPi experiencing problems. It would be better for deployment to check the wiring connections are not loose and the wires are not too long.

sauravuprety21, 2021-09-10:
Thank you very much, this fixed it!
@2bndy5

closed 2021-09-10 by sauravuprety21