nRF24/RF24 #792: [Question] Can pipe address be changed periodically in the main loop?

dkoung, 2021-09-10:
Hello, I am wondering if we can keep changing the pipe address for listening/reading in the main loop. I want to listen to more than 6 different transmitters whose addresses are all different. So, if it is possible to first open all 6 pipes to 6 transmitters' addresses, listen and save the data. Then change these 6 pipes to the other 6 transmitters' addresses. Thanks.

TMRh20, 2021-09-10:
Yup, you just need to call stopListening before changing the addresses

dkoung, 2021-09-17:
> Yup, you just need to call stopListening before changing the addresses
> […](#)

Thanks for the quick response. Do we need to put any delay between "stopListening" and "openReadingPipe" of the new address? or between "startListening" of the new address and before calling "available"?

2bndy5, 2021-09-17:
If you look at the source code, you'll see that there is a delay already implemented (as mandated by the datasheet).

dkoung, 2021-09-17:
Ok, got it. Big thanks!

closed 2021-09-17 by dkoung