nRF24/RF24 #795: Send and Receive multiple Arrays

akilakavinda, 2021-09-26:
I'm doing research for Weather Station. For that, I need to send a large number of values which will be more than 32 Bytes. Since the Max Payload is about 32 Bytes, I tried sending those data by using 2 Arrays (Tx and Rx files are attached).  But it didn't give an output what I anticipated.  My requirement is to send all those values at once (using any method) and use all those values in different places in the receiver programme. Therefore, highly appreciate help in this scenario.

[Project Files.zip](https://github.com/nRF24/RF24/files/7230880/Project.Files.zip)

2bndy5, 2021-09-26:
I haven't looked at your code (yet), but you could try using RF24Network because it has builtin capability to send payloads (aka "network messages") of up to 144 bytes; RF24Network lib fragments large messages into 32 byte payloads. Are you using an ATTiny (there's some default limitations in RF24Network for ATTiny MCUs)?

akilakavinda, 2021-09-26:
I'm using Arduino Mega2560 as Transmitter and ESP32 as Receiver. I include the code in the question itself now.

2bndy5, 2021-09-26:
I still think using RF24Network would be easier for your case.

Otherwise, you would likely need to use pointers and `memcpy()` fragments of the total data to send into a temporary 32 byte buffer (usually a uint8_t array) that is used to send the data. I'm not sure about your skill level for such a task. Keep in mind the temporary 32 byte buffer would need to be cleared after each fragmented transmission, and you need to keep track of the amount of data that you `memcpy()` into the temporary buffer (then pass that amount to `memcpy()` & `RF24::write()`).

2bndy5, 2021-09-26:
wait a minute! A `float` occupies 4 bytes. So, 
```cpp
float Weather_Data_01[6]; // allocates 24 bytes
float Weather_Data_02[5]; // allocates 20 bytes
```
These should fit in 2 transmissions. You want to transmit it all at once? That's not possible with this radio. The 32 byte limit is a hardware limit not a library limit.

2bndy5, 2021-09-26:
```cpp
  if (radio.available()) {
    radio.read(&Weather_Data_01, sizeof(Weather_Data_01));
    radio.read(&Weather_Data_02, sizeof(Weather_Data_02));
```
this is likely the problem. `available()`  is asserted for each received payload. So if it only receives 1 out of 2 payloads, then the data fetched by the second `radio.read()` is dummy data from the radio's empty FIFO.

what you need is
```cpp
bool gotPayload1 = false;

void loop() {
  if (radio.availble()) {
    if (gotPayload1) {
      radio.read(&Weather_Data_02, sizeof(Weather_Data_02));
      gotPayload1 = false; // reset for the next pair of payloads
      /* do stuff with data from both arrays */
    }
    else {
      radio.read(&Weather_Data_01, sizeof(Weather_Data_01));
      gotPayload1 = true;
  }
}

2bndy5, 2021-09-26:
you could also send a payload ID number with each payload to help identify which one was received, but that be replicating RF24Network.

akilakavinda, 2021-09-26:
That code worked and I was able to get what I anticipate from the output. Thank you very much and highly appreciate your tremendous help.
Also, I was able to use more than 2 arrays to get the same output using the Switch statement. But, it's better to use Payload ID, because from time to time, the sequence of values gets changed.

I used this code in the receiver and if there is any other easy method to do the same, high appreciate help on that too.
`
bool gotPayload1 = false;
int ArrayID = 1;

void loop() {

  if (radio.available()) {
    if (gotPayload1) {
      radio.read(&Weather_Data_04, sizeof(Weather_Data_04));
      gotPayload1 = false; // reset for the next pair of payloads
      /* do stuff with data from both arrays */
     // I printer all values received from those arrays
    }
    
    else {
      
      switch (ArrayID) {
        case 1:
          radio.read(&Weather_Data_01, sizeof(Weather_Data_01));
          ArrayID = 2;
          break;
        case 2:
          radio.read(&Weather_Data_02, sizeof(Weather_Data_02));
          ArrayID = 3;
          break;
        case 3:
          radio.read(&Weather_Data_03, sizeof(Weather_Data_03));
          gotPayload1 = true;
          ArrayID = 1;
          break;
      }
    }
  }
}`

2bndy5, 2021-09-26:
Glad to see we answered your first question ever on github. It is customary to close an issue that you opened if there is nothing more to be discussed. You can still comment here even if it is closed.

2bndy5, 2021-09-26:
>  if there is any other easy method to do the same

look up RF24Network. If you ever decide to get data from more than 1 weather station, then using RF24Network can simplify your program(s).

akilakavinda, 2021-09-26:
Highly appreciate your support again. I'll look into RF24Network.

closed 2021-09-26 by akilakavinda