nRF24/RF24 #796: error: 'printf_P' was not declared in this scope

Onkar35, 2021-09-29:
When tried to complie for the ESP32 it says this error 

Arduino\libraries\RF24-master\RF24.cpp:442:127: error: 'printf_P' was not declared in this scope

     printf_P(PSTR("OBSERVE_TX=%02x: POLS_CNT=%x ARC_CNT=%x\r\n"), value, (value >> PLOS_CNT) & 0x0F, (value >> ARC_CNT) & 0x0F);

unable to solve the issue

2bndy5, 2021-09-29:
What exact board are you compiling for (there are many esp32 boards)?

Did you make any alteration to the library?

Are you using Arduino IDE or platformIO?

We simply need more details about your scenario.

Onkar35, 2021-09-30:
Compiling for esp devkit v1

No alteration done in library

Using Arduino IDE 15

2bndy5, 2021-09-30:
>Using Arduino IDE 15

I think its time to update your Arduino IDE. They haven't used a version number without a decimal in it for many years now. After visiting the https://www.arduino.cc/en/software site, the latest version is v1.8.16 (they're working towards v2.0). Version 15 came well before they switch to v1.0. I'm guessing your working from Linux and installed it via `apt` (which is strongly discouraged by the Arduino people).

I just compiled the GettingStarted.ino sketch with v1.8.15, and it worked fine for the "ESP32-Dev Module". I don't actually have that board to wire it up, but it compiles for me with no problems.

To be clear, I did un-comment the following lines so that `printf_P()` line was invoked.
https://github.com/nRF24/RF24/blob/e22a8be1a56e563fff3b08c01040d6bd42f4cf2f/examples/GettingStarted/GettingStarted.ino#L89-L90

Onkar35, 2021-09-30:
Thanks for the help. Clean installation both ESP32 and Arduino is resolved the issue

closed 2021-09-30 by Onkar35