nRF24/RF24 #798: Hey @2bndy5 , thanks for your hard work on this library - much appreciated.

2bndy5, 2021-10-07:
Hey @2bndy5 , thanks for your hard work on this library - much appreciated. 

If this should be in a new issue, please split it out.

Just wanted to point out that there's a really important use case for setting the first address byte to `0x00`, namely protocol reverse engineering. Here's some reading if you're unfamiliar:

* https://travisgoodspeed.blogspot.com/2011/02/promiscuity-is-nrf24l01s-duty.html
* http://www.remote-exploit.org/articles/keykeriki_v2_0__8211_2_4ghz/index.html
* https://github.com/Yveaux/NRF24_Sniffer

Right now, to get around the limitations of this library, something like this is required:

```c++
radio.setAutoAck(false);
writeRegister(RF_SETUP, 0x21); // Disable PA, 250kbps, LNA enabled in a single op - no ctor to do this in one go?
radio.setPayloadSize(sz);
radio.setChannel(ch);
writeRegister(EN_RXADDR, 0x00);    // this is critical - seems no way to do this with RF24?
writeRegister(SETUP_AW, 0x00);  // though perhaps passing in a_width of 2 would work? https://github.com/nRF24/RF24/blob/2819745c727cccc2cf2308595a77c637cc84cd95/RF24.cpp#L1417
radio.openReadingPipe(0, 0xAALL);
radio.openReadingPipe(1, 0x55LL);
radio.disableCRC();
radio.startListening();
// ...
```

where `writeRegister` is a raw SPI transfer that also handles `CSN` toggling. Not pleasant!

I see 70786aa which looks encouraging to help with some of this - can you clarify how the use case I've outlined might be supported by this commit? I expect other language bindings (like the Python one) will need to have this variable exposed as well...) Thanks!

_Originally posted by @wohali in https://github.com/nRF24/RF24/issues/496#issuecomment-937375570_

2bndy5, 2021-10-07:
## This is dense; grab a drink now - then continue

> I see 70786aa which looks encouraging to help with some of this

That commit is on [my experimental fork's "revamped" branch](https://github.com/2bndy5/RF24Revamped/tree/revamp)... I haven't tested it in a while (at least a few months at this point). The main idea is that the added private bool `_is_pipe0_rx` is toggled true when pipe 0 is used for RX operations (via calling `openReadingPipe(0, some_addr)`, then toggled false when pipe 0 is closed for RX operations (via calling `closeReadingPipe()`). This allows more intuitive handling of pipe 0 when `startListening()` is called. Really, this was hinted by @hexwell (thus the tag in commit title) to bypass the "null vs 0" problem that #496 raises/discusses.

The use of an extra bool to do this would occupy a whole byte in memory and only serves to resolve #496 seeming edge case (in spite of [poor addressing outlined by ManiacBug](http://maniacalbits.blogspot.com/2013/04/rf24-addressing-nrf24l01-radios-require.html))

>I expect other language bindings (like the Python one) will need to have this variable exposed as well...) 

Not likely because the added bool would not need a public scope (only the lib would need to use it), and the python wrapper does not expose private members (using either boost.python or pybind11). If it was a pure python lib (like my CirPy lib), then it could be accessed (because nothing is really private in python). However, the entire "null vs 0" problem is avoided with the difference between python's `None` vs the buffer `b"\0"`

---

>```cpp
>writeRegister(RF_SETUP, 0x21); // Disable PA, 250kbps, LNA enabled in a single op - no ctor to do this in one go?
>```

Correct, each part of this register is handled according to it's relative feature. Instead, this would be accomplished with
```cpp
radio.stopConstCarrier(); // would only reset PLL bit for non-plus variants of nRF24L01
// stopConstCarrier() needs to first because it resets some features to lib defaults (for non-plus variants only)
radio.setDataRate(RF24_1MBPS);
radio.setPALevel(RF24_PA_MIN, false); // false controls the LNA bit
```

---

>```cpp
>writeRegister(EN_RXADDR, 0x00);    // this is critical - seems no way to do this with RF24?
>```

This can be done with RF24 lib like so
```cpp
for (uint8_t i = 0; i < 6; ++i)
    radio.closeReadingPipe(i);
```
However, it is no where near as efficient as clearing the whole register at once. I remember a comment in src from @TMRh20 explaining the lib's intention about handling pipes individually:
https://github.com/nRF24/RF24/blob/e22a8be1a56e563fff3b08c01040d6bd42f4cf2f/RF24.cpp#L1402-L1405

---

>```cpp
>writeRegister(SETUP_AW, 0x00);  // though perhaps passing in a_width of 2 would work? https://github.com/nRF24/RF24/blob/2819745c727cccc2cf2308595a77c637cc84cd95/RF24.cpp#L1417
>```

I often refer to the datasheet when faced with such questions. Observe
[image]
However, the datasheet **is known** to be inaccurate in some areas. Maybe the "'00' - Illegal" is yet another instance of inaccuracy(?).

---

> where `writeRegister` is a raw SPI transfer that also handles CSN toggling. Not pleasant!

1. This actually a common library convention for SPI implementation. Although your use case could be an exception (I'll come back to that later). I should note that I had at one point (when developing my CirPy lib) tried performing several SPI commands while the CSN remained HIGH. This resulted in the nRF24L01 ignoring everything after the first command was completed. The datasheet mandates, 
   > Every new command must be started by a high to low transition on CSN

   My experience concurs.
2. Unfortunately, this function is a private member. We can discuss making `writeRegister()` protected in more detail on #782 ; personally, I'm open to accepting it as a protected member.
   - As a protected member, one could easily inherit from `RF24` class and handle the CSN pin as needed.

---

# Where to go from here

I think your use case actually falls outside the scope of the RF24 lib's original intention which was to provide a beginner-friendly interface for the nRF24L01 transceiver. I talked a bit about inheriting from the RF24 class because, it could be extended to suite your use case. Although, some things might need to be made protected for 2 reasons
1. backward compatibility
2. The RF24 class simply wasn't designed to be altered via inheritance. I'm willing to help fix this, but any "tooling" derivative would need a home of its own (not belonging to this repo).

2bndy5, 2021-10-07:
I forgot to mention 1 thing:

>```cpp
>radio.setPayloadSize(sz);
>```

The minumum payload size that the RF24 class allows is 1 byte. This is because the datasheet states that setting the payload size for any pipe would effectively disables the pipe.
[image]

And, yes it could be done individually per pipe, but this lib doesn't support that (for code-size reasons). My CirPy lib does support individual pipe management for `payload_size`, `dynamic_payloads`, and even `auto-ack` 😉 .

2bndy5, 2021-10-08:
> - https://travisgoodspeed.blogspot.com/2011/02/promiscuity-is-nrf24l01s-duty.html

This is great reading! I've been wanting to reverse engineer Logitech's unifying receiver dongle, so that maybe an open-source mouse doesn't need any new receiver plugged into the PC's USB ports... These findings can't be shared publicly as Logitech does well to patch wireless exploits. The last exploit published prompted Logitech to use encryption on keyboards (& possibly mice now too).

wohali, 2021-10-08:
Hey @2bndy5 , thanks for splitting this out. I've skimmed most of this and understand, but wanted to focus on one thing:

> I think your use case actually falls outside the scope of the RF24 lib's original intention which was to provide a beginner-friendly interface for the nRF24L01 transceiver. I talked a bit about inheriting from the RF24 class because it could be extended to suite your use case.

My understanding was that this was a forked version of the original beginners' library, as the library's documentation states:

> More compliant with the manufacturer specified operation of the chip, **while allowing advanced users to work outside the recommended operation.**

I guess I gravitated towards this library in the hopes that it would be more amenable to RE use cases, rather than having to bitbang SPI to get what I am looking for. It's so very close -- and only needs a couple of very minor tweaks.

Given I'd prefer to be calling this most often from something like JS or Python (and there are already many issues installing this library in a managed Python dependency toolchain such as Poetry), I'd rather not have to sub-class everything myself. Otherwise I'm looking at a) maintaining a fork of the C++ code (ugh), and b) having to sort out publishing Python bindings to that myself, whereas I'd much rather some sort of compile-time option to your library (something like `-DUNSAFE_RX`) and to help contribute to the Python bindings to make this library's use simpler for standard tx/rx use cases.

FYI my intention has nothing to do with sniffing keyboard/mouse protocol stuff; I actually only got into this library to reverse engineer the protocol for my house's baseboard heaters, which turned out to be rather simple -- ONCE I got past the issues mentioned in my first post. 😉 

Does my counter-proposal interest you at all?

2bndy5, 2021-10-08:
> Does my counter-proposal interest you at all?

Yes! Sorry about the info dump; I got overexcited when you mentioned RE. I've also been looking at appropriating the firmware for the (now deprecated) steam controller which literally uses the Nordic ESB protocol when its not in BLE mode. IIRC, controller uses nRF51822 while the receiver uses a nRF24LU1.

>It's so very close -- and only needs a couple of very minor tweaks.

Specifically, what are you requesting? The code snippet in the OP is kinda broad, and leaves me with a partial idea.
- A solution for #496 is now being considered.
- Were you hoping for some new methods (like `open/closeAllReadingPipes()`)?
- How exactly were you planning to manipulate the CSN?
- Do you actually ever assert the PLL bit for RE usage (its mainly related to emitting the Carrier Wave in hardware tests)?

>a managed Python dependency toolchain such as Poetry
>...
>I don't know if CirPy is fast enough in this situation

1. I did some preliminary work on re-writing the python wrappers with pybind11 on one of my forks (see nRF24/pyrf24#3). The original intention is to have pre-built wheels released to pypi.org. 
2. My rf24-network branch of my CirPy lib has some speed-ups implemented. The most promising of which is depending on the SpiDev module (a C extension available from pypi.org) for Linux - this proved to be at least 7X faster than using adafruit's pureIO lib for SPI transactions (but still slower than this lb's python wrapper). I don't have BeagleBone, so I haven't been able to test if that branch works on that hardware.
   - I'm not fluent with poetry, but you can try the branch with
      ```
      pip3 install git+https://github.com/nRF24/CircuitPython_nRF24L01@rf24-network
      ```
      More info on the features of that branch (there's a lot) are listed in the PR (see nRF24/CircuitPython_nRF24L01#33)

Ultimately the difference between this lib and the CirPy lib is C++'s amazing execution speed. Although I generally like to do rapid developing in python (for experiments and proof-of-concepts), then port the python algorithms to C++. Although, having a pip installable C extension of this lib would be more ideal.

2bndy5, 2021-10-17:
@wohali I just pushed a solution to #496 on the rp2xxx branch. I would greatly appreciate any other advice or requests you may need to fulfill your RE needs.

wohali, 2021-10-18:
Hey @2bndy5 , sorry for the delay - been very busy with other stuff this past week, I'll try and review everything during the course of this coming week.

PR #804 (2021-11-27): Reverse Engineering tricks
This resolves #798 with the addition of 2 new public functions (& 2 new private functions for code re-use).

1. `toggleAllPipes()` can be used to open or close all data pipes (EN_RXADDR register) in 1 SPI transaction.
2. `setRadiation()` can be used to configure the RF_SETUP register in 1 SPI transaction. To re-use the existing code in `setDataRate()` & `setPALevel()` functions, there are 2 new private functions:
   -  `_data_rate_reg_value(speed)` replaces a bulk of `setDataRate()` by returning an uint8_t value that can be `|`'d into the value written to the RF_SETU [... cut]
closed 2021-11-27 by TMRh20