nRF24/RF24 #800: Funciona con arduino nano Every?

kikesandi, 2021-10-15:
Uso una Mac Pro, y haciendo pruebas con el Arduino uno, compila bien. En cambio, al utilizar un Arduino Nano Every no compila... hice el intento con 3 placas Every y sigue sin compilar. El error mostrado es "Imposible compilar para Every". Alguien más con este problema?

2bndy5, 2021-10-15:
Esa placa usa un ATMega4809 y otro chip ATSAMD11D14A como su chip USB serial. Tendré que investigar un poco más, pero supongo que esa placa no es compatible explícitamente.

2bndy5, 2021-10-15:
Puedo compilar para Arduino Nano Every fine. No puedo probar físicamente si funciona porque no soy dueño de esa placa.

[image]

Tenga en cuenta que Arduino Uno usa el núcleo AVR y Arduino Nano Every usa el núcleo megaAVR. El hecho de que mega esté en el nombre del chip no significa que los núcleos sean compatibles. Esta fue una mala elección de palabras por parte de la gente de Arduino (estaban enfrentando algunas razones muy técnicas).

kikesandi, 2021-10-20:
> Uso una Mac Pro, y haciendo pruebas con el Arduino uno, compila bien. En cambio, al utilizar un Arduino Nano Every no compila... hice el intento con 3 placas Every y sigue sin compilar. El error mostrado es "Imposible compilar para Every". Alguien más con este problema?

El asunto se resolvió reiniciando la computadora...