nRF24/RF24 #805: NRF24L01+PA+LNA transmitting/receiving 'blank' data and no ack

watsonpaul6, 2021-11-02:
I purchased a set of these: https://www.amazon.com/gp/product/B08M997T42/
and I am looking to have a few arduinos around the house that communicate to a central raspberry pi.

I have started with just trying to get a simple example running (gettingstarted.cpp) between an arduino uno and a raspberry pi and I am not having much luck. I have ran through all of the solutions to common problems with no luck, but I have yet to try the electromagnetic shielding as I currently have no electrical tape(in the mail). I will try that as soon as I can. It seems that there is some signal being picked up, but 99% of the time no correct data is coming through and no ack is successful. 

I have quadruple checked my pins are hooked up correctly, and I've swapped the wires I've been using. 

Here is a sample output for one session on each device. I refactored the gettingstarted examples on each device so the RPi is hardcoded as the slave and vice versa for the arduino. 

### RPi 3 Model B v1.2
'================ SPI Configuration ================
CSN Pin         = /dev/spidev0.0
CE Pin          = Custom GPIO22
SPI Speedz      = 4 Mhz
================ NRF Configuration ================
STATUS          = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1    = 0x65646f4e31 0x65646f4e32
RX_ADDR_P2-5    = 0xc3 0xc4 0xc5 0xc6
TX_ADDR         = 0x65646f4e31
RX_PW_P0-6      = 0x04 0x04 0x04 0x04 0x04 0x04
EN_AA           = 0x3f
EN_RXADDR       = 0x03
RF_CH           = 0x4c
RF_SETUP        = 0x01
CONFIG          = 0x0e
DYNPD/FEATURE   = 0x00 0x00
Data Rate       = 1 MBPS
Model           = nRF24L01+
CRC Length      = 16 bits
PA Power        = PA_MIN
ARC             = 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0
Received 4 bytes on pipe 1: 0

### Arduino Uno
18:25:44.525 -> SPI Speedz	= 4 Mhz
18:25:44.525 -> STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
18:25:44.525 -> RX_ADDR_P0-1	= 0x65646f4e32 0x65646f4e31
18:25:44.525 -> RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
18:25:44.571 -> TX_ADDR		= 0x65646f4e32
18:25:44.571 -> RX_PW_P0-6	= 0x04 0x04 0x04 0x04 0x04 0x04
18:25:44.571 -> EN_AA		= 0x3f
18:25:44.571 -> EN_RXADDR	= 0x03
18:25:44.571 -> RF_CH		= 0x4c
18:25:44.571 -> RF_SETUP	= 0x01
18:25:44.571 -> CONFIG		= 0x0e
18:25:44.571 -> DYNPD/FEATURE	= 0x00 0x00
18:25:44.571 -> Data Rate	= 1 MBPS
18:25:44.571 -> Model		= nRF24L01+
18:25:44.571 -> CRC Length	= 16 bits
18:25:44.571 -> PA Power	= PA_MIN
18:25:44.571 -> ARC		= 15
18:25:44.571 -> SPI Frequency		= 4 Mhz
18:25:44.571 -> Channel			= 76 (~ 2476 MHz)
18:25:44.571 -> RF Data Rate		= 1 MBPS
18:25:44.571 -> RF Power Amplifier	= PA_MIN
18:25:44.571 -> RF Low Noise Amplifier	= Enabled
18:25:44.571 -> CRC Length		= 16 bits
18:25:44.571 -> Address Length		= 5 bytes
18:25:44.571 -> Static Payload Length	= 4 bytes
18:25:44.571 -> Auto Retry Delay	= 1500 microseconds
18:25:44.571 -> Auto Retry Attempts	= 15 maximum
18:25:44.617 -> Packets lost on
18:25:44.617 ->     current channel	= 0
18:25:44.617 -> Retry attempts made for
18:25:44.617 ->     last transmission	= 15
18:25:44.617 -> Multicast		= Disabled
18:25:44.617 -> Custom ACK Payload	= Disabled
18:25:44.617 -> Dynamic Payloads	= Disabled
18:25:44.617 -> Auto Acknowledgment	= Enabled
18:25:44.617 -> Primary Mode		= TX
18:25:44.617 -> TX address		= 0x65646f4e32
18:25:44.617 -> pipe 0 ( open ) bound	= 0x65646f4e32
18:25:44.617 -> pipe 1 ( open ) bound	= 0x65646f4e31
18:25:44.617 -> pipe 2 (closed) bound	= 0xc3
18:25:44.617 -> pipe 3 (closed) bound	= 0xc4
18:25:44.617 -> pipe 4 (closed) bound	= 0xc5
18:25:44.617 -> pipe 5 (closed) bound	= 0xc6
18:25:44.663 -> Transmission failed or timed out
18:25:46.704 -> Transmission failed or timed out
18:25:48.701 -> Transmission failed or timed out
18:25:50.745 -> Transmission failed or timed out
18:25:52.779 -> Transmission failed or timed out
18:25:54.818 -> Transmission failed or timed out
18:25:56.850 -> Transmission failed or timed out
18:25:58.848 -> Transmission failed or timed out
18:26:00.899 -> Transmission failed or timed out
18:26:02.902 -> Transmission failed or timed out
18:26:04.949 -> Transmission failed or timed out
18:26:06.981 -> Transmission failed or timed out
18:26:08.985 -> Transmission failed or timed out
18:26:11.021 -> Transmission failed or timed out
18:26:13.063 -> Transmission failed or timed out
18:26:15.094 -> Transmission failed or timed out
18:26:17.102 -> Transmission failed or timed out
18:26:19.164 -> Transmission failed or timed out
18:26:21.185 -> Transmission failed or timed out
18:26:23.190 -> Transmission failed or timed out
18:26:25.240 -> Transmission failed or timed out
18:26:27.247 -> Transmission failed or timed out

I will occasionally get a success, but its rare. On the RPi I will also occasionally receive a "Received 4 bytes on pipe 5: 0" when I never set up pipe 5. I assume that's general interference but I'm not sure. 

I have swapped the rf24s, the power regulatror breakouts, and the antennas between eachother all with the same result. 

Any suggestions? Besides the shielding that I am going to try as soon as I can. Did I just buy bad boards? 

Let me know if I can provide any more information, thanks.

TMRh20, 2021-11-02:
How are you powering the device attached to the RPi? Usually these pa+lna modules require more power than the RPi can provide. It may be the same with the Arduino but I would start with a low power module connected to the RPi or try a different power supply/batteries.

watsonpaul6, 2021-11-03:
I am using a 5v out pin on the RPi connected to the 5v->3.3v breakout. Same for the Arduino, 5v.

2bndy5, 2021-11-03:
It is very likely a "not enough current to transmit" problem. Given the output on the RPi, I'd suspect the RPi isn't able to transmit the ACK packets. But, the received data values tell me there something else afoot. I see you've slowed the SPI speed, so it isn't likely a problem with the data connections. The shielding might help, but this smells like a power supply problem.

The PA/LNA modules are good example of when bigger is not better. Usually people only use those when they need to transmit beyond  ~32 ft. I've had better success with non PA/LNA modules reaching up to at least 50 ft with walls and stuff obstructing line of sight.

watsonpaul6, 2021-11-03:
That makes a lot of sense, thank you. Ill order a few of the non-PA/LNA modules and give those a go!

closed 2021-11-03 by watsonpaul6

2bndy5, 2021-11-03:
While you're shopping, I would suggest you also get some various sized capacitors (if you didn't already). The non-PA/LNA modules are still sensitive to noise from power supply.