nRF24/RF24 #806: [Question][Help Wanted] .writeAckPayload() dont work with some buttons library.

WeAreGeronimo, 2021-11-09:
Hello! I was read about common issues and did not find an answer on my question. Google also dont know about this problem. 
My question / problem is as follows: when i uncomment code fragment on last lines .writeAckPayload() won't work anymore, but it continues to receive an input signal from another module.
Software: RF24 latest version.
Hardware: arduino nano + nrf24 with amplifier and additional anthena (with voltage stabilization module 3.3V);
(I also tried on another arudino nano with built-in Wi-Fi - same situation). In fact, I have tried so many options.

Someone can tell me, what i do incorrect? 
Thank you in advance.
```
//#include <JC_Button.h> //................. im tryied this library
//#include <EncButton.h> //................. this
#include <ezButton.h> //....................... and this
#include <SPI.h>

#include "RF24.h"
//EncButton<EB_TICK, 2> powerButton;
ezButton powerButton(2);

RF24 radio(9, 10, 4000000); //............. im tried this after reading common issues, initial setting was <<RF24 radio(9, 10);>>

byte address[][2] = {"NodeS", "SlaveS",};
unsigned long lightTimer, sendTimer, clickTimer;
int settingsPayload[3];
int settingsAnswer[3];
int settingsHolder[3] = {2, 2 , 2};
int lastSettingsBeforeOff[3];
bool powerState1 = 1;
bool settingsChangedFromBase = false;

void setup() {
  Serial.begin(9600);
  radio.begin();

  radio.enableAckPayload();
  radio.openReadingPipe(1, address[0]);
  radio.setChannel(0x67);
  radio.setPALevel (RF24_PA_LOW);
  radio.setDataRate (RF24_1MBPS);
  radio.powerUp();
  radio.startListening();
  radio.writeAckPayload (1, &settingsHolder, sizeof(settingsHolder));
}

void loop() {

  //powerButton.tick(); //................... all buttons libraries add unique listening functions for connected button

  powerButton.loop();

  byte pipeNo;

  if (radio.available()) {
    radio.read(&settingsPayload, sizeof(settingsPayload));
    for (byte i = 0; i < 3; i++) {
      settingsHolder[i] = settingsPayload[i];
    }
    radio.writeAckPayload(1, &settingsHolder, sizeof(settingsHolder));
  }

  if (millis() - lightTimer > 1000) {
    for (byte i = 0; i < 3; i++) {
      Serial.print(i);
      Serial.print(",");
      Serial.print(settingsHolder[i]);
      Serial.print(";");

    }
    lightTimer = millis();
  }

  if (millis() - clickTimer > 500) {
    Serial.println(powerState1);
    clickTimer = millis();
  }

  //  if (powerButton.isPressed()) {      //..................if will uncommented this fragment, 
  //    Serial.println("On");                   //.................. .writeAckPayload() dont will be work 
  //    powerState1 = !powerState1;
  //    if (powerState1) {
  //
  //
  //    } else {
  //
  //      Serial.println("off");
  //    }
  //
  //  }

}
```

2bndy5, 2021-11-09:
I believe the answer you seek is [noted in the docs](https://nrf24.github.io/RF24/classRF24.html#a36b9fc363d3560358fe430a600a6f385). ACK Payloads are dynamically sized, so you need to `enableDynamicPayloads()` (on both RX & TX ends).

WeAreGeronimo, 2021-11-10:
> I believe the answer you seek is [noted in the docs](https://nrf24.github.io/RF24/classRF24.html#a36b9fc363d3560358fe430a600a6f385). ACK Payloads are dynamically sized, so you need to `enableDynamicPayloads()` (on both RX & TX ends).

Thank for answer. Nope. Dont helped. What is the relationship between commented out code and enableDynamicPayloads()? and why dynamic if i use an array with length 3?

2bndy5, 2021-11-10:
> What is the relationship between commented out code 

The commented out code has nothing to do with the RF24 lib, so I don't know why that would affect the radio at all. I suspect that if uncommenting the code helps then you are facing a timing issue because any `Serial.print*()` adds a significant delay to a program's run time.

> why dynamic if i use an array with length 3?

dynamic payload sizes has nothing to do with your code. It is more about how the radio handles ACK payloads over-the-air. It would be too technical to explain further. Just know that any application that uses ACK payloads needs to have dynamic payloads enabled (its not a recommendation - its a requirement). This is also demonstrated in the ACK payloads example.

WeAreGeronimo, 2021-11-10:
> > What is the relationship between commented out code
> 
> The commented out code has nothing to do with the RF24 lib, so I don't know why that would affect the radio at all. I suspect that if uncommenting the code breaks things, then you are facing a timing issue because any `Serial.print*()` adds a significant delay to a program's run time.
> 
> > why dynamic if i use an array with length 3?
> 
> dynamic payload sizes has nothing to do with your code. It is more about how the radio handles ACK payloads over-the-air. It would be too technical to explain further. Just know that any application that uses ACK payloads needs to have dynamic payloads enabled (its not a recommendation - its a requirement). This is also demonstrated in the ACK payloads example.

The magic is that it doesn't matter if you press the button or not. . ackpayload does not work even when the condition is not met.

```
   if (powerButton.isPressed()) {
      Serial.println("On");
      if (powerState1) {
        powerState1 = false; // this line break .ackPayload even when the condition is not met.
      } 
  
    }
```

2bndy5, 2021-11-10:
> I suspect that if uncommenting the code breaks things, then you are facing a timing issue

Your problem might be related to debouncing the button because this literally has to keep track of `millis()`.

I personally recommend you use a small capacitor in parallel with the button. This way your program doesn't have to overcompensate for electrical noise by using a button debouncing lib. Edge detection is easy enough to do without a library.

TMRh20, 2021-11-10:
`byte address[][2] = {"NodeS", "SlaveS",};`

Is creating a memory issue also I think. It should be `byte address[][6] = {"NodeS", "SlaveS",};`
When random code causes things to stop working, look for these types of issues in your code.

If you want to use shorter addresses, use setAddressWidth along with appropriate sized addresses.

WeAreGeronimo, 2021-11-10:
Thanks all for answers!

> > I suspect that if uncommenting the code breaks things, then you are facing a timing issue
> 
> Your problem might be related to debouncing the button because `powerButton.isPressed()` literally has to keep track of `millis()`.
> 
> I personally recommend you use a small capacitor in parallel with the button. This way your program doesn't have to overcompensate for electrical noise by using a button debouncing library. Edge detection is easy enough to do without a library.

I tried your guess and delete all buttons libraries, but it wasnt helped.

> `byte address[][2] = {"NodeS", "SlaveS",};`
>
> Is creating a memory issue also I think. It should be `byte address[][6] = {"NodeS", "SlaveS",};` When random code causes things to stop working, look [...]

Yes!! The problem was really this. After change to [6] everything started working properly. Also with buttons libraries. Thank you very much.

closed 2021-11-10 by WeAreGeronimo

2bndy5, 2021-11-10:
I would recommend you google the differences between strings and c-strings. Long story short:
```cpp
char address[6] = "Slaves" // this is actually 7 bytes in memory, but you only allocate 6
char address[7] = "Slaves" // in memory this looks like { 'S', 'l', 'a', 'v', 'e', 's', '\0' }

// thus no bad c-strings looks like
char address[2][6] = {"Nodes", Slave"};
```
If you weren't using an AVR based MCU, then your entire processor would freeze up when it encounters malformed c-strings.