nRF24/RF24 #808: Receiver is receiving nonexistent packets and garbling existing ones

foodini, 2021-11-15:
I have been through the "common problems" page and have tried to mitigate potential issues. (fail_check() just looks at failureDetected and halts with a message if it's nonzero. I set failureDetected to 0 after allocating the RF24 because after allocation, it indicates a failure. As you'll see, I can clearly send, so I thought the failrueDetected was bogus. But hopefully my problem is just somewhere there and this will be easy.
```
  m_rf24 = new RF24(RADIO_CE_PIN, RADIO_CS_PIN);
  m_rf24->failureDetected=0;
  fail_check(__LINE__);
  if(!m_rf24->begin()) {
    while(true) {
      Serial.println("Fatal failure initializing radio.");
      delay(1000);
    }
  }
  fail_check(__LINE__);
  m_rf24->setAddressWidth(5);
  m_rf24->setChannel(13);
  m_rf24->setDataRate(RF24_1MBPS);
  m_rf24->setAutoAck(false); //// maybe true?
  m_rf24->disableDynamicPayloads(); /// no idea
  m_rf24->disableAckPayload();
  m_rf24->setPayloadSize(sizeof(RadioMessage));
  m_rf24->setCRCLength(RF24_CRC_16);
  fail_check(__LINE__);
```

I set up for listening:
```
  if(m_sender) {  //m_sender has been verified to be true for the sender and false for the receiver.
    Serial.println("Setting up new Radio as a sender");
    m_rf24->openWritingPipe(address);
    m_rf24->setPALevel(RF24_PA_MIN);
    m_rf24->stopListening();
  } else {
    Serial.println("Setting up new Radio as a receiver");
    m_rf24->openReadingPipe(0, address);
    m_rf24->setPALevel(RF24_PA_MIN);
    m_rf24->startListening();
  }
  fail_check(__LINE__);
```

As soon as I start trying to get packets, I start getting garbage even though I've not powered up my sender. The packets look like this:
```
typedef struct {
  float altitude;
  float pressure;
  float temp;
  unsigned long proxy_id; //Using a char here will cause the struct to be different sizes on different hardware.
} RadioMessage;
```

And the code to receive a packet is this:
```
  Serial.println("Entering while(m_rf24->available())...");
  while(m_rf24->available()) {
    RadioMessage tmp;
    m_rf24->read(&tmp, sizeof(RadioMessage));

    Serial.print("Read:       ");
    debug_print(&tmp);
  }
  Serial.println("Exiting while(m_rf24->available())...");
```

debug_print() just dumps the content of the packet, bit-by-bit, to Serial. As soon as I power up the receiver, I start getting this:
```
Entering while(m_rf24->available())...
Exiting while(m_rf24->available())...
Entering while(m_rf24->available())...
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Exiting while(m_rf24->available())...
Entering while(m_rf24->available())...
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Exiting while(m_rf24->available())...
Entering while(m_rf24->available())...
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Exiting while(m_rf24->available())...
Entering while(m_rf24->available())...
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Read:       00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0.00m 0.00mbar 0.00C ID:0
Exiting while(m_rf24->available())...
```

Again, this is nothing broadcasting. If I do send a few packets, 1) they're garbled when they arrive, and 2) every subsequent "ghost" packet will no longer be all zeroes. The ghosts will have other "stuff" in them that won't change until a new packet is received. It'll even survive a reboot of the hardware. The only thing that will take it back to zeroes in to power down the processor.

For the sake of debugging, I replaced the floating point values with just 16 bytes of incrementing unsigned chars:
```
unsigned char * char_ptr = (unsigned char *)&message;
for(int i=0; i<sizeof(message); i++)
  char_ptr[i] = i+1;
```

When the debug_print outputs this, I see:
```
Sent: 00000001 00000010 00000011 00000100 00000101 00000110 00000111 00001000 00001001 00001010 00001011 00001100 00001101 00001110 00001111 00010000 0.00m 0.00mbar 0.00C ID:269422093
```

So that's what I should see on the receiving end. However, what is read back is:
```
Read: 00000000 00000001 00000001 00000010 00000010 00000011 00000011 00000100 00000100 00000101 00000101 00000110 00000110 00000111 00000111 00001000 0.00m 0.00mbar 0.00C ID:134678278 bytes:16
```

It's interesting to note that the received packet is _similar_ to what was expected, but not correct. I've tried this with 10 different radios now on multiple different processors. I have seen successful reads before, and have confirmed that all my boards can send correctly, but the problem seems to be on the receiving end. Particularly since after getting a good packet, the bogus/ghost packets have invalid data in them. Regardless of what's happening on the sending end, the receiver should not be handing me anything that wasn't sent.

What's going on here? I thought I had some bogus radios because, after setting up 10 radios and doing nearly the entire grid of testing, nearly every radio had shown the ability to send reliably, but less than half could receive. I quarantined every radio that failed that test and am now using hardware that worked 100% before, but I'm now getting this.

I've been very careful with static. I've not shorted any of the radios. I've re-checked my wiring more times than I can count. I'm totally out of ideas here.

2bndy5, 2021-11-15:
> I have been through the "common problems" page

Did you try the advice about lowering the SPI speed?

foodini, 2021-11-15:
> > I have been through the "common problems" page
> 
> Did you try the advice about lowering the SPI speed?

There's nothing on the troubleshooting page (https://nrf24.github.io/RF24/md_COMMON_ISSUES.html) about it. Looking at where spi_speed is mentioned in the class documentation, it's not obvious that this is a fix for anything. Maybe I missed something.

I cranked down my speed to 500k and I seem to be getting a correct result on the first test. I could binary search for a working value then build in a safety margin, but it would be better to just ask what should be a safe value?

Thanks. I was really tearing my hair out over this one.

2bndy5, 2021-11-15:
The advice maybe a newer addition to the COMMON_ISSUES.md doc, so it may not have been in the hosted doxygen-rendered html related pages.

It is in the github copy: [Here you go](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#payloads-receivedsent-are-inaccurate-or-printdetails-outputs-the-unexpected-values)

Originally the RF24 lib used a default of 4MHz because that's what the AVR core had defaulted to, and ManiacBug found it most reliable. It has since been changed to the max to push the limits of the radio. However, some boards seem to have super sensitive SPI buses (I've been seeing this problem from people with the Arduino Nano 33 BLE), or it could be something up with the actual wires your using (maybe too long or simply terminated loosely).

closed 2021-12-01 by 2bndy5