nRF24/RF24 #809: Arduino web editor compile errors

DavethaSlave, 2021-11-22:
Hello all. I am new to GitHub,  it I have been playing with Arduinos for a couple of years now. I just started using the online web editor to use the cloud dashboard functions. I use the RF24 library with the desktop IDE with no issues. I tried to upload a known good sketch into the web editor and it won’t compile. I have checked and make sure that the needed libraries are installed and it seems fine. The specific error message is ‘const void*’ not a pointer-to-object type. What does this mean and how do you fix it?  Thank you

2bndy5, 2021-11-22:
I'm not entirely sure what  to make of this error output. It looks like it doesn't like the esp8266 core's definition of `pgm_read_ptr`

```
/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:76:81: error: 'const void*' is not a pointer-to-object type

#define pgm_read_ptr(addr) (*reinterpret_cast<const void*>(addr))
```
@DavethaSlave The RF24 lib's use of `pgm_read_ptr` is cross-platform compatible - meaning its definition is dependent on the platform specified to the compiler. I think this may be more of a problem with the esp8266 core or the compiler used in the Arduino web editor (or both).

2bndy5, 2021-11-26:
@DavethaSlave if you take out the call to `printDetails()` or `printPrettyDetails()`, then I believe the compiler should stop complaining. We can't fix this on our end.

closed 2022-02-11 by 2bndy5